TheImagingSource/tiscamera #101: Question:  Fixed exposure through gstreamer

ben00052, 2017-06-29:
Is there any way to set a fixed exposure time through Gstreamer, as opposed to using tcamautoexposure in the pipeline to let it find one iteself?

TIS-Stefan, 2017-06-29:
Hello

The answer to your question is "it depends upon...."

Which camera do you use? 
Which Gstreamerversion do you use?
Do you use the pipeline within a program or from command line only?

ben00052, 2017-06-29:
Sorry for not giving enough information.  Using a DFK 27AUJ003 from within a C++ application using GStreamer 1.10.4 (on fedora 25 with KDE if it matters).

TIS-Stefan, 2017-06-30:
OK, I see, I did not ask enough questions. However, I will try to give a start. I presume, you use our Gstreamer modules. I guess, you have all these little GStreamer preparations in your code.
```
GValue val = {}; 
g_value_init(&val, G_TYPE_INT);
g_value_set_int(&val, Value);  // Value can be the exposure time to be set.
// Now lets set the value:
tcam_prop_set_tcam_property( (TcamProp *)_TcamBin,( char*)Property, &val);
```
The variable "Property" is the property name, as return by e.g. the tcam-ctrl program when listing available properties.

The variable "_TcamBin" contains the pointer of the tcambin or tcamsrc GStreamer module.
You can query the tcambin from a pipeline as follows:

```
GstElement *_TcamBin; 
GstElement  *_Pipeline

// Create the pipeline as string:
sprintf(pipeline, "tcambin serial=\"%s\" name=source ! %s,format=\"%s\",width=%d,height=%d,framerate=%d/%d ! queue ! videoconvert ! videoscale ! ximagesink async=false name=display",
			_serial,
			_VideoFormat.getType(),
			_VideoFormat.getFormat(),
			_VideoFormat._width,
			_VideoFormat._height,
			_VideoFormat.getFrameRate()._numerator,
			_VideoFormat.getFrameRate()._denominator
   		); 

_Pipeline.pipeline = gst_parse_launch (pipeline, &error); 
// Now get the pointer to the tcambin:
GstElement *_Tcambin= gst_bin_get_by_name(GST_BIN(_pipeline), "source"); 
```
I hope, this helps.

Stefan

ben00052, 2017-06-30:
I am actually currently using a Qt Gstreamer wrapper, but if I need to move away from it to get functionality to manually control exposure, so be it.  

```
   m_pipeline      = QGst::Pipeline::create("Video Player + Snapshots pipeline");
    m_source        = QGst::ElementFactory::make("tcamsrc", "tcamsrc");
    m_videoSink     = QGst::ElementFactory::make("xvimagesink", "video-sink");
    m_videoConvert  = QGst::ElementFactory::make("videoconvert", "videoconvert");
    m_Filter       = QGst::ElementFactory::make("capsfilter", "capsfilter");
    m_Bayer2Rgb     = QGst::ElementFactory::make("bayer2rgb","bayer2rgb");
    m_AutoWhiteBalance = QGst::ElementFactory::make("tcamwhitebalance","tcamwhitebalance");
    m_Filter->setProperty("caps", QGst::Caps::fromString("video/x-bayer, width=3872, height=2764, framerate=6/1"));

    m_pipeline->setProperty("video-sink", m_videoSink);
    m_videoSink->setProperty("enable-last-sample", true);
    m_pipeline->add(m_source, m_Filter, m_AutoWhiteBalance, m_Bayer2Rgb,m_videoConvert, m_videoSink);

    m_source->link(m_Filter);
    m_Filter->link(m_AutoWhiteBalance);
    m_AutoWhiteBalance->link(m_Bayer2Rgb);
    m_Bayer2Rgb->link(m_videoConvert);
    m_videoConvert->link(m_videoSink);

    QGlib::connect(m_pipeline->bus(), "message", this, &Player::onBusMessage);
    m_pipeline->bus()->addSignalWatch();
    m_pipeline->bus().clear();

    watchPipeline(m_pipeline);
    m_pipeline->setState(QGst::StatePlaying);
```

Which creates something equivelent to
`gst-launch-1.0 tcamsrc  ! video/x-bayer,width=3872,height=2764,framerate=6/1 ! tcamwhitebalance ! bayer2rgb ! videoconvert ! xvimagesink`

I had originally been using tcambin instead of tcamsrc, but for what I am doing tcamautoexposure was causing problems, hence looking for a way to set exposure to a specific value and trust that it will stay there.

ben00052, 2017-08-04:
Thank you.  I did get this working thanks to your advice, testing it just got pushed back significantly because of competing priorities.

closed 2017-08-09 by TIS-Arne