TheImagingSource/tiscamera #102: DFK 33UX264 in GStreaming through v4l2src in Python. Problems setting/getting properties and slow gain adaptation

ivanprado, 2017-07-19:
Hi, 

I'm writing an app that uses camera DFK 33UX264 from Ubuntu 14.04 from Python using GStreamer 1.8.0 and v4l2src as source. 

The equivalente gstreamer pipeline I'm using is:

`v4l2src ! videoconvert ! appsink ! video/x-raw, format=(string){BGR}, width=(int)2448, height=(int)2048, framerate=(fraction)25/1`

I have some issues.

## 1) Impossible to set camera setting from GStreaming v4l2src

I have tried to manage camera controls through v4l2src. For some properties it works, but some others are not changed. I have found a workaround that is to launch `v4l2-ctl` through a system call, and it works. So that is my *first question*, why some properties cannot be set though v4l2src? 

That is the way I'm setting the properties: 

```python
    srcprop = Gst.Structure.new_empty("extra-controls")
    for name, value in config.items():
        lg.info("Setting camera property '{}' to '{}'".format(name, value))
        srcprop.set_value(name, value)
    v4l2src_element.set_property("extra-controls", srcprop)
```
## 2) Imposible to get actual exposure time and gain on automatic modes

I would like to know which is the exact exposure time or gain used to take the photos. I have tried to obtain them through the properties `exposure_time_us` and `gain` but they don't reflect real values (they are just fixed a some particular value). How can I know which the actual value of exposure and gain used by the camera?

## 3) Slow gain adaptation

I'm using the camera in trigger mode with the software trigger. I fix the exposure time and leave the gain free to adapt in automatic mode. The rate I'm using is one shoot each half a second or so. I have noticed that gain adaptation is too slow. It is a problem in situations of high light change, which in my case happen many times as photos are taken on night with a light: when there are no object, scene is too dark. When objects start to come, first photos get burned. It takes many shoots until the camera adapts. Any way where we can change gain adaptation so that it gets much faster?

## 4) Instabilities. 

I have noticed some instabilities on the pipeline:

* Some times I only get empty samples. 
* I'm finishing the pipeline by sending an EOS and then setting the pipeline state to NULL, but it is not always working. Some times EOS is ignored. Other times pipeline is not ended, so I have to kill the app manually.
* At some points camera gets unstable, and I have to unplug/plug it to get it working. I have noticed that this problem doesn't affect guvcview, as it is always working.

One possibility is that maybe I'm not starting/finish the pipeline properly, but I have not good references/documentation about how to do it properly.

TIS-Edgar, 2017-07-21:
Hi,

1. 
Is there a reason you are not using out elements? 
The "extra-controls" property is relatively new. Since other applications handle the camera properties correctly I would assume v4l2src is not behaving correctly. Does v4l2src say something when you enable debug?
2.
This is probably due to an older firmware.
You can check your version with the firware-update tool in our repository.
    sudo tools/firmware-update/firmware-update -id <SERIAL>

3.
The auto algorithms can only work when the camera is actively recording.
Since trigger mode is active it does not record, thus the algorithms only get the frames you trigger.
4.
Do you manually send an EOS? Simply going to state NULL should be enough, EOS, etc. have to be automatically handled by v4l2src.

ivanprado, 2017-07-28:
My firmware version is:

```
Device manufacturer: The Imaging Source Europe GmbH
Product name:        DFK 33UX264
Serial number:       28710063
VendorID:ProductID:  199e:9089
Firmware version:    USB3c2-IMX/47
```
Should I update framework to be able to get the gain the camera has selected? That's what I obtain when getting Gain from camera:

```
tcam-ctrl -p 28710063|grep "Gain  "
Gain                                    (int) min=0 max=480 step=1 default=0 value=0 category=Exposure
```
As you can see, reported gain is 0. 

Regarding the rest of questions, I'll do some trials and will report results.

TIS-Edgar, 2017-07-28:
If you could send an email to support@theimagingsource.com referencing this issue.
We will send you a firmware file to update your camera. That will fix the update issues with the camera properties.

TIS-Stefan, 2017-07-28:
Hello

If I do
`v4l2-ctl -d /dev/video0 -c gain=4`
There is an effect.
`v4l2-ctl -d /dev/video0 -C gain`
returns the correct, previously set value.
But if Gain automatic is enabled and I see Gain rising in the image, because there is more noise and the image becomes brighter, then
`v4l2-ctl -d /dev/video0 -C gain`
returns always 4.
Same goes with tcam-ctrl. Set Gain value has an effect on the image and the read value is the previously set value. But if Gain automatic is enabled, the returned value is always the value, I have set previously, even in Gain automatic did advance the Gain.
In Windows the Gain value is always updated and readable, when automatic is running.

Stefan

ivanprado, 2017-07-28:
Exactly, that's the behavior I'm seen. I would like to be able to see which gain the came has selected when automatic gain is set on Linux. Is that possible? Would it be possible with the new firmware?

TIS-Arne, 2017-08-09:
This seems to be an issue with the Linux uvcvideo kernel driver. We are currently investigating this issue. Most likely a fix for this issue will require a kernel update though.

ivanprado, 2017-08-09:
@TIS-Arne thanks.

ivanprado, 2017-08-16:
Hi @TIS-Arne. Any news on gain issue?

TIS-Arne, 2017-08-17:
Our developer prepared a patch but it needs to be reworked in order to get accepted by the kernel developer. If you are able to apply the patch to a kernel source tree and compile your own kernel you could find the first version of the patch here:

http://www.mail-archive.com/linux-media@vger.kernel.org/msg117087.html

devellison, 2018-05-11:
Running into this now - any updates or is this where it still stands?

TIS-Edgar, 2018-05-15:
The concerning patch has been merged into the mainline kernel in late February.
That means all linux kernel >= 4.16 should not exhibit this issue.

closed 2018-06-22 by TIS-Stefan