TheImagingSource/tiscamera #105: Software trigger stops working

ivanprado, 2017-08-02:
Hi, 

I'm having the following issue. I have my own GStreamer pipeline equivalent to "v4l2src ! videoconvert ! mycustomsink" where mycustomsink is a sink I have written in python that just get the buffers and save them. I want to trigger the photos manually, so I enabled the trigger mode. 

The pipeline works well, but some times it stops receiving images. It seems the camera gets to a inconsistent state or so, as issue persist even if I restart the pipeline. In that case, no photo is received at all. That are the state of the properties at that moment:

```
tcam-ctrl -p 28710063
Found 46 propert(y/ies)
Brightness                              (int) min=0 max=4095 step=1 default=240 value=240 category=Exposure
Saturation                              (int) min=0 max=255 step=1 default=64 value=64 category=Color
Hue                                     (int) min=-180 max=180 step=1 default=0 value=0 category=Color
Whitebalance Auto                       (bool) default=true value=true category=Color
Whitebalance Red                        (int) min=0 max=255 step=1 default=64 value=67 category=Color
Whitebalance Blue                       (int) min=0 max=255 step=1 default=64 value=209 category=Color
Gamma                                   (int) min=1 max=500 step=1 default=100 value=100 category=Image
Gain                                    (int) min=0 max=480 step=1 default=0 value=0 category=Exposure
Exposure                                (int) min=25 max=4000000 step=1 default=333 value=33532 category=Exposure
Exposure Auto                           (bool) default=true value=false category=Exposure
Exposure Auto Reference                 (int) min=0 max=255 step=1 default=128 value=90 category=Exposure
Gain Auto                               (bool) default=true value=true category=Exposure
Trigger Mode                            (bool) default=false value=true category=Special
Software Trigger                        (button) category=Special
Trigger Delay (us)                      (int) min=0 max=1000000 step=1 default=0 value=31 category=Special
Strobe Enable                           (bool) default=false value=false category=Special
Strobe Polarity                         (bool) default=false value=false category=Special
Strobe Exposure                         (bool) default=true value=true category=Special
Strobe Duration                         (int) min=0 max=32767 step=1 default=100 value=100 category=Special
Strobe Delay                            (int) min=0 max=32767 step=1 default=0 value=0 category=Special
GPOut                                   (int) min=0 max=1 step=1 default=0 value=0 category=Special
GPIO                                    (int) min=0 max=1 step=1 default=0 value=0 category=Special
Offset X                                (int) min=0 max=2192 step=2 default=0 value=0 category=Partial Scan
Offset Y                                (int) min=0 max=2044 step=2 default=0 value=0 category=Partial Scan
Offset Auto Center                      (bool) default=true value=true category=Partial Scan
Trigger Polarity                        (enum)  default=RisingEdge category=Special
						value=FallingEdge
							FallingEdge
							RisingEdge
Trigger Exposure Mode                   (enum)  default=FrameStart category=Special
						value=FrameStart
							ExposureActive
							FrameStart
Trigger Burst Count                     (int) min=1 max=1000 step=1 default=1 value=1 category=Special
Trigger Debounce Time (us)              (int) min=0 max=1000000 step=1 default=0 value=0 category=Special
Trigger Mask Time (us)                  (int) min=0 max=1000000 step=1 default=0 value=0 category=Special
Trigger Noise Surpression Time (us)     (int) min=0 max=1000000 step=1 default=0 value=0 category=Special
Auto Functions ROI Control              (bool) default=true value=true category=Special
Auto Functions ROI Left                 (int) min=0 max=2432 step=1 default=0 value=480 category=Special
Auto Functions ROI Top                  (int) min=0 max=2032 step=1 default=0 value=270 category=Special
Auto Functions ROI Width                (int) min=16 max=2448 step=1 default=0 value=960 category=Special
Auto Functions ROI Height               (int) min=16 max=2048 step=1 default=0 value=540 category=Special
Whitebalance                            (bool) default=false value=false category=Color
Whitebalance Auto Preset                (enum)  default=Any    category=Color
						value=Any
							Any
							Cool White
							Daylight
							Deep Shade
							Warm White
Whitebalance Green                      (int) min=0 max=255 step=1 default=64 value=64 category=Color
Whitebalance Temperature                (int) min=2500 max=10000 step=100 default=6500 value=4000 category=Color
Flip Horizontal                         (bool) default=false value=false category=Unknown
Flip Vertical                           (bool) default=false value=false category=Unknown
Highlight Reduction                     (bool) default=false value=false category=Unknown
Exposure Auto Upper Limit Auto          (bool) default=true value=true category=Unknown
Exposure Auto Lower Limit               (int) min=25 max=1000000 step=1 default=333 value=60 category=Unknown
Exposure Auto Upper Limit               (int) min=25 max=1000000 step=1 default=333 value=11111 category=Unknown
```

The unique way of make it back to work is re-plugging the camera. 

My camera is:
```
Device manufacturer: The Imaging Source Europe GmbH
Product name:        DFK 33UX264
Serial number:       28710063
VendorID:ProductID:  199e:9089
Firmware version:    USB3c2-IMX/47
```

I'm using GStreamer 1.8, Ubuntu 14.04.

ivanprado, 2017-08-02:
Please, if it would be possible assign that with priority as it is blocking for me and I need the camera for being used in crops that are mature at this precise moment, so I have no margin. 

More data just if it is useful. 

If I open the camera using VLC at that state I got:

```
[0x7f46cc003c08] v4l2 demux warning: cannot reset cropping limits: Inappropriate ioctl for device
[0x7f46d0000a68] main input error: Invalid PCR value in ES_OUT_SET_(GROUP_)PCR !
```
And nothing is shown. Then I run 

`v4l2-ctl -d /dev/video0 -c trigger_mode=0`

And I open VLC, and I can see a video but for a small amount of seconds. Then it freezes, and message is the same:

```
[0x7f3dd0003c08] main input error: Invalid PCR value in ES_OUT_SET_(GROUP_)PCR !
```

If I open guvcview, nothing is shown, and I got these messages:

```
GUVCVIEW: version 2.0.3
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm_dmix.c:1022:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
bt_audio_service_open: connect() failed: Connection refused (111)
bt_audio_service_open: connect() failed: Connection refused (111)
bt_audio_service_open: connect() failed: Connection refused (111)
bt_audio_service_open: connect() failed: Connection refused (111)
ALSA lib pcm_equal.c:196:(_snd_pcm_equal_open) No slave configuration for equal pcm
ALSA lib pcm_equal.c:196:(_snd_pcm_equal_open) No slave configuration for equal pcm
ALSA lib pcm_dmix.c:1022:(snd_pcm_dmix_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
V4L2_CORE: ioctl (-1067952623) retried 4 times - giving up: Resource temporarily unavailable)
V4L2_CORE: (VIDIOC_DQBUF) Unable to dequeue buffer: Resource temporarily unavailable
V4L2_CORE: Could not grab image (select timeout): Resource temporarily unavailable
V4L2_CORE: Could not grab image (select timeout): Resource temporarily unavailable
V4L2_CORE: Could not grab image (select timeout): Resource temporarily unavailable
V4L2_CORE: Could not grab image (select timeout): Resource temporarily unavailable
V4L2_CORE: Could not grab image (select timeout): Resource temporarily unavailable
```
Running 
`GST_DEBUG=*:4 gst-launch-1.0 "v4l2src ! videoconvert ! autovideosink"`

just freezes. These are the logs:

```
0:00:00.000050510 19216      0x20a3a00 INFO                GST_INIT gst.c:502:init_pre: Initializing GStreamer Core Library version 1.2.4
0:00:00.000080339 19216      0x20a3a00 INFO                GST_INIT gst.c:503:init_pre: Using library installed in /usr/lib/x86_64-linux-gnu
0:00:00.000096887 19216      0x20a3a00 INFO                GST_INIT gst.c:513:init_pre: Linux ivan-Dell-Precision-M3800 4.4.0-83-generic #106~14.04.1-Ubuntu SMP Mon Jun 26 18:10:19 UTC 2017 x86_64
0:00:00.000552360 19216      0x20a3a00 INFO                GST_INIT gstmessage.c:123:_priv_gst_message_initialize: init messages
0:00:00.000630251 19216      0x20a3a00 INFO                GST_INIT gstcontext.c:77:_priv_gst_context_initialize: init contexts
0:00:00.000756590 19216      0x20a3a00 INFO      GST_PLUGIN_LOADING gstplugin.c:317:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.000845895 19216      0x20a3a00 INFO      GST_PLUGIN_LOADING gstplugin.c:225:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.000853591 19216      0x20a3a00 INFO      GST_PLUGIN_LOADING gstplugin.c:227:gst_plugin_register_static: added static plugin "staticelements", result: 1
[... 657 lines cut ...]
0:00:12.255415057 19558      0x1d5da00 INFO         GST_REFCOUNTING gstelement.c:2953:gst_element_finalize:<tempsink> finalize parent
0:00:12.255426987 19558      0x1d5da00 INFO         GST_REFCOUNTING gstelement.c:2873:gst_element_dispose:<autovideosink0> dispose
0:00:12.255433496 19558      0x1d5da00 INFO        GST_ELEMENT_PADS gstelement.c:766:gst_element_remove_pad:<autovideosink0> removing pad 'sink'
0:00:12.255446670 19558      0x1d5da00 INFO         GST_REFCOUNTING gstelement.c:2917:gst_element_dispose:<autovideosink0> parent class dispose
0:00:12.255453708 19558      0x1d5da00 INFO         GST_REFCOUNTING gstelement.c:2948:gst_element_finalize:<autovideosink0> finalize
0:00:12.255460147 19558      0x1d5da00 INFO         GST_REFCOUNTING gstelement.c:2953:gst_element_finalize:<autovideosink0> finalize parent
0:00:12.255469607 19558      0x1d5da00 INFO        GST_ELEMENT_PADS gstpad.c:1877:gst_pad_unlink: unlinking tcambin0:src(0x1f20110) and videoconvert0:sink(0x1f2a110)
0:00:12.255483198 19558      0x1d5da00 INFO        GST_ELEMENT_PADS gstpad.c:1931:gst_pad_unlink: unlinked tcambin0:src and videoconvert0:sink
[... cut ...]
0:00:12.256324517 19558      0x1d5da00 INFO         GST_REFCOUNTING gstelement.c:2917:gst_element_dispose:<pipeline0> parent class dispose
0:00:12.256329930 19558      0x1d5da00 INFO         GST_REFCOUNTING gstelement.c:2948:gst_element_finalize:<pipeline0> finalize
0:00:12.256333624 19558      0x1d5da00 INFO         GST_REFCOUNTING gstelement.c:2953:gst_element_finalize:<pipeline0> finalize parent
0:00:12.256339668 19558      0x1d5da00 INFO                GST_INIT gst.c:938:gst_deinit: deinitializing GStreamer
0:00:12.258828717 19558      0x1d5da00 INFO                GST_INIT gst.c:1068:gst_deinit: deinitialized GStreamer
```

TIS-Edgar, 2017-08-02:
What hardware are you using? 

    V4L2_CORE: (VIDIOC_DQBUF) Unable to dequeue buffer: Resource temporarily unavailable

This often indicates a problem with the usb connection, be it bandwidth, power or something else.
Also why are you still on the old firmware?

Does the problem also occur when you use a small resolution with a low framerate? Something like 640x480@5fps.

The gst pipeline would be:

    gst-launch-1.0 --gst-debug=tcam*:5 --gst-debug-no-color tcambin ! video/x-raw,width=640,height=480,framerate=5/1

ivanprado, 2017-08-02:
Hi Edgar, 

My camera is DFK33UX264 with a 3m USB3 wire sold with the camera (USB 3.0 Cable E119932 AWM 2725 80ºC 30V VW-1 COPARTNER). I plug the camera to a Dell Precision M3800 laptop.

I didn't know I was with an older firmware. I requested one by email but the answer was no new firmware was available. I'll request it again. 

I just want to notice that the problems start once the camera gets a particular wrong state, because before it works properly (even at high framerates and resolutions)

This is the result of the pipeline you suggest when the camera is in the bad state (nothing is shown, no buffers flowing):

```
 gst-launch-1.0 --gst-debug=tcam*:5 --gst-debug-no-color tcambin ! video/x-raw,width=640,height=480,framerate=5/1 ! videoconvert ! autovideosink

(gst-plugin-scanner:23461): GStreamer-WARNING **: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsteglglessink.so': /usr/lib/x86_64-linux-gnu/libgstegl-1.0.so.0: undefined symbol: eglDestroyImageKHR
0:00:00.514203700 23460       0x9b50f0 DEBUG                tcambin /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcambin.cpp:1228:gst_tcambin_init: init
Setting pipeline to PAUSED ...
0:00:00.519225893 23460       0x9b50f0 INFO                 tcambin /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcambin.cpp:1037:gst_tcambin_change_state: NULL_TO_READY
0:00:00.519237722 23460       0x9b50f0 DEBUG                tcambin /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcambin.cpp:767:gst_tcambin_create_source: Creating source...
0:00:00.521185073 23460       0x9b50f0 INFO                 tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1586:gst_tcam_src_set_property: Set camera name to 
0:00:00.521209525 23460       0x9b50f0 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1333:gst_tcam_src_change_state: State change: NULL -> READY
0:00:00.521213356 23460       0x9b50f0 INFO                 tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1337:gst_tcam_src_change_state: must initialize device
0:00:00.521218284 23460       0x9b50f0 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1209:gst_tcam_src_init_camera:<tcambin-source> Initializing device.
0:00:02.525881074 23460       0x9b50f0 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1219:gst_tcam_src_init_camera:<tcambin-source> Found 1 devices.
0:00:02.525905893 23460       0x9b50f0 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1227:gst_tcam_src_init_camera:<tcambin-source> No serial given. Opening first available device.
0:00:02.615592280 23460       0x9b50f0 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:675:gst_tcam_src_get_all_camera_caps: Found 5 pixel formats
0:00:02.615618024 23460       0x9b50f0 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:691:gst_tcam_src_get_all_camera_caps: Found 'video/x-bayer, format=(string)rggb' pixel format string
0:00:02.615628126 23460       0x9b50f0 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:697:gst_tcam_src_get_all_camera_caps: Found 4 resolutions
0:00:02.615682251 23460       0x9b50f0 ERROR                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:685:gst_tcam_src_get_all_camera_caps: Format has empty fourcc. Ignoring.
0:00:02.615687339 23460       0x9b50f0 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:691:gst_tcam_src_get_all_camera_caps: Found 'video/x-raw, format=(string)YUY2' pixel format string
0:00:02.615693034 23460       0x9b50f0 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:697:gst_tcam_src_get_all_camera_caps: Found 4 resolutions
0:00:02.615728149 23460       0x9b50f0 ERROR                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:685:gst_tcam_src_get_all_camera_caps: Format has empty fourcc. Ignoring.
0:00:02.615732014 23460       0x9b50f0 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:691:gst_tcam_src_get_all_camera_caps: Found 'video/x-raw, format=(string)BGR' pixel format string
0:00:02.615736790 23460       0x9b50f0 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:697:gst_tcam_src_get_all_camera_caps: Found 4 resolutions
0:00:02.615834036 23460       0x9b50f0 INFO                 tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:833:gst_tcam_src_get_all_camera_caps: video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction){ 300/1, 5000000/20833, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-bayer, format=(string)rggb, width=(int)1920, height=(int)1080, framerate=(fraction){ 90/1, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-bayer, format=(string)rggb, width=(int)2048, height=(int)2048, framerate=(fraction){ 45/1, 30/1, 15/1, 5/1, 1/1 }; video/x-bayer, format=(string)rggb, width=(int)2448, height=(int)2048, framerate=(fraction){ 35/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, framerate=(fraction){ 600/1, 480/1, 5000000/20833, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, framerate=(fraction){ 90/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)YUY2, width=(int)2048, h [... cut ...]
0:00:02.635943750 23460       0xb7e230 INFO                 tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1176:gst_tcam_src_set_caps: Start acquisition
0:00:02.642542385 23460       0xb7e230 INFO                 tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1191:gst_tcam_src_set_caps: Successfully set caps to: video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction)5/1
```
I don't think problem is related with USB bandwith or wire, as I'm just getting one full resolution photo each 700 ms or so, so I'm far from full frame rate.

ivanprado, 2017-08-02:
Unrelated problem, 

When running (on a stable state):

`gst-launch-1.0 --gst-debug=tcam*:5 --gst-debug-no-color tcambin ! video/x-raw,format=BGR,width=2448,height=2048,framerate=25/1 ! videoconvert ! autovideosink`

I got the following error:

```
 gst-launch-1.0 --gst-debug=tcam*:5 --gst-debug-no-color tcambin ! video/x-raw,format=BGR,width=2448,height=2048,framerate=25/1 ! videoconvert ! autovideosink 
0:00:00.018980516 24540       0x93d360 DEBUG                tcambin /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcambin.cpp:1228:gst_tcambin_init: init
Setting pipeline to PAUSED ...
0:00:00.024405369 24540       0x93d360 INFO                 tcambin /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcambin.cpp:1037:gst_tcambin_change_state: NULL_TO_READY
0:00:00.024420350 24540       0x93d360 DEBUG                tcambin /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcambin.cpp:767:gst_tcambin_create_source: Creating source...
0:00:00.026240858 24540       0x93d360 INFO                 tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1586:gst_tcam_src_set_property: Set camera name to 
0:00:00.026255395 24540       0x93d360 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1333:gst_tcam_src_change_state: State change: NULL -> READY
0:00:00.026259415 24540       0x93d360 INFO                 tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1337:gst_tcam_src_change_state: must initialize device
0:00:00.026264541 24540       0x93d360 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1209:gst_tcam_src_init_camera:<tcambin-source> Initializing device.
0:00:02.031531787 24540       0x93d360 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1219:gst_tcam_src_init_camera:<tcambin-source> Found 1 devices.
0:00:02.031558042 24540       0x93d360 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1227:gst_tcam_src_init_camera:<tcambin-source> No serial given. Opening first available device.
0:00:02.123247302 24540       0x93d360 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:675:gst_tcam_src_get_all_camera_caps: Found 5 pixel formats
0:00:02.123272765 24540       0x93d360 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:691:gst_tcam_src_get_all_camera_caps: Found 'video/x-bayer, format=(string)rggb' pixel format string
0:00:02.123288577 24540       0x93d360 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:697:gst_tcam_src_get_all_camera_caps: Found 4 resolutions
0:00:02.123352523 24540       0x93d360 ERROR                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:685:gst_tcam_src_get_all_camera_caps: Format has empty fourcc. Ignoring.
0:00:02.123359075 24540       0x93d360 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:691:gst_tcam_src_get_all_camera_caps: Found 'video/x-raw, format=(string)YUY2' pixel format string
0:00:02.123371673 24540       0x93d360 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:697:gst_tcam_src_get_all_camera_caps: Found 4 resolutions
0:00:02.123415913 24540       0x93d360 ERROR                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:685:gst_tcam_src_get_all_camera_caps: Format has empty fourcc. Ignoring.
0:00:02.123421181 24540       0x93d360 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:691:gst_tcam_src_get_all_camera_caps: Found 'video/x-raw, format=(string)BGR' pixel format string
0:00:02.123432600 24540       0x93d360 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:697:gst_tcam_src_get_all_camera_caps: Found 4 resolutions
0:00:02.123551600 24540       0x93d360 INFO                 tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:833:gst_tcam_src_get_all_camera_caps: video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction){ 300/1, 5000000/20833, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-bayer, format=(string)rggb, width=(int)1920, height=(int)1080, framerate=(fraction){ 90/1, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-bayer, format=(string)rggb, width=(int)2048, height=(int)2048, framerate=(fraction){ 45/1, 30/1, 15/1, 5/1, 1/1 }; video/x-bayer, format=(string)rggb, width=(int)2448, height=(int)2048, framerate=(fraction){ 35/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, framerate=(fraction){ 600/1, 480/1, 5000000/20833, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, framerate=(fraction){ 90/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)YUY2, width=(int)2048, height=(int)2048, framerate=(fraction){ 40/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)YUY2, width=(int)2448, height=(int)2048, framerate=(fraction){ 35/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)BGR, width=(int)640, height=(int)480, framerate=(fraction){ 400/1, 5000000/20833, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)BGR, width=(int)1920, height=(int)1080, framerate=(fraction){ 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)BGR, width=(int)2048, height=(int)2048, framerate=(fraction){ 25/1, 15/1, 5/1, 1/1 }; video/x-raw, f [... cut ...]
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
0:00:02.274530514 24540       0x93d360 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1315:gst_tcam_src_stop:<tcambin-source> Stopped acquisition
Setting pipeline to NULL ...
Freeing pipeline ...
Caught SIGSEGV
```

Meanwhile executing it using v4l2src works:

`gst-launch-1.0 --gst-debug=tcam*:5 --gst-debug-no-color v4l2src ! video/x-raw,format=BGR,width=2448,height=2048,framerate=25/1 ! videoconvert ! autovideosink`

ivanprado, 2017-08-02:
New findings: 

It seems that problem is not related with trigger mode. I have seen video freeze on video (video/x-raw,format=BGR,width=2448,height=2048,framerate=25/1) using v4l2src. 

Now testing with tcambin at 640x480 5Hz and seems to be working (I'll keep it running). I couldn't test on tcambin in full resolution because of the problem in the last comment. 

¿Any ideas on what could be the problem? ¿heating? ¿the cable?

ivanprado, 2017-08-02:
I can confirm it also fails with tcambin at higher resolutions:

```
gst-launch-1.0 --gst-debug=tcam*:5 --gst-debug-no-color tcambin ! video/x-raw,width=2448,height=2048,framerate=35/1 ! videoconvert ! autovideosink:

....
0:10:44.284161690 25602       0xbcb230 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1461:gst_tcam_src_create: Timestamp(ns): 88507627320000
0:10:44.284192510 25602       0xbcb230 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1475:gst_tcam_src_create:<tcambin-source> Pushing buffer...
0:10:54.159413443 25602 0x7f2350001320 ERROR                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1201:gst_tcam_src_device_lost_callback: Received lost device notification. Stopping stream.
0:10:54.162736418 25602 0x7f2350001320 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1315:gst_tcam_src_stop:<tcambin-source> Stopped acquisition
Got EOS from element "pipeline0".
Execution ended after 0:10:52.025104766
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
0:10:54.163394257 25602       0xb36360 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1315:gst_tcam_src_stop:<tcambin-source> Stopped acquisition
Setting pipeline to NULL ...
Freeing pipeline ...
```

After the failure, the camera gets into mentioned bad state, so re-executing the former pipeline gets:

```
...
0:00:02.135176157 26186       0xc6f230 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1084:gst_tcam_src_set_caps: In tcam_set_caps
0:00:02.135181442 26186       0xc6f230 INFO                 tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1096:gst_tcam_src_set_caps: Requested caps = video/x-bayer, format=(string)rggb, width=(int)2448, height=(int)2048, framerate=(fraction)35/1
0:00:02.140635748 26186       0xc6f230 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1141:gst_tcam_src_set_caps:<tcambin-source> Frame rate = 35 Hz
0:00:02.140677535 26186       0xc6f230 INFO                 tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1176:gst_tcam_src_set_caps: Start acquisition
0:00:02.152996569 26186       0xc6f230 INFO                 tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1191:gst_tcam_src_set_caps: Successfully set caps to: video/x-bayer, format=(string)rggb, width=(int)2448, height=(int)2048, framerate=(fraction)35/1
0:00:12.174021326 26186 0x7fd618002560 ERROR                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1201:gst_tcam_src_device_lost_callback: Received lost device notification. Stopping stream.
0:00:12.178729135 26186 0x7fd618002560 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1315:gst_tcam_src_stop:<tcambin-source> Stopped acquisition
Got EOS from element "pipeline0".
Execution ended after 0:00:10.044668091
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
0:00:12.179812511 26186       0xbda360 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1315:gst_tcam_src_stop:<tcambin-source> Stopped acquisition
Setting pipeline to NULL ...
Freeing pipeline ...
```

TIS-Stefan, 2017-08-02:
I asked per email for a sample project for us, so we can check the project on our own and see, what happens. 

The firmware in the camera 459. We have no newer firmware. 

Stefan

ivanprado, 2017-08-02:
Acording to Stefan Geissler, the camera already has the latest firmware version.

I also got the error with 640x480 30 frames:
```
gst-launch-1.0 --gst-debug=tcam*:5 --gst-debug-no-color tcambin ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! autovideosink
```

ivanprado, 2017-08-03:
I'm doing tests with a brand new USB3 cable. Things seems to improve, but still the problem arises. 

I have run `gst-launch-1.0 --gst-debug=tcam*:5 --gst-debug-no-color tcambin ! video/x-raw,width=2448,height=2048,framerate=35/1 ! videoconvert ! autovideosink` and it stopped working after 28 minutes, as can be seen here:

```
...
0:28:15.826372814 13291      0x255e230 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1475:gst_tcam_src_create:<tcambin-source> Pushing buffer...
0:28:15.997617439 13291      0x255e230 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1461:gst_tcam_src_create: Timestamp(ns): 10535911249000
0:28:15.997657082 13291      0x255e230 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1475:gst_tcam_src_create:<tcambin-source> Pushing buffer...
0:28:16.168340090 13291      0x255e230 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1461:gst_tcam_src_create: Timestamp(ns): 10535996929000
0:28:16.168369202 13291      0x255e230 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1475:gst_tcam_src_create:<tcambin-source> Pushing buffer...
0:28:26.077027121 13291 0x7f0448001320 ERROR                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1201:gst_tcam_src_device_lost_callback: Received lost device notification. Stopping stream.
0:28:26.080660437 13291 0x7f0448001320 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1315:gst_tcam_src_stop:<tcambin-source> Stopped acquisition
Got EOS from element "pipeline0".
Execution ended after 0:28:23.947872179
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
0:28:26.081375067 13291      0x24c9360 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1315:gst_tcam_src_stop:<tcambin-source> Stopped acquisition
Setting pipeline to NULL ...
Freeing pipeline ...
```

If I try to execute the same just after, it simply didn't work. The message is the same `Received lost device notification. Stopping stream.`

```
...
0:00:02.682273468 15175      0x1174230 INFO                 tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1191:gst_tcam_src_set_caps: Successfully set caps to: video/x-bayer, format=(string)rggb, width=(int)2448, height=(int)2048, framerate=(fraction)35/1
0:00:12.704185404 15175 0x7f657c001d50 ERROR                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1201:gst_tcam_src_device_lost_callback: Received lost device notification. Stopping stream.
0:00:12.707339421 15175 0x7f657c001d50 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1315:gst_tcam_src_stop:<tcambin-source> Stopped acquisition
Got EOS from element "pipeline0".
Execution ended after 0:00:10.043004773
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
0:00:12.708089062 15175       0xfab0f0 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1315:gst_tcam_src_stop:<tcambin-source> Stopped acquisition
Setting pipeline to NULL ...
Freeing pipeline ...
```

Please, @TIS-Edgar any idea what is happening? 

Regards

TIS-Arne, 2017-08-09:
This looks like a USB communication issue. Does your host computer have multiple USB ports? If so, could you please try to connect the camera to a different USB port. 
Also please post the output of "dmesg" after the issue happens.

ivanprado, 2017-08-09:
@TIS-Arne dmesg after the issue happens:

```
[18614.712965] usb 4-6: new SuperSpeed USB device number 10 using xhci_hcd
[18614.729881] usb 4-6: LPM exit latency is zeroed, disabling LPM.
[18614.730656] usb 4-6: New USB device found, idVendor=199e, idProduct=9089
[18614.730658] usb 4-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[18614.730659] usb 4-6: Product: DFK 33UX264
[18614.730660] usb 4-6: Manufacturer: The Imaging Source Europe GmbH
[18614.730661] usb 4-6: SerialNumber: 28710063
[18614.731402] uvcvideo: Unknown video format 36314752-0000-0010-8000-00aa00389b71
[18614.731405] uvcvideo: Unknown video format 31313459-0000-0010-8000-00aa00389b71
[18614.731408] uvcvideo: Found UVC 1.10 device DFK 33UX264 (199e:9089)
[18615.399606] ------------[ cut here ]------------
[18615.399617] WARNING: CPU: 6 PID: 3511 at /build/linux-lts-xenial-ep3zLI/linux-lts-xenial-4.4.0/drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd3e/0x10c0 [videodev]()
[18615.399618] Unknown pixelformat 0x00000000
[18615.399619] Modules linked in: cdc_ether uas usbnet usb_storage cdc_acm mii ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_conntrack_netlink xfrm_user xfrm_algo iptable_nat nf_nat_ipv4 br_netfilter bridge stp llc aufs nvram drbg ansi_cprng ctr ccm pci_stub vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) nfnetlink_queue nfnetlink_log nfnetlink input_polldev cmac uvcvideo btusb btrtl videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core btbcm v4l2_common btintel videodev hid_multitouch media nvidia_uvm(POE) arc4 ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 xt_hl ip6t_rt nf_conntrack_ipv6 nf_defrag_ipv6 ipt_REJECT nf_reject_ipv4 nf_log_ipv4 nf_log_common xt_LOG xt_limit xt_tcpudp xt_addrtype nf_conntrack_ipv4 nf_defrag_ipv4 xt_conntrack ip6table_filter nvidia_modeset(POE) ip6_tables bnep rfcomm nf_conntrack_netbios_ns nf_conntrack_broadcast bluetooth nf_nat_ftp nf_nat nf_conntrack_ftp nvidia(POE) nf_conntrack iptable_filter ip_tables pn544_mei mei_phy x_tables pn544 hci nfc snd_hda_codec_hdmi intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp snd_hda_codec_realtek dell_laptop kvm_intel snd_hda_codec_generic dcdbas kvm snd_hda_intel snd_hda_codec irqbypass iwlmvm dell_smm_hwmon snd_hda_core crct10dif_pclmul snd_hwdep crc32_pclmul snd_pcm mac80211 ghash_clmulni_intel snd_seq_midi snd_seq_midi_event aesni_intel snd_rawmidi snd_seq aes_x86_64 lrw gf128mul glue_helper ablk_helper snd_seq_device cryptd input_leds joydev iwlwifi serio_raw snd_timer rtsx_pci_ms cfg80211 memstick snd lpc_ich mei_me mei ie31200_edac soundcore shpchp edac_core binfmt_misc acpi_als kfifo_buf industrialio mxm_wmi dell_wmi sparse_keymap int3403_thermal mac_hid processor_thermal_device int3402_thermal int3400_thermal dell_smo8800 dell_rbtn wmi int340x_thermal_zone acpi_thermal_rel intel_soc_dts_iosf parport_pc ppdev lp parport btrfs xor raid6_pq hid_apple hid_generic usbhid hid i915 rtsx_pci_sdmmc i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops psmouse drm ahci libahci rtsx_pci video fjes
[18615.399693] CPU: 6 PID: 3511 Comm: AudioThread Tainted: P        W  OE   4.4.0-83-generic #106~14.04.1-Ubuntu
[18615.399694] Hardware name: Dell Inc. Dell Precision M3800/Dell Precision M3800, BIOS A10 08/17/2015
[18615.399695]  0000000000000000 ffff8803ff04fbe8 ffffffff813ddefc ffff8803ff04fc30
[18615.399697]  ffffffffc13f4ab0 ffff8803ff04fc20 ffffffff8107ea36 ffff8803ff04fd98
[18615.399698]  0000000000000000 0000000000000000 ffff880229c60c00 ffffffffc13f7180
[18615.399700] Call Trace:
[18615.399704]  [<ffffffff813ddefc>] dump_stack+0x63/0x87
[18615.399708]  [<ffffffff8107ea36>] warn_slowpath_common+0x86/0xc0
[18615.399709]  [<ffffffff8107eabc>] warn_slowpath_fmt+0x4c/0x50
[18615.399713]  [<ffffffffc146a147>] ? uvc_ioctl_enum_fmt+0xb7/0x140 [uvcvideo]
[18615.399718]  [<ffffffffc13de80e>] v4l_enum_fmt+0xd3e/0x10c0 [videodev]
[18615.399721]  [<ffffffffc13dedea>] __video_do_ioctl+0x25a/0x2e0 [videodev]
[18615.399723]  [<ffffffff81220eb4>] ? mntput+0x24/0x40
[18615.399725]  [<ffffffff8120ef89>] ? path_openat+0x2e9/0x12d0
[18615.399729]  [<ffffffffc13e13fb>] video_usercopy+0x21b/0x590 [videodev]
[18615.399732]  [<ffffffffc13deb90>] ? v4l_enum_fmt+0x10c0/0x10c0 [videodev]
[18615.399735]  [<ffffffffc13e1785>] video_ioctl2+0x15/0x20 [videodev]
[18615.399738]  [<ffffffffc13da644>] v4l2_ioctl+0xa4/0xc0 [videodev]
[18615.399741]  [<ffffffff812143bd>] do_vfs_ioctl+0x2dd/0x4c0
[18615.399742]  [<ffffffff812103d3>] ? putname+0x53/0x60
[18615.399744]  [<ffffffff81200548>] ? do_sys_open+0x1b8/0x270
[18615.399745]  [<ffffffff81214619>] SyS_ioctl+0x79/0x90
[18615.399748]  [<ffffffff8180c876>] entry_SYSCALL_64_fastpath+0x16/0x75
[18615.399749] ---[ end trace 6bcd2dfb7f344bac ]---
[18615.399924] ------------[ cut here ]------------
[18615.399929] WARNING: CPU: 6 PID: 3511 at /build/linux-lts-xenial-ep3zLI/linux-lts-xenial-4.4.0/drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd3e/0x10c0 [videodev]()
[18615.399930] Unknown pixelformat 0x00000000
[18615.399931] Modules linked in: cdc_ether uas usbnet usb_storage cdc_acm mii ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_conntrack_netlink xfrm_user xfrm_algo iptable_nat nf_nat_ipv4 br_netfilter bridge stp llc aufs nvram drbg ansi_cprng ctr ccm pci_stub vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) nfnetlink_queue nfnetlink_log nfnetlink input_polldev cmac uvcvideo btusb btrtl videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core btbcm v4l2_common btintel v [... cut ...]
[20654.841772] uvcvideo: Non-zero status (-71) in video completion handler.
[20654.845270] uvcvideo: Non-zero status (-71) in video completion handler.
```

I'll try in a different port and I'll let you know. 

I have also discovered the following:
* It seems that my pipeline is only able to give me 15 fps even if I set camera to give me 35 (so there are buffers dropping at the pipeline). 
* I have noticed that I don't get the problem if I set fps to 15 in the pipeline. So probably the issue is related with the situation where there are buffers dropping.

ivanprado, 2017-08-09:
@TIS-Arne I can confirm that the same problem arises in a different USB port.

TIS-Arne, 2017-08-10:
Thank you very much for the log files. 

There seems to be an issue with the USB communication. First of all you should be able to receive the full 35 FPS, as long as your host machine has enough performance to process this amount of data. From the dmesg output it looks like you are using a Core i7 machine which should suffice. I got stable 35 FPS on a Core i3 machine with the following launch line: 

gst-launch-1.0 tcambin serial=48610630 ! video/x-raw,width=2448,height=2048,framerate=35/1 ! queue ! videoconvert ! fpsdisplaysink video-sink=xvimagesink

The lines containing "uvcvideo: Non-zero status (-71) in video completion handler." indicate a low level hardware issue. It is difficult to pinpoint the exact cause but here are some things you should try:

- It is possible that your laptop does not provide enough power on the USB ports. This could explain why the camera runs stable on lower resolutions and frame rates since the device draws less power in this situation. You could connect the camera to a sufficiently powered USB 3.0 hub to see if it runs stable then. 

- Ubuntu 14.04 is quite an old distribution. It is possible that newer Linux kernel versions have specific fixes for issues with the USB interface used on your laptop. It would be worth testing the device with a more recent Linux version. If a permanent installation on your machine is not possible, an installation on a USB drive might work.

ivanprado, 2017-08-10:
Thanks @TIS-Arne 

- Although I'm using Ubuntu 14.04, mi kernel version is not so old: `4.4.0-83-generic`. Don't you think is enough recent version?
- Regarding fps, I don't know why when using fpsdisplaysink `fps` and `reendered` are always 0 and only `dropped` and `drop_rate` is shown. Any idea? Other way of measuring fps?
- Drop rate is **7** when running your command `gst-launch-1.0 tcambin ! video/x-raw,width=2448,height=2048,framerate=35/1 ! queue ! videoconvert ! fpsdisplaysink video-sink=xvimagesink` , so still buffers are dropped. My computer has a core i7 as you told.
- I don't have a powered USB3 hub, so I can't test that by the moment.

TIS-Arne, 2017-08-10:
The kernel version does not sound too old. Probably it is not worth trying a newer one. 

To test the frame rate, you could use this script:

https://www.dropbox.com/s/8x29vrx536jt0xf/camera.py?dl=0

It will prompt you for a camera, video format and frame rate. It should then present a live video image. You could enter "f" on the console and it will print the currently achieved frame rate.

For the instabilities, my best bet is still an USB power issue since this would explain why the camera only fails at higher frame rates and resolutions. Maybe you can borrow a USB hub from someone for testing?

ivanprado, 2017-08-10:
Frame rate reported by this tool is 34.879171. So it seems that fps is good as reported by this tool. 

I have also noticed that the problem also appears when using this tool.

TIS-Arne, 2017-08-10:
A web search with your notebook brand and model turned up some USB issues which look similar to what you are encountering (for example here: http://forum.notebookreview.com/threads/xps-9530-usb-connects-disconnects.760728/page-2#post-9809393). Unfortunately there does not seem to be a definite solution to the problem. You could try the solutions discussed in this thread (ie. change some USB settings in the BIOS and/or use ridiculously short cables) but it looks like there really might be an issue with the laptop hardware. 
My best bet would still be to use a powered USB hub but even this is not guaranteed to work. 
I am sorry I could not come up with a better solution for this issue :/

ivanprado, 2017-08-28:
@TIS-Arne I have tested with a powered USB3 hub with no luck. But then I have also tested in **another laptop** with same OS and kernel version and **It seems to work** :-D. So by now (crossing fingers) I conclude that you were right and there is particular strange problem with USB on my laptop. Thanks for the support. 

Anyway I'll run more exhaustive tests as I want to be completely sure. If issue returns on that laptop I'll tell you here. 

Regards, 
Iván

closed 2017-08-28 by TIS-Arne