TheImagingSource/tiscamera #106: Debayering bug on tcamsrc

ivanprado, 2017-08-03:
As you can see with [that images](https://gist.github.com/ivanprado/6f9ccee664f8f18d74862ad3839647c7) tcamsrc is not debayering properly the images. 

Getting images in rggb format and debayering with rgb2bayer element seems to provide better results than just request images in BGR format

TIS-Arne, 2017-08-09:
Hello ivanprado, 

the "tcamsrc" is never debayering any issues. Only the "tcambin" element embeds the bayer2rgb filter if needed. 

With the launch line you provided, the debayering of the images takes place on the camera itself. Due to resource restrictions on the camera hardware, the result is not as good as if the debayering is performed on the host machine. 

This means that if you need the best image quality, you need to use the bayer2rgb element on the host machine but if you need to reduce CPU consumption on the host machine, you could use the cameras on-board debayering.

ivanprado, 2017-08-09:
@TIS-Arne understood, thanks.

closed 2017-08-10 by TIS-Arne