TheImagingSource/tiscamera #107: Interfacing my GIGE mono chrome camera with OpenCV

louispetitjean, 2017-08-10:
Hi, 
I just bought a GigE camera, and i'm currently trying to access it with an openCV capture function. I first fixed the camera's ip address and now i'm kind of stuck, cause the camera is unreachable. 
I know that it's possible to access a GigE camera thanks to this process, it works for exemple with a PtGrey GigE Camera. 
So do you have any clue why it's not working ? Or do you provide some facilities between opencv and your camera ? 
Best regards, Louis

TIS-Arne, 2017-08-10:
@louispetitjean I am not sure what you mean with "unreachable". If the camera is not listed with "tcam-ctrl -l" then please try to start the "gige-daemon" first:

```
sudo systemctl start gige-daemon
tcam-ctrl -l
```

If the camera still can not be found by tcam-ctrl, then try the following command:

```
tcam-gigetool list
```

If this correctly detect the camera, please check the IP settings of the camera. 

If you need an example on how to interact with opencv, please let me know which programming language you are using.

louispetitjean, 2017-08-10:
Hi, 
Sorry if i wasn't clear ! I can access to the camera with the aravis arv-viewer program, but i can't compile the tiscamera example. 
I'm using C++ i would be really glad if you could provide me some examples !

louispetitjean, 2017-08-28:
Hi again, 
Last time you said you had an example using C++ and openCV, is it possible to see this example ?

TIS-Arne, 2017-08-30:
Hi, 
there is no ready made sample. I am currently preparing one and will post it on this site tomorrow.

louispetitjean, 2017-09-06:
Hi, 
Is the sample ready ?

TIS-Stefan, 2018-04-12:
Sorry for the long delay of my answer. 

In the attachment is a C++ sample for GStreamer, our tiscamera modules and how to convert a GStBuffer to cv::Mat.
The interesting part is in main.cpp.

I hope, this helps. 
Stefan
[tcambin-autofocus.zip](https://github.com/TheImagingSource/tiscamera/files/1903143/tcambin-autofocus.zip)

closed 2018-04-12 by TIS-Stefan