TheImagingSource/tiscamera #108: Framerate affecting software trigger delay

ivanprado, 2017-08-28:
I'm using camera for taking photos at particular moments using a GStreamer pipeline. The pipeline is:

`tcamsrc serial="28710063" name=source ! capsfilter caps="video/x-bayer, format=(string){rggb}, width=(int)2448, height=(int)2048, framerate=(fraction)35/1" !  bayer2rgb !  videoconvert ! capsfilter caps="video/x-raw,format=BGR" ! identity name=identity ! fakesink sync=false
`
And I'm capturing the frames on the identity element.

I have noticed the following behavior. The selected framerate on caps is affecting to when a buffer is received. So for example, if I set a framerate of 1 frame per second, I cannot grab more than one image per second even if I have triggered more than one image in that second.

Because of that I have set framerate to the maximum possible: 35 for the highest resolution.

Is that the expected behavior?

It also raises another question. Could I expect to have the same delay between photos where trigger and they where shooted, or that depends on the framerate, so what I receive is just the next frame on the video stream at that framerate?

In other words, let's imagine that framerate is 1 per second and more than one second took since last photo was taken. If I trigger a photo will it come as soon as possible (in a few milliseconds) or maybe it can take longer as camera is following a strict internal 1 second timer?

TIS-Arne, 2017-08-28:
The actual implementation is dependend on the camera you are using but on most cameras (inlcuding the DFK 33UX264), the frame rate setting directly affects the sensor read out speed. If you set a frame rate of 1 frames per second, it takes about one second to read out the sensor and thus you can not trigger another image during that period. This also means that for rolling shutter sensors, the rolling shutter effect is increased with lower frame rates. 

To be exact, the frame readout time in triggered mode is "framerate + exposure time", so if you set an exposure time of 1 second and a frame rate of 1 second you can only trigger a new image each 2 seconds.

In triggered mode, the camera pauses the sensor until a trigger event is registered. The exposure of the video frame starts immediately after receiving the trigger signal, regardless of the frame rate.

ivanprado, 2017-08-28:
Ok, understood, thanks.

closed 2017-08-28 by TIS-Arne