TheImagingSource/tiscamera #109: Erroneus  frame rate sometimes

ivanprado, 2017-08-28:
Hi, 

I'm experiencing bad frame rate some times. I want 35 FPS, but sometime I just get 5 FPS for some strange reason. Why? Do you need more logs or traces to look into the problem? Look at the following trace from your tool camera.py to see the extrange behaviour:

```
Available video formats:
1: rggb (640x480)
2: rggb (1920x1080)
3: rggb (2048x2048)
4: rggb (2448x2048)
5: YUY2 (640x480)
6: YUY2 (1920x1080)
7: YUY2 (2048x2048)
8: YUY2 (2448x2048)
9: BGR (640x480)
10: BGR (1920x1080)
11: BGR (2048x2048)
12: BGR (2448x2048)
Select format (0 exits) : 4
Available frame rates:
1 : 35/1
2 : 30/1
3 : 15/1
4 : 5/1
5 : 1/1
Select frame rate (0 exits) : 1
Caps set for capture: video/x-raw, format=(string)RGBx, width=(int)2448, height=(int)2048, framerate=(fraction)35/1
cmd : f
Current FPS: 5.177840
cmd : f
Current FPS: 5.177840
cmd : f
Current FPS: 5.177840
cmd : f
Current FPS: 5.177840
cmd : f
Current FPS: 5.177840
cmd : f
Current FPS: 5.177840
cmd : f
Current FPS: 5.177840
cmd : f
Current FPS: 5.177840
cmd : f
Current FPS: 5.177840
cmd : f
Current FPS: 5.177840
cmd : f
Current FPS: 5.177840
cmd : f
Current FPS: 5.177840
cmd : f
Current FPS: 5.177840
cmd : f
Current FPS: 5.177840
cmd : F
Available video formats:
1: rggb (640x480)
2: rggb (1920x1080)
3: rggb (2048x2048)
4: rggb (2448x2048)
5: YUY2 (640x480)
6: YUY2 (1920x1080)
7: YUY2 (2048x2048)
8: YUY2 (2448x2048)
9: BGR (640x480)
10: BGR (1920x1080)
11: BGR (2048x2048)
12: BGR (2448x2048)
Select format (0 exits) : 4
Available frame rates:
1 : 35/1
2 : 30/1
3 : 15/1
4 : 5/1
5 : 1/1
Select frame rate (0 exits) : 1
Caps set for capture: video/x-raw, format=(string)RGBx, width=(int)2448, height=(int)2048, framerate=(fraction)35/1
cmd : f
Current FPS: 35.209662
cmd : f
Current FPS: 34.820390
cmd : F
Available video formats:
1: rggb (640x480)
2: rggb (1920x1080)
3: rggb (2048x2048)
4: rggb (2448x2048)
5: YUY2 (640x480)
6: YUY2 (1920x1080)
7: YUY2 (2048x2048)
8: YUY2 (2448x2048)
9: BGR (640x480)
10: BGR (1920x1080)
11: BGR (2048x2048)
12: BGR (2448x2048)
Select format (0 exits) : 4
Available frame rates:
1 : 35/1
2 : 30/1
3 : 15/1
4 : 5/1
5 : 1/1
Select frame rate (0 exits) : 1
Caps set for capture: video/x-raw, format=(string)RGBx, width=(int)2448, height=(int)2048, framerate=(fraction)35/1
cmd : f
Current FPS: 9.055457
cmd : f
Current FPS: 34.865206
cmd : f
Current FPS: 35.108679
cmd : f
Current FPS: 34.658953
cmd : f
Current FPS: 35.034114
cmd : f
Current FPS: 34.643074
cmd : f
Current FPS: 34.850007
cmd : F
Available video formats:
1: rggb (640x480)
2: rggb (1920x1080)
3: rggb (2048x2048)
4: rggb (2448x2048)
5: YUY2 (640x480)
6: YUY2 (1920x1080)
7: YUY2 (2048x2048)
8: YUY2 (2448x2048)
9: BGR (640x480)
10: BGR (1920x1080)
11: BGR (2048x2048)
12: BGR (2448x2048)
Select format (0 exits) : 4
Available frame rates:
1 : 35/1
2 : 30/1
3 : 15/1
4 : 5/1
5 : 1/1
Select frame rate (0 exits) : 1
Caps set for capture: video/x-raw, format=(string)RGBx, width=(int)2448, height=(int)2048, framerate=(fraction)35/1
cmd : F
Available video formats:
1: rggb (640x480)
2: rggb (1920x1080)
3: rggb (2048x2048)
4: rggb (2448x2048)
5: YUY2 (640x480)
6: YUY2 (1920x1080)
7: YUY2 (2048x2048)
8: YUY2 (2448x2048)
9: BGR (640x480)
10: BGR (1920x1080)
11: BGR (2048x2048)
12: BGR (2448x2048)
Select format (0 exits) : 4
Available frame rates:
1 : 35/1
2 : 30/1
3 : 15/1
4 : 5/1
5 : 1/1
Select frame rate (0 exits) : 1
Caps set for capture: video/x-raw, format=(string)RGBx, width=(int)2448, height=(int)2048, framerate=(fraction)35/1
cmd : f
Current FPS: 4.275281
cmd : f
Current FPS: 4.017933
cmd : f
Current FPS: 3.835013
cmd : f
Current FPS: 6.971494
cmd : f
Current FPS: 6.970205
cmd : f
Current FPS: 6.969016
cmd : f
Current FPS: 6.964576
cmd : f
Current FPS: 6.972172
```

ivanprado, 2017-08-28:
Here you have an example of log traces when that happens just using gst-launch-1.0. As you can see by `Pushing buffer` lines timestamp, only 4 frames per second are served. When that happens, my process is affected as photos cannot be taken at enough frequency.

```
gst-launch-1.0 --gst-debug=tcam*:5 --gst-debug-no-color tcamsrc ! video/x-bayer,width=2448,height=2048,framerate=35/1 ! bayer2rgb ! videoconvert ! autovideosink
0:00:00.055473330  6127       0x88c460 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1209:gst_tcam_src_init_camera:<tcamsrc0> Initializing device.
0:00:00.058669433  6127       0x88c460 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1219:gst_tcam_src_init_camera:<tcamsrc0> Found 1 devices.
0:00:00.058711405  6127       0x88c460 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1227:gst_tcam_src_init_camera:<tcamsrc0> No serial given. Opening first available device.
0:00:00.151654893  6127       0x88c460 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:675:gst_tcam_src_get_all_camera_caps: Found 5 pixel formats
0:00:00.151714789  6127       0x88c460 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:691:gst_tcam_src_get_all_camera_caps: Found 'video/x-bayer, format=(string)rggb' pixel format string
0:00:00.151739780  6127       0x88c460 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:697:gst_tcam_src_get_all_camera_caps: Found 4 resolutions
0:00:00.151997867  6127       0x88c460 ERROR                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:685:gst_tcam_src_get_all_camera_caps: Format has empty fourcc. Ignoring.
0:00:00.152025953  6127       0x88c460 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:691:gst_tcam_src_get_all_camera_caps: Found 'video/x-raw, format=(string)YUY2' pixel format string
0:00:00.152076471  6127       0x88c460 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:697:gst_tcam_src_get_all_camera_caps: Found 4 resolutions
0:00:00.152265867  6127       0x88c460 ERROR                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:685:gst_tcam_src_get_all_camera_caps: Format has empty fourcc. Ignoring.
0:00:00.152285424  6127       0x88c460 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:691:gst_tcam_src_get_all_camera_caps: Found 'video/x-raw, format=(string)BGR' pixel format string
0:00:00.152314647  6127       0x88c460 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:697:gst_tcam_src_get_all_camera_caps: Found 4 resolutions
0:00:00.152590986  6127       0x88c460 INFO                 tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:833:gst_tcam_src_get_all_camera_caps: video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction){ 300/1, 5000000/20833, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-bayer, format=(string)rggb, width=(int)1920, height=(int)1080, framerate=(fraction){ 90/1, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-bayer, format=(string)rggb, width=(int)2048, height=(int)2048, framerate=(fraction){ 45/1, 30/1, 15/1, 5/1, 1/1 }; video/x-bayer, format=(string)rggb, width=(int)2448, height=(int)2048, framerate=(fraction){ 35/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, framerate=(fraction){ 600/1, 480/1, 5000000/20833, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, framerate=(fraction){ 90/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)YUY2, width=(int)2048, height=(int)2048, framerate=(fraction){ 40/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)YUY2, width=(int)2448, height=(int)2048, framerate=(fraction){ 35/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)BGR, width=(int)640, height=(int)480, framerate=(fraction){ 400/1, 5000000/20833, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)BGR, width=(int)1920, height=(int)1080, framerate=(fraction){ 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)BGR, width=(int)2048, height=(int)2048, framerate=(fraction){ 25/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)BGR, width=(int)2448, height=(int)2048, framerate=(fraction){ 25/1, 15/1, 5/1, 1/1 }; ANY
0:00:00.152863633  6127       0x88c460 INFO                 tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1063:gst_tcam_src_get_caps: Available caps = video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction){ 300/1, 5000000/20833, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-bayer, format=(string)rggb, width=(int)1920, height=(int)1080, framerate=(fraction){ 90/1, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-bayer, format=(string)rggb, width=(int)2048, height=(int)2048, framerate=(fraction){ 45/1, 30/1, 15/1, 5/1, 1/1 }; video/x-bayer, format=(string)rggb, width=(int)2448, height=(int)2048, framerate=(fraction){ 35/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, framerate=(fraction){ 600/1, 480/1, 5000000/20833, 120/1, 60/1, 30/1, 15/1, 5/1, 1/1 }; video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, framerate=(fraction){ 90/1, 60/1, 30/1,  [... cut ...]
0:00:04.915143080  6127       0x8c7540 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1475:gst_tcam_src_create:<tcamsrc0> Pushing buffer...
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:04.836595537
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
0:00:05.014918088  6127       0x88c460 DEBUG                tcamsrc /home/ivan/sw/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1315:gst_tcam_src_stop:<tcamsrc0> Stopped acquisition
Setting pipeline to NULL ...
Freeing pipeline ...
```

TIS-Arne, 2017-08-30:
@ivanprado I do not have an idea what might cause this. I will try to reproduce this issue here and get back to you shortly.

ivanprado, 2017-09-04:
Hi @TIS-Arne , 

Where you able to reproduce it?

Regards, 
Iván

TIS-Arne, 2017-09-04:
@ivanprado I can reproduce the issue but was not able to find the cause just yet. I am still looking into it.

TIS-Arne, 2017-09-05:
@ivanprado We have a temporary workaround for the frame rate issue. Setting "do-timestamp" property on the camera source element leads to a full frame rate. To do this with the camera.py script, insert the line:
```
        self.source.set_property("do-timestamp", True)
```
at the end of the "start" method after line 142. Could you try that?

ivanprado, 2017-09-05:
It seems to work, thanks!. What does `do-timestamp` property means? Any secondary effect of using that?

TIS-Arne, 2017-09-05:
According to the docs, the `do-timestamp` property does
```
Configure src to automatically timestamp outgoing buffers based on the current running_time of the pipeline. This property is mostly useful for live sources.
```
What this means is that with this property set, the timestamps are generated in software on GStreamer level, instead of coming from the driver which would be more accurate. There should not be any side effect unless you need highly accurate timestamps in your application. Our developer is working on this issue so that in a future version of the software, this property is no longer needed.

ivanprado, 2017-09-05:
Nice, thanks @TIS-Arne . 

This workaround is enough for my current case, so thanks. 

Regards, 
Iván

closed 2018-04-10 by TIS-Stefan