TheImagingSource/tiscamera #110: How to use Digital trigger in Linux with ROS package likes uvc_camera?

AbnerCSZ, 2017-08-30:
I bougtht the DFK22BUC03. I can run and get picture with Digital trigger in Windows. 
But I can't do the same things in Linux(Ubuntu). 
I have to use the Digital trigger by the ROS package "uvc_camera". There is the link of the package:
http://wiki.ros.org/uvc_camera
I believe your camera can be used by this package in ROS(Robot Operating System). Because I found the source codes about UVC in your github: 
https://github.com/TheImagingSource/tiscamera/wiki/UVC-Extension-Units
Unfortunately, although I can run the" gst-launch-1.0 tcambin ! videoconvert ! autovideosink" in the README.md and get pictures normally, I can't run the "sudo uvcdynctrl -i tisEUVC.xml -d /dev/video0" in wiki of UVC normally. 
In general, I have to use the ROS package, maybe uvc_camera or some package from your campany. Do you have any ROS package to trigger and save images?
Thanks very much! I need your help.
Tips: The Chinese technical supports are familiar with Windows, But can't give me help about Linux. 
Looking forward to your reply

AbnerCSZ, 2017-08-31:
This is my launch:
`<launch>
  <group ns="camera">
    <node pkg="libuvc_camera" type="camera_node" name="mycam">
      
      <param name="vendor" value="0x199e"/>
      <param name="product" value="0x8302"/>
      <param name="serial" value="30714360"/>
      
      <param name="index" value="0"/>

      
      <param name="width" value="640"/>
      <param name="height" value="480"/>
      
      <param name="video_mode" value="uncompressed"/> 
      <param name="frame_rate" value="15/1"/>

      <param name="timestamp_method" value="start"/> 
      

      <param name="auto_exposure" value="3"/> 
      <param name="auto_white_balance" value="true"/>
    </node>
  </group>
</launch>`

and the result is 

> NODES
  /camera/
    mycam (libuvc_camera/camera_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[camera/mycam-1]: started with pid [3510]
unsupported descriptor subtype VS_COLORFORMAT
uvc_get_stream_ctrl_format_size: Invalid mode (-51)
[ERROR] [1504147640.013810307]: check video_mode/width/height/frame_rate are available
DEVICE CONFIGURATION (0000:0000/[none]) ---
Status: idle
VideoControl:
	bcdUVC: 0x6972
END DEVICE CONFIGURATION
[camera/mycam-1] process has died [pid 3510, exit code 255, cmd /home/abner/catkin_ws/devel/lib/libuvc_camera/camera_node __name:=mycam __log:=/home/abner/.ros/log/63508676-8df6-11e7-aee5-f81654cf1656/camera-mycam-1.log].
log file: /home/abner/.ros/log/63508676-8df6-11e7-aee5-f81654cf1656/camera-mycam-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete

TIS-Arne, 2017-08-31:
I can not tell specifics about ROS right now since the website of this project is unresponsive.
Triggering should work with any application supporting the "video4linux" interface. You need to install the extension unit XML files as per the instructions you posted above in order to enable the trigger functionality in the usual way. 
However for the DFK22BUC03 there is another workaround if you can not access the extension unit: The camera has a standard UVC property called "privacy mode". If you enable this property, the camera also switches to the triggered mode.

AbnerCSZ, 2017-08-31:
This is my result when I run "   v4l2-ctl --list-formats-ext"  
```
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'GREY'
	Name        : 8-bit Greyscale
		Size: Discrete 744x480
			Interval: Discrete 0.013s (76.046 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.011s (87.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.006s (161.290 fps)
			Interval: Discrete 0.008s (120.000 fps)
			Interval: Discrete 0.013s (80.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'GBRG'
	Name        : 8-bit Bayer GBGB/RGRG
		Size: Discrete 744x480
			Interval: Discrete 0.013s (76.046 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.011s (87.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.006s (161.290 fps)
			Interval: Discrete 0.008s (120.000 fps)
			Interval: Discrete 0.013s (80.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)

```
Pixel Format is GREY and GBRG . But the option of uvc_cam are: uncompressed or yuyv/nv12/jpeg. So I can't use the uvc_cam for this camera?

TIS-Arne, 2017-08-31:
Likely not since the software needs to do the color processing (de-bayering) of the raw video data provided by the camera. Does the software also support the video4linux interface?

AbnerCSZ, 2017-08-31:
I believe it support the v4l. There is its Overview:

> This package provides a ROS interface for digital cameras meeting the USB Video Class standard (UVC) using libuvc. Most webcams are UVC-compliant. 

What's the "You need to install the extension unit XML files as per the instructions you posted above in order to enable the trigger functionality in the usual way." means?

```
when I follow the wiki "UVC Extension Units".This is my result:
[bner@abner-XPS-8700:~/tiscamera/data/uvc-extensions$ sudo uvcdynctrl -i tisEUVC.xml -d /dev/video0
[libwebcam] Unknown V4L2 user control ID encountered: 0x00980930 (V4L2_CID_USER_BASE + 48)
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E208
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E208, name = 'Trigger'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E209
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E209, name = 'Software Trigger'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E920
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E920, name = 'ExtIO'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E927
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E927, name = 'X Offset'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E928
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E928, name = 'Y Offset'
Importing dynamic controls from file tisEUVC.xml.
tisEUVC.xml: error: video0: unable to map 'ExtIO' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
tisEUVC.xml: error: video0: unable to map 'GainR' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video0: unable to map 'GainG' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video0: unable to map 'GainB' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video0: unable to map 'Trigger' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
tisEUVC.xml: error: video0: unable to map 'Trigger Global Reset Shutter' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
tisEUVC.xml: error: video0: unable to map 'Software Trigger' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
tisEUVC.xml: error: video0: unable to map 'Binning' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video0: unable to map 'Skipping' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video0: unable to map 'X Offset' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
tisEUVC.xml: error: video0: unable to map 'Y Offset' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
](url)
```

AbnerCSZ, 2017-08-31:
Can I use the OpenCV to Digital trigger(not software trigger)?
[http://docs.opencv.org/trunk/dd/d01/group__videoio__c.html](url)

AbnerCSZ, 2017-08-31:
Can I use the OpenCV to Digital trigger(not software trigger)?
http://docs.opencv.org/trunk/dd/de7/group__videoio.html

TIS-Arne, 2017-08-31:
"libuvc" does not use the v4l2 interface, it seems to communicate directly with the camera. I am sorry I can not tell much about the issue as long as the ROS webserver is unresponsive.

TIS-Arne, 2017-08-31:
Update: It takes a long time for the server to come up with a page but I finally found a list of ROS modules. You could try this module for v4l access: http://www.ros.org/browse/details.php?distro=lunar&name=usb_cam

AbnerCSZ, 2017-08-31:
So what is the "UVC Extension Units" means? How can I install the XML files?Should I have to modify the file?
And how about the uvc_camera of the ROS?

AbnerCSZ, 2017-08-31:
Yeah, I can use the usb_cam, But the image is bad(With many bad cube).

TIS-Arne, 2017-08-31:
I just saw there is also a driver for gstreamer cameras: https://github.com/ros-drivers/gscam

With this driver you should be able to use our SDK with the launch line you mentioned.

AbnerCSZ, 2017-08-31:
I try the gscam and found this error:

```
NODES
  /v4l/
    gscam_driver_v4l (gscam/gscam)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[v4l/gscam_driver_v4l-1]: started with pid [22592]
[ INFO] [1504168239.070819551]: Using gstreamer config from rosparam: "v4l2src device=/dev/video0 ! video/x-raw,format=RGBx,framerate=30/1 ! ffmpegcolorspace"
[ INFO] [1504168239.074545623]: camera calibration URL: package://gscam/examples/uncalibrated_parameters.ini
[ INFO] [1504168239.075751855]: Loaded camera calibration from package://gscam/examples/uncalibrated_parameters.ini
**[FATAL] [1504168239.077082500]: 无组件“v4l2src”
[FATAL] [1504168239.077101513]: Failed to initialize gscam stream!**
[v4l/gscam_driver_v4l-1] process has finished cleanly
log file: /home/abner/.ros/log/f1cc3450-8e09-11e7-aee5-f81654cf1656/v4l-gscam_driver_v4l-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

But I think I have the "v4l2src", because I can run the" gst-launch-1.0 v4l2src ! xvimagesink" and get picture normally.

AbnerCSZ, 2017-08-31:
I think the problem is:

```
abner@abner-XPS-8700:~/catkin_ws$ rosrun gscam gscam
[FATAL] [1504168469.591123489]: Problem getting GSCAM_CONFIG environment variable and 'gscam_config' rosparam is not set. This is needed to set up a gstreamer pipeline.
[FATAL] [1504168469.591161250]: Failed to configure gscam!

```
**So how can i set the 'gscam_config' ?**

TIS-Arne, 2017-08-31:
I will have a look at ROS and let you know if I find a working configuration.

AbnerCSZ, 2017-08-31:
Thank you very much . The ROS Wiki is normal now.

TIS-Arne, 2017-08-31:
@AbnerCSZ Could you tell me how I could quickly verify whether a camera is working with ROS after I installed the ROS packages?

AbnerCSZ, 2017-08-31:
If it goes well, it should be confirmed soon.

I tried to run gst-inspect-0.10 v4l2src,and the result is no element "v4l2src"
and I run gst-inspect-1.0 v4l2src,the result is 
```

actory Details:
  Rank                     primary (256)
  Long-name                Video (video4linux2) Source
  Klass                    Source/Video
  Description              Reads frames from a Video4Linux2 device
  Author                   Edgard Lima <edgard.lima@indt.org.br>, Stefan Kost <ensonic@users.sf.net>

Plugin Details:
  Name                     video4linux2
  Description              elements for Video 4 Linux
  Filename                 /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstvideo4linux2.so
  Version                  1.8.3
  License                  LGPL
  Source module            gst-plugins-good
  Source release date      2016-08-19
  Binary package           GStreamer Good Plugins (Ubuntu)
  Origin URL               https://launchpad.net/distros/ubuntu/+source/gst-plugins-good1.0

GObject
 +----GInitiallyUnowned
       +----GstObject
             +----GstElement
                   +----GstBaseSrc
                         +----GstPushSrc
                               +----GstV4l2Src
…………

```
So I guess the error reason is that the ROS can't judge the version of gst-launch.
Can you tell me the different of two versions. Should I delete the 0.1?

TIS-Arne, 2017-08-31:
@AbnerCSZ Could you please give me an example on how I need to use ROS to see if our cameras are working?

AbnerCSZ, 2017-08-31:
http://wiki.ros.org/gscam

https://github.com/ros-drivers/gscam

http://wiki.ros.org/usb_cam

AbnerCSZ, 2017-08-31:
How can i use the Digital I/O trigger by your sdk? I found the example of software trigger.(trigger-image.c)
So after set the "Trigger Mode"  be true, than **how can i set the callback function when camera is triggered by the Digital pulse?**

TIS-Arne, 2017-09-01:
I finally seem to get ROS running. It looks like I could get the camera to work with the following command line: 
```
export GSCAM_CONFIG="tcambin ! video/x-raw,format=RGBA,width=640,height=480 ! videoconvert"
rosrun gscam gscam                                                                         
[ INFO] [1504250608.535509499]: Using gstreamer config from env: "tcambin ! video/x-raw,format=RGBA,width=640,height=480 ! videoconvert"
[ INFO] [1504250608.542329159]: using default calibration URL
[ INFO] [1504250608.542382048]: camera calibration URL: file:///home/arne/.ros/camera_info/camera.yaml
[ INFO] [1504250608.542452586]: Unable to open camera calibration file [/home/arne/.ros/camera_info/camera.yaml]
[ WARN] [1504250608.542484811]: Camera calibration file /home/arne/.ros/camera_info/camera.yaml not found.
[ INFO] [1504250608.542507586]: Loaded camera calibration from 
[ WARN] [1504250608.543212381]: No camera frame_id set, using frame "/camera_frame".
[ INFO] [1504250608.669830842]: Time offset: 1504215659.508
[ INFO] [1504250609.447217524]: Publishing stream...
[ INFO] [1504250609.447398401]: Started stream.
```

I do not know how to set parameters on the camera using ros in order to enable the trigger functionality and how to display a live video to make sure the video is correct. If you know ROS, can you tell me an easy way to test these things?

AbnerCSZ, 2017-09-01:
I do the same thing..But the result is error.
ROS is only responsible for publishing the images. So can you tell me how to use your SDK in linux to trigger? You can ignore ROS.

TIS-Arne, 2017-09-01:
Which programming language? You could try the
```
examples/python/trigger_images.py
```
example for instance.

AbnerCSZ, 2017-09-01:
C++。
This example is triggered by software. I need use the PIF(Process Interface) to hardware trigger. I can use it in Windows, but i need Ubuntu support with C++.

TIS-Arne, 2017-09-01:
The software trigger of the demo is just so you do not need to connect hardware. There is no functional difference in the camera, you should be able to use the hardware trigger input as well.

AbnerCSZ, 2017-09-01:
But how can i modify the code to command the camera to do what i want to? How can i save the picture when the camera be triggered?

AbnerCSZ, 2017-09-02:
@TIS-Arne @TIS-Stefan

AbnerCSZ, 2017-09-02:
Now I can "rosrun gscam gscam" Successful . You can use "rviz" in ROS to make sure the video is correct.
[RVIZ](http://wiki.ros.org/rviz/UserGuide)
You can add the iamge and set the image topic be image_raw(you can run "rostopic list -v "in a new Terminal to see all topics running , and find the topic with type of "sensor_msg/Image" or "sensor_msg/compressedImage"). And then you can see the images.

Now, the only problem is trigger. The gscam is based on Gstreamer, so maybe it use the same Software library with your SDK. But I know nothing about Gstreamer. Can you tell me how to use the trigger method of your example(examples/c/trigger_images.c), to modify the example of the gscam(gscam/src/gscam.cpp and gscam_node.cpp).

**How to achieve the same thing(set trigger mode to be true) in gscam.cpp?In your example:**
`tcam_prop_set_tcam_property(TCAM_PROP(source), "Trigger Mode", &trigger_val);`
**In addition to let the photo display, how can I save the images?**

 Hope your reply

AbnerCSZ, 2017-09-04:
Is there any conflict between gstreamer-1.0 and -0.1?
Now I can run the gscam to capture the pictures. But when i run "gst-launch-1.0 tcambin ! videoconvert ! autovideosink" in your README.md, there is the result :
```
(gst-plugin-scanner:26158): GStreamer-WARNING **: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsteglglessink.so': /usr/lib/x86_64-linux-gnu/libgstegl-1.0.so.0: undefined symbol: eglDestroyImageKHR
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Caught SIGSEGV
Spinning.  Please run 'gdb gst-launch-1.0 26157' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:04.054932799
Setting pipeline to PAUSED ...
Setting pipeline to READY ...

```
and no image showed.

I'm sure that i have libgstegl-1.0.so.0 in /usr/lib/x86_64-linux-gnu/
@TIS-Arne @TIS-Stefan

AbnerCSZ, 2017-09-04:
When I use my cheap webcam to run `gst-launch-1.0 v4l2src device=/dev/video0 ! videoconvert ! autovideosink`
the result is good and I can get picture normally.
But when I change device to video1(imagingsource camera), `gst-launch-1.0 v4l2src device=/dev/video1 ! videoconvert ! autovideosink`
the result is 
```
xlab@xlab-ThinkPad-T460p:~$ gst-launch-1.0 v4l2src device=/dev/video1 ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Caught SIGSEGV
Spinning.  Please run 'gdb gst-launch-1.0 26309' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.

```
and I can't get any picture. **Why? Do the camera not support v4l2src? But Why I can ust v4l2src in gscam to get pictures from imagingsource camera?**

AbnerCSZ, 2017-09-07:
I solved the problem!
v4l2-ctl -d /dev/videoX -c privacy=1
means enable trigger mode. and rosrun gscam gscam, Can achieve what I want.
Thanks for your help. @TIS-Arne

closed 2017-09-07 by AbnerCSZ