TheImagingSource/tiscamera #113: RPI3 tcam-capture

cappelaere, 2017-09-22:
As soon as I select my device in menu, I get:

DEBUG:tcam_capture.TcamView:Created pipeline and set to READY
INFO:tcam_capture.TcamView:Setting state to PAUSED
Traceback (most recent call last):
  File "tcam-capture", line 226, in open_device
    self.play()
  File "tcam-capture", line 231, in play
    self.view.play(video_format)
  File "/home/pi/tiscamera/tools/tcam-capture/tcam_capture/TcamView.py", line 242, in play
    if self.pipeline.get_state(1000000).state == Gst.State.PLAYING:
AttributeError: 'tuple' object has no attribute 'state'

cappelaere, 2017-09-22:
I can comment that line out but not sure if it is a solution!

TIS-Arne, 2018-04-12:
This looks like gst-python overrides are not installed or the gstreamer / gst-python package is outdated. Re-open if this is still an issue on a current Raspbian.

closed 2018-04-12 by TIS-Arne