TheImagingSource/tiscamera #114: Can't live stream with camera DFK23UV024 on OpenSUSE Leap 42.3

mbless, 2017-10-05:
Dears,

I'm trying to use this camera on OpenSUSE for the first time and I'm having problem to make it work. This camera works perfect with the drivers and program distribuited for Windows. I've followed the installation steps listed in the following link https://github.com/TheImagingSource/tiscamera (README.md). The function ARAVIS was also included. After the instalation I try to execute the following line an error pops up:
[image]

So then I check which plug in I have available for Gstream:

[image]

So the plug-in are there and recognized by the program. If I look specifically for tcambin with gst-inspect I get that the plug-in could not be loaded:

[image]

I think that this problem is related with how the plug-ins are called but I don't have any clue to check this or making this available. 

Thanks you in advance for your help and I'm looking forward to your prompt replay.

Kind regards,

Maxi

TIS-Edgar, 2017-10-06:
Hi Maxi,

There are a few things of...
gst-inspect-1.0 should list all elements. 
When you only get 6 listed the majority of elements is not found.
Did you set any environment variables like `GST_PLUGIN_SYSTEM_PATH_1_0`?
That could be one possible explanation.

This is supported by the output of `gst-inspect-1.0 tcambin`.
You said you installed it according to our instructions, so it should be found under /usr.
The output states that it is in your home directory, that should not be the case.

Cheers,

Edgar

mbless, 2017-10-06:
Hi Edgar,

Thanks for your answer! I've been reviewing your comment "When you only get 6 listed the majority of elements is not found." and I found out that when I used the command $export GST_PLUGIN_SYSTEM_PATH_1_0=/.. I overwrote the other paths. So I add some new paths and now it looks like this:

[image]

Only for confirmation gst-inspect-1.0 shows me all the gstreamer plug in (or at list a long long list) and tcambin shows no error (a win!):

[image]

But I'm afraid that I've done the same mystake for GI_TYPELIB_PATH and now the path looks like this:
[image]

Can this bring me problems?

After these modifications on the environment variables I´ve tried again the command $gst-launch-1.0 tcambin ! videoconvert ! autovideosink and this came out:

[image]

Do you happend to know why I still can't use my camera?

Just in case I leave a printscreen of the ccmake configuration before I used $make and $sudo make isntall.

[image]

I'm sorry if this post seems to long, but I wanted to make my issue as clear as possible.
Thank you in advance and I really appreciate your help.

Kind regards,

Maxi

TIS-Edgar, 2017-10-09:
Hi Maxi,

`I'm sorry if this post seems to long, but I wanted to make my issue as clear as possible.`
No problem at all, it's better to post more information than not enough. 
In fact your last screenshot shows something.

Is there a reason you install binaries and libraries into your home directory?

For debugging purposes you can launch the gst pipeline like:
`gst-launch-1.0 --gst-debug=tcamsrc:5 tcambin ! videoconvert ! ximagesink`

If the camera makes problems we should now get an output what is causing them.
My current suspicion is that ~/Programas/tis/lib is not in your ld search path.
This leads to the gstreamer elements not being able to find the backend libraries which do things like opening the camera.

Two additional steps that ensure it is a software and not a hardware/kernel:
Does the camera work with qv4l?
Does `tcam-ctrl -l` list the camera?

-Edgar

mbless, 2017-10-09:
Hi Edgar,

> Is there a reason you install binaries and libraries into your home directory?

The reason I've  install binaries and libraries into your home directory is to have this program only for this user. But nothing special. Will this cause any troubles?

> My current suspicion is that ~/Programas/tis/lib is not in your ld search path.

Your suspicious were right, it was not in the GST_PLUGIN_SYSTEM_PATH_1_0, but now it is!
[image]

I've lauch `gst-launch-1.0 --gst-debug=tcamsrc:5 tcambin ! videoconvert ! ximagesink` as recommed and this is my outcome:

[image]

I've tried another pipeline that specifies the camera seria number `gst-launch-1.0 tcamsrc serial=XXXXXXXX --gst-debug=tcamsrc:5 ! videoconvert ! xvimagesink`

[image]

So it seems that the device can't be opened by gst-launch-1.0 despite it is recognice with the command `lsusb`:

[image]

I'm thinking perhaps that the UDEV Rules are required, but according to my reading camera DFK 23UV024 does not required it. Am I wrong?
Regarding  

> Does the camera work with qv4l?

I've no clue how to set up a pipeline to make my camera work with qv4l. Could you please gave me an example?

>Does tcam-ctrl -l list the camera?

It seems that this line is not setting up in my environment variables or I'm missing something on my typping:

[image]

Thanks again for your help!

Kind regards,

Maxi

TIS-Edgar, 2017-10-10:
Hi Maxi,

> The reason I've install binaries and libraries into your home directory is to have this program only for this user. But nothing special. Will this cause any troubles?

It seems that way. Your setup is something we haven't really thought of, so there seems to be a problem we haven't thought of. Our backends (v4l2, aravis, libusb) are all in separate libraries that are loaded during runtime. You can find them in a folder named tcam-0. My suspicion is that your setup causes problem due to the folder not being found.

To test this:
```
export TCAM_LOG=DEBUG
gst-launch-1.0 tcambin ! videoconvert ! ximagesink
```
If it tells you that the backends could not be loaded we likely have found the problem.
The remedy would either be to install everything systemwide or to further alter your environment.

`export LD_LIBRARY_PATH="$LD_LIBRARY_PATH:/home/mys/Programas/tis/lib/"`

That is the ld I was refering to. It expands the library search path for your library linker.

> I'm thinking perhaps that the UDEV Rules are required, but according to my reading camera DFK 23UV024 does not required it. Am I wrong?

udev rules are only required when dealing with legacy devices or special cases. The dfk23 cameras are neither. You can check whether something on that level is wrong by looking at your 'dmesg' output after attaching a camera. It should have a line like `uvcvideo: Found UVC 1.10 device DFK 23UV024 (199e:8414)`  and a device like /dev/video0 should be there (if you have a built in webcam it should be /dev/video1)

> I've no clue how to set up a pipeline to make my camera work with qv4l. Could you please gave me an example?
qv4l is a standalone application that directly interacts with v4l2. You should be able to install it through your package manager. 

-Edgar

mbless, 2017-10-10:
Hi Edgar,

So first at all I can confirm that my camera is alive and working perfectly! With the firmware updates given by Stefan (send him a warn regard) and with your help with the path's configuration it know shows image!! So this issue can be considered as closed
Just to let you know the problem was in my ccmake setting up in the folders path. I've made the correction recommended by you and added the new path in .bashrc to have them uploaded at the begging of my session.
Thanks you very much for your help! 
Kind regards,

Maxi

closed 2017-10-10 by mbless