TheImagingSource/tiscamera #115: How can I change the resolution?

AbnerCSZ, 2017-10-12:
I want to change the resolution but:
when I run tcam-ctrl -p 30714360
result is 
```
Found 5 propert(y/ies)
Brightness                              (int) min=0 max=255 step=1 default=128 value=128 category=Exposure
Gain                                    (int) min=16 max=63 step=1 default=16 value=16 category=Exposure
Exposure                                (int) min=100 max=250000 step=100 default=12700 value=50000 category=Exposure
Focus                                   (int) min=0 max=1000 step=1 default=0 value=0 category=Lens
Trigger Mode                            (bool) default=false value=false category=Special

```

and when I run tcam-ctrl -f 30714360
result is 
```
Available format settings:
Format: 8-bit Greyscale - Fourcc(808466521)
	Resolution: 744x480
		 76.0456 fps
		 60.0000 fps
		 30.0000 fps
		 25.0000 fps
		 15.0000 fps
		  7.5000 fps
		  5.0000 fps
	Resolution: 640x480
		 87.0000 fps
		 60.0000 fps
		 30.0000 fps
		 25.0000 fps
		 15.0000 fps
		  7.5000 fps
		  5.0000 fps
	Resolution: 320x240
		161.2903 fps
		120.0000 fps
		 80.0000 fps
		 60.0000 fps
		 30.0000 fps
		 25.0000 fps
		 15.0000 fps

Format: 8-bit Bayer GBGB/RGRG - Fourcc(1196573255)
	Resolution: 744x480
		 76.0456 fps
		 60.0000 fps
		 30.0000 fps
		 25.0000 fps
		 15.0000 fps
		  7.5000 fps
		  5.0000 fps
	Resolution: 640x480
		 87.0000 fps
		 60.0000 fps
		 30.0000 fps
		 25.0000 fps
		 15.0000 fps
		  7.5000 fps
		  5.0000 fps
	Resolution: 320x240
		161.2903 fps
		120.0000 fps
		 80.0000 fps
		 60.0000 fps
		 30.0000 fps
		 25.0000 fps
		 15.0000 fps

```
And when I run tcam-ctrl -s -f "format=RGB32,width=640,height=480,framerate=15.0,binning=0" 30714360
result is 
```
Given property string is faulty!

```

TIS-Edgar, 2017-10-13:
Hi,

There was indeed a mistake in the tool. I have just pushed a fix.

The following command example should work now:
`tcam-ctrl -f -s "format=GBRG,width=640,height=480,framerate=15.0" 30714360`

-Edgar

AbnerCSZ, 2017-10-13:
Still the same error:
```
xlab@xlab-ThinkPad-T460p:~/catkin_ws$ tcam-ctrl -f -s "format=GBRG,width=640,height=480,framerate=15.0" 30714360
Given property string is faulty!

```

And when I use 
```
gst-launch-1.0 tcambin serial=30714360 ! video/x-raw,format=RGBx,width=640,height=480,framerate=15/1 ! videoconvert ! ximagesink
```
I can change the resolution by tcambin.The result is good.

However I need to change it by v4l2src,when I run it , error is 
```
xlab@xlab-ThinkPad-T460p:~/catkin_ws$ gst-launch-1.0 v4l2src ! xvimagesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2865): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.000558290
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

```

so how can I use the v4l2src ?

AbnerCSZ, 2017-10-13:
I what to use v4l2src because I have to run the gscam in ROS.
[gscam](http://wiki.ros.org/gscam)
I can  get pictures in 72Hz when I run the  example launch. The most important place is there:
`<env name="GSCAM_CONFIG" value="v4l2src device=/dev/video0 ! video/x-raw-yuv,framerate=30/1,width=640,height=480 ! ffmpegcolorspace " />`
Unfortunately, When I change the framerate, the actual framerate is always 72Hz.

TIS-Edgar, 2017-10-13:
Ok, I will have to look into set-format....

Concerning your problem with v4l2src:

The source only gives you the formats the camera has. In your case Y800 and bayer 8-bit.
The following line should work (assuming gstreamer-1.0):
```v4l2src device=/dev/video1 ! bayer2rgb ! video/x-aw,format=RGBx,width=640,height=480,framerate=15/1! videoconvert``

AbnerCSZ, 2017-10-13:
The result is 
```
NODES
  /v4l/
    gscam_driver_v4l (gscam/gscam)

auto-starting new master
process[master]: started with pid [3268]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to dcdbe842-afd8-11e7-ac61-e4b318cbbf0b
process[rosout-1]: started with pid [3281]
started core service [/rosout]
process[v4l/gscam_driver_v4l-2]: started with pid [3285]
[ INFO] [1507873161.484114323]: Using gstreamer config from rosparam: "v4l2src device=/dev/video1 ! bayer2rgb ! video/x-raw,format=RGBx,width=640,height=480,framerate=15/1! videoconvert"
[ INFO] [1507873161.487853507]: camera calibration URL: package://gscam/examples/uncalibrated_parameters.ini
[ INFO] [1507873161.488084374]: Loaded camera calibration from package://gscam/examples/uncalibrated_parameters.ini
[ INFO] [1507873161.508969260]: Time offset: 1507873026.238
[ INFO] [1507873161.770559792]: Publishing stream...
[ INFO] [1507873161.770767625]: Started stream.
[ERROR] [1507873161.770786089]: Could not get gstreamer sample.
[ INFO] [1507873161.770802086]: Stopping gstreamer pipeline...
[ INFO] [1507873161.771217369]: GStreamer stream stopped!
[ INFO] [1507873161.771232886]: Cleaning up stream and exiting...
[v4l/gscam_driver_v4l-2] process has finished cleanly
log file: /home/xlab/.ros/log/dcdbe842-afd8-11e7-ac61-e4b318cbbf0b/v4l-gscam_driver_v4l-2*.log

```
And the ROS program exited.

When I use v4l2src by this 
v4l2src device=/dev/video1 ! video/x-raw,format=RGBx,framerate=15/1! ffmpegcolorspace
I can get pictures by ROS.(Only problem is that I cant change the framerate)

AbnerCSZ, 2017-10-13:
And I can run
gst-launch-1.0 v4l2src device=/dev/video1 ! video/x-raw,format=GRAY8,width=640,height=480,framerate=87/1 ! videoconvert ! ximagesink
and get Gray pictures.
But why I can't run to get RGB?
```
xlab@xlab-ThinkPad-T460p:~$ gst-launch-1.0 v4l2src device=/dev/video1 ! video/x-raw,format=RGBx,width=640,height=480,framerate=87/1 ! videoconvert ! ximagesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2865): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.000581708
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

```
Or
```
xlab@xlab-ThinkPad-T460p:~$ gst-launch-1.0 v4l2src device=/dev/video1 ! video/x-raw,format=gbrg,width=640,height=480,framerate=87/1 ! videoconvert ! ximagesink
WARNING: erroneous pipeline: could not link v4l2src0 to videoconvert0

```

Or
```
xlab@xlab-ThinkPad-T460p:~$ gst-launch-1.0 v4l2src device=/dev/video1 ! bayer2rgb ! video/x-bayer,format=gbrg,width=640,height=480,framerate=87/1 ! videoconvert ! ximagesink
WARNING: erroneous pipeline: could not link bayer2rgb0 to videoconvert0

```

TIS-Edgar, 2017-10-13:
You can not run RGB because the camera does not have RGB. It has bayer.
You need to debayer the image to get rgb.

gst-launch-1.0 v4l2src device=/dev/video1 ! bayer2rgb ! video/x-raw,format=RGBx,width=640,height=480,framerate=87/1 ! videoconvert ! ximagesink

That image will have a green cast. We have a whitebalance module for our source.
Is there a particular reason you are unable to use the tcambin with gscam? It would do all this automatically for you.

AbnerCSZ, 2017-10-13:
The result is
```
xlab@xlab-ThinkPad-T460p:~/catkin_ws$ gst-launch-1.0 v4l2src device=/dev/video1 ! bayer2rgb ! video/x-raw,format=RGBx,width=640,height=480,framerate=87/1 ! videoconvert ! ximagesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2865): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.000467333
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

```

And can you tell me how to set the privacy to 1 or 0 in code, instead of by command line?

AbnerCSZ, 2017-10-16:
How can I save images with time stamps by your APIs? I think I have to know where the image buffer is. If I know it, maybe I can write a program in ROS by myself(juest publish images). I can't find this function in your examples.
By the way, do you knew OpenCV? How can I use APIs from OpenCV to save images?
For example, imwrite(const string& filename, InputArray img). So I need konw where the 'inputArray'(image cache) is.

AbnerCSZ, 2017-10-16:
In the code of examples/c/pioelinc.c, I found some codes about saving image. 
```
g_signal_connect (save_button, "clicked", G_CALLBACK (save_image), NULL);

void save_image (GtkWidget* button,
                 gpointer data)
{
    printf("saving image\n");
    close_valve(FALSE);
}
```

But I am puzzled by the caching of image data. I found the "gpointer" means void*. So I guess the image data is in the "data". But I can't get any usefull data of "data" by converting it to "char" or "int".
The function of "save_image" do nothing when I click the butto "Save". There is no picture is saved.

AbnerCSZ, 2017-10-16:
I found some codes in /examples/c/pipeline.c. Now I can get picture.But the picture quality is bad.Look like this:
[image]

How did these rasters come from? How to remove? I need your help...!~

TIS-Arne, 2017-10-17:
You are seeing the raw bayer pattern which needs to be color processed by the bayer2rgb filter. The tcambin element does all that for you, what is the reason to use the v4l2src element?

AbnerCSZ, 2017-10-18:
Because I need to use it in ROS. So I have to use gscam.[gscam](http://wiki.ros.org/gscam)
**So I have to save images and set their name be time stamps.** e.g. TimeofNow.png.
I don't know how to use tcambin to achieve it. 
I found some codes in examples/c/properties.c. But I found it set images'name by "gst_parse_launch" and set ! multifilesink name= location= ! . And this example save only raw images likes my result. 
**How can I save RGB images with time stamps by this example?**

AbnerCSZ, 2017-10-23:
In example/c/propeties.c

```
void save_image (GtkWidget* button,
                 gpointer data)
{
    close_valve(FALSE);
}

```
gpointer  means void*
But when I convert it to char* or short*, there is no data.

AbnerCSZ, 2017-11-10:
I removed libgstreamer-1.0 and used libgstreamer-0.1. Then I can used ROS package gscam normally. I don't know why but now I can change the resolution and fps in version 0.1 .

closed 2017-11-10 by AbnerCSZ

TIS-Stefan, 2017-11-10:
Hello

first of all we will provide a nice ROS sample that uses GSTreamer 1.0 for accessing the camera and the ROS gscam 0.10  module for ROS. It will be encapsulated in a little python script.

TIS-Arne, 2017-11-14:
@AbnerCSZ A ROS Example is now available in the current master branch. Maybe you want to give it a try.