TheImagingSource/tiscamera #116: auto_exposure: gstreamer 1.0 doesn't support exposure-min property

g0hl1n, 2017-10-16:
Commit 7e3838d45f1c80abd506ad6c599737b82ee20460 in the gstreamer-0.10 branch introduced the exposure-min property for auto_exposure. This property is missing in the gstreamer-1.0 implementation of auto_exposure.

Would it be possible to port this feature to the master branch?

Thanks & Regards,
Richard L.

TIS-Arne, 2017-10-17:
You are right, this is a missing feature. We will port this feature to the master branch.

g0hl1n, 2017-10-22:
Ok. Thanks. Do you already know when it will be available?

g0hl1n, 2017-11-15:
Any updates/timeline on this issue?

TIS-Edgar, 2017-11-15:
The feature is currently being tested. You can expect the feature by next week.

TIS-Edgar, 2017-11-16:
Hi,
I pushed a preview branch:
https://github.com/TheImagingSource/tiscamera/tree/exposure-min

Can you try those changes and verify that everything works as you would expect?
After a few additional internal tests and your OK I would push them to master.

g0hl1n, 2017-11-23:
Introduced to master in ddf7bc3271beaaf8d711330f8a183876d62f1fe1. Thanks.

closed 2017-11-23 by g0hl1n