TheImagingSource/tiscamera #117: Python example pipeline... What does it do???

cappelaere, 2017-10-18:
Would be nice to have some more info on examples like pipeline and make sure they work.
$python3 pipeline

** (pipeline:1785): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files

Then I have a window with 2 buttons to push. "Snap Image" and "Play"
When you hit Snap Image, nothing happens.
When you hit Play, It says "Starting stream" and button turns into "Stop".  You hit stop, it says "Stopping stream"... HUMMMMM!!!!

What are we supposed to do and/or learn from this???????

TIS-Arne, 2017-10-18:
The example should come up with a live video when hitting "Play" and save images when clicking "Snap Image". 

What camera are you using?

cappelaere, 2017-10-18:
I am using DMK 21BU618.
Thanks,
Pat.

TIS-Stefan, 2017-10-18:
Hi

If the cameras does not show up in as  /dev/video*, then please follow the instructions at https://github.com/TheImagingSource/tiscamera/wiki/UDEV-Rules 
(In case you did not already do that.)

cappelaere, 2017-10-18:
I have two cameras plugged in and I do see /dev/video0 and /dev/video1
When using examples/python/trigger-images and selecting a device, I do see an image.

TIS-Stefan, 2017-10-18:
Only one image or a stream? And also from the DMK 21BU618 or another camera? DMK 21BU618 should be expected.

cappelaere, 2017-10-18:
I only see an image from either DMK 21BU618 that are connected based on device selection.

TIS-Stefan, 2017-10-18:
Sorry for asking, but that means, you get only one image, but this is never refreshed? If yes, what happen, if you  use
gst-launch tcambin ! videoconverter ! ximagesink 
on the command line?

cappelaere, 2017-10-18:
$ gst-launch tcambin ! videoconverter ! ximagesink
-bash: gst-launch: command not found

$ gst-launch-1.0 tcambin ! videoconverter ! ximagesink
WARNING: erroneous pipeline: no element "videoconverter"

TIS-Stefan, 2017-10-18:
sorry, must have been "gst-launch-1.0" and "videoconvert".

cappelaere, 2017-10-18:
The interest here is not to run the cameras in video mode but to snap pictures at some regular interval like 7.5 per seconds

cappelaere, 2017-10-18:
$ gst-launch-1.0 tcambin ! videoconvert ! ximagesink works fine and I do see a video stream

TIS-Stefan, 2017-10-18:
Aha. I understood you have trouble running the camera at all. Therefore all my questions. So the camera is running in live stream with the python code?

TIS-Stefan, 2017-10-18:
Sorry, remove the last question.

cappelaere, 2017-10-18:
If I do:
$ export NO_AT_BRIDGE=1
The warning at startup disappears, but still no video on start....
see: https://gist.github.com/jeffcogswell/62395900725acef1c0a5a608f7eb7a05

TIS-Stefan, 2017-10-18:
Following pipeline set the camera to 7.5 fps and writes png files as fast as possible, up to 7.5 frames per second to hard disc:
 gst-launch-1.0 tcambin  ! video/x-raw,framerate=15/2 ! videoconvert ! pngenc ! multifilesink location="image%04d.png"

Unfortunately I was not able to figure out how to write BMP files. But as far as I know, PNG is lossless too.

TIS-Stefan, 2017-10-18:
You wrote
"The warning at startup disappears, but still no video on start...." and you also wrote "and I do see a video stream". Now you lost me. Do you get a video stream or not?

cappelaere, 2017-10-18:
Thanks Stefan.  This helps to get started.  I would need to do this in Python and control both cameras.  It might be nice to also see the video as images get saved... but that would be icing on the cake :)... This is for a drone application with 3D stereo + hyperspectral imaging...

TIS-Stefan, 2017-10-18:
Please give me a few minutes...

cappelaere, 2017-10-18:
$ export NO_AT_BRIDGE=1
This removes the warning but I do not see a video stream.

TIS-Stefan, 2017-10-18:
The following pipeline shows the live video in an ximagesink and saves images to png files in parallel:
`gst-launch-1.0 tcambin serial=15210439 ! video/x-raw,framerate=15/2 ! tee name=t t. ! queue  !  videoconvert ! pngenc ! multifilesink location="image%04d.png" t. ! queue ! videoconvert ! ximagesink`

The "tee" module is used to split the input pipeline into branches. The tee is named "t" here. Each new branch starts with this name as "t." and a queue is needed. So we get
Input part:
`tcambin serial=15210439 ! video/x-raw,framerate=15/2 ! tee name=t `
We use the camera's serial number to make sure, we open the one of both, we what to open. Then we set the frame and add the tee.

Save image branch:
` t. ! queue  !  videoconvert ! pngenc ! multifilesink location="image%04d.png"`
This simply saves the images. For the second camera you should change the file name.
Display branch:
`t. ! queue ! videoconvert ! ximagesink`
Explains itself.

From this you can see, you have two similar pipelines, only the serial number and file names differ.
You could use this already in a Python script, since python GStreamer has a function that gets a pipeline string passed. However, the stuff is not as simple, because there is also the starting and stopping etc. needed. We'll get that later, since I am running out of time now.

cappelaere, 2017-10-18:
This is a great start with great explanations.  THANK YOU!
[What about timestamping the filenames? :)]

TIS-Stefan, 2017-10-18:
[Two-Cams.zip](https://github.com/TheImagingSource/tiscamera/files/1394917/Two-Cams.zip)

I created a small python sample, consisting of two py files. The "TIS.py" contains a class, that encapsulates the GStreamer stuff, handles camera properties and converts the GStreamer image buffers in OpenCV images.
The "Program.py" is the main program. We instantiate two Objects of our TIS class:
```
LeftCam = TIS.TIS("15210439",640, 480, 15,1,False )
RightCam = TIS.TIS("48010377",640, 480, 15,1,False )
```
The first parameter is the serial number of each camera. "15,1" is the frame rate, currently it is 15 fps. For 7.5 fps, "15/2" should be passed.

We create two OpenCV windows, so we have a live video:
```
cv2.namedWindow('Left Camera', 1)
cv2.namedWindow('Right Camera', 1)
```
Then we start the streams:
```
LeftCam.Start_pipeline()
RightCam.Start_pipeline()

cv2.waitKey(1000)
```

In a loop we get the images from the cameras and show them:

```
        while ( lastkey != 27):
                LeftImage = LeftCam.Get_image()
                RightImage = RightCam.Get_image()
                
                 cv2.imshow('Left Camera', LeftImage)
                 cv2.imshow('Right Camera', RightImage)

                lastkey = cv2.waitKey(67) # 67 = 1000 / 15 fps
```
In the source code this is longer, since there is error handling and other stuff in. The waitKey() wait time in ms should be somewhat longer than the frame rate's time span, otherwise we will receive tons of error message, because we ask too fast for new images.

Since we have OpenCV image now, we can do, what ever we want with same. Saving, Stitching and create file names also as we want, using the Python functions for this.

I suppose, you will add some more user interaction, like a key press for start of image saving, end of image saving. That has the advantage of setting up your scene first, see the result in the camera live view and then start saving.
The OpenCV windows will be created overlapped. So move one of them away.

I hope, that gives a starting point.

cappelaere, 2017-10-18:
Awesome.
Will study the code and try it.
Thanks,
Pat.

Pat Cappelaere
pat@cappelaere.com

Cell: 410 340 4868
Skype: patrice_cappelaere
gmail: cappelaere@gmail.com

TIS-Stefan, 2017-10-18:
Would be nice to mix the images to an RGB24 image: left camera to the red channel, right camera to the green or blue channel, so you get an anaglyph image. You only need  red /green or red /blue glasses and you have your own cheap VR system. (When I tried that, I felt sick.)

closed 2018-04-12 by TIS-Stefan