TheImagingSource/tiscamera #122: How to get the gige-daemon enabled？

JoelinChee, 2017-10-19:
We bought the camera DFK 23U445. When following the readme file downloaded from github,we found that our linux 17.04 system can hardly enable gige-daemon.service. 

sudo systemctl daemon-reload  This step is OK.
sudo systemctl enable gige-daemon.service This step get rise to a problem, saying that Failed to enable unit: File gige-daemon.service: No such file or directory.

What should we do? Should we download some files?

TIS-Stefan, 2017-10-19:
Hello

if you do not plan to use GigE cameras at, then you do not need the Gige-daemon. Also you build the tiscamera Gstreamer modules without Aravis support as shown in the Readme:

```
# Without ARAVIS
cmake -DBUILD_ARAVIS=OFF -DBUILD_GST_1_0=ON -DBUILD_TOOLS=ON -DBUILD_V4L2=ON -DCMAKE_INSTALL_PREFIX=/usr ..

make
sudo make install
```
Of course I will discuss the gige-deamon issue with the programmers.

TIS-Arne, 2017-10-19:
If you build tiscamera without support for GiGE cameras, ie. you used the "Without ARAVIS" way to configure the compiler like it is mentioned by @TIS-Stefan , then the gige-daemon is also *not* being build. This is the expected behaviour since using the gige-daemon does not make any sense in this configuration. 

I admit this should be stated clearer in the README file.

closed 2017-10-19 by TIS-Arne