TheImagingSource/tiscamera #123: Some problems about gtk+-3.0

JoelinChee, 2017-10-24:
After ARAVIS installed and Gige-vision devices installed, I want to run some examples. When I make the code in the examples/c， the program always says that 'Package gtk+-3.0 was not found in the pkg-config search path'. How can I get the gtk 3.0 files? 

Thanks!

TIS-Stefan, 2017-10-24:
Hello

you may try
`sudo apt-get install libgtk-3-dev libnotify-dev`

Stefan

JoelinChee, 2017-10-24:
Thans for your help. 
When I want to compile save-video.c, the program says fatal error: gst/gst.h: no such file or dictionary. 
#include <gst/gst.h>
Could you tell me where can I get the gst.h and gst file? How to solve this problem?

TIS-Stefan, 2017-10-24:
Hi

As shown in the readme at https://github.com/TheImagingSource/tiscamera you may follow the "Build dependencies" and "Runtime dependencies" installation. That should do the job, even for Aravis only.

```
# Build dependencies
sudo apt-get install git g++ cmake pkg-config libudev-dev libudev1 libtinyxml-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libglib2.0-dev libgirepository1.0-dev libusb-1.0-0-dev libzip-dev uvcdynctrl python-setuptools libxml2-dev libpcap-dev libaudit-dev libnotify-dev autoconf intltool gtk-doc-tools

# Runtime dependencies
sudo apt-get install gstreamer1.0-tools gstreamer1.0-x gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly libxml2 libpcap0.8 libaudit1 libnotify4
```

JoelinChee, 2017-10-24:
Yes， I have followed the readme file and follow the "Build dependencies" and "Runtime dependencies" installation, but the problem still exist. Which part goes wrong?

JoelinChee, 2017-10-24:
I use the command ./pipeline to run the code, it works. Thank you for your support. 
The program is running, but the window 'TCam Pipeline Example' appearing shows no video. When I click the save button, no video is saved. What should I do?

TIS-Stefan, 2017-10-24:
First of all
Which camera model do you use?
Secondly:
Does the green LED lights up constantly and the yellow blinks on the camera?
If not, the camera does not have power. Power the camera either by PoE switch / injector or an external power supply.

The green LED does not light up constantly, the yellow one blinks- The camera is not connected to a Gigabit ethernet network. If you use a PoE injector or switch, check, whether it supports GigE/1000mBit. Also make sure, your computer has a GigE Ethernet port. (The Raspberry has 100mBit only.)

P addresses should be:

If DHCP enabled and the camera is connected directly to the computer:

NIC: 169.254.a.b : 255.255.0.0

cam: 169.254.c.d : 255.255.0.0

a,b,c and d will have random values between 0 and 255.

NIC = Network Interface Controller

cam = camera.

If DHCP enabled and the camera is connected through a network with DHCP server:

NIC: 192.168.0.a : 255.255.255.0

cam: 192.168.0.b : 255.255.255.0

a and b will have random values between 0 and 255. Also the 192.168.0 depends on your DHCP server, but the three numbers should be the same on NIC and camera.

If persistant IP is enabled:

NIC: 192.168.1.a : 255.255.255.0

cam: 192.168.1.b : 255.255.255.0

JoelinChee, 2017-10-25:
I use the camera DFK 23U445. I think there is no LED in the camera.

JoelinChee, 2017-10-25:
Should I use the same way you told me to check the camera?

TIS-Stefan, 2017-10-25:
Ah, ok. Then the Aravis GigE Vision library does not make any sense, right? With GigE Vision libraries you can not handle an USB camera. 

In order do get that sorted out correctly, please follow the instructions on the readme and build without Aravis (not necessary).
Start the "tcam-ctrl" :
`tcam-ctrl -l`
This should list the camera.
Get a video with
`gst-launch-1.0 tcambin ! videoconvert ! ximagesink`

Stefan

JoelinChee, 2017-10-25:
Finally, I can use the DFK 23U445 to get a video . Really thank you for your support TIS-Stefan. I will be highly praise for the imagingsource camera. 
The next step, I will use this camera to do the pedestrian detection. I still have a little questions. What is aravis? Why the DFK 23U445 camera do not need the aravis? In order to run the program, the library gst.h and the library gtk.h are included. What are the two libraries? Thank you.

TIS-Stefan, 2017-10-25:
Aravis is a "GigEVision" library. "GigEVision" is a protocol for network cameras only. The USB cameras are no network cameras. 
The DFK 23U445 is an UVC compatible USB camera. The Linux v4l2 driver handles UVC compatible cameras automatically. 
gst.h is a header file of the GStreamer SDK
gtk.h is a header file of the GTK SDK. 
SDK = Software Development Kit,
GTK is a frame work for graphical user interfaces (GUI).

willingtonvr, 2017-10-28:
Hello
how do i do to get the appsink example to actually capture some frames, i run it but nothing seems to happen.  (using DFK 33GX249)

thank you

closed 2019-07-04 by TIS-Edgar