TheImagingSource/tiscamera #125: `make install` installs tcam header but not dependent headers

Nomadluap, 2017-10-30:
tcam.h includes the following: 

```
#include "version.h"
#include "base_types.h"
#include "CaptureDevice.h"
#include "Properties.h"
#include "MemoryBuffer.h"
#include "DeviceInfo.h"
#include "DeviceIndex.h"
#include "ImageSink.h"
#include "serialization.h"
#include "public_utils.h"
```

However these files are not themselves installed into the installation root. This makes the installed ticam.h file useless.

Copying all header files in the source directory into a subfolder of the include directory would alleviate this issue.

TIS-Stefan, 2017-11-01:
Did you follow the readme on https://github.com/TheImagingSource/tiscamera and also install the runtime files as mentioned?

TIS-Arne, 2017-11-02:
@Nomadluap The "tcam.h" header file is only used internally and gets installed in error. Applications using our software should not include this file by themself. The file will not be installed into the include path with future versions of our software.

closed 2017-11-02 by TIS-Arne

Nomadluap, 2017-11-02:
So then using the c++ api directly is not supported?

TIS-Arne, 2017-11-02:
No it is not. There is neither API documentation nor are there examples for the usage. The internal API is not even guaranteed to be fixed - it might get changed without notice anytime.