TheImagingSource/tiscamera #126: where are the plugins like "tiscolorize"?

yulpun, 2017-11-03:
After I **sudo make install** 
then
GST_PLUGIN_SYSTEM_PATH_1_0=:/home/so/tiscamera/build/src/gstreamer-1.0:/home/so/.$://usr/lib/x86_64-linux-gnu/gstreamer-1.0 
but when i run
**gst-launch-0.10 v4l2src device=/dev/video1 ! video/x-raw-gray ! tisvideobufferfilter ! tis_auto_exposure ! tiscolorize ! queue ! tiswhitebalance ! queue ! bayer2rgb ! queue ! ffmpegcolorspace ! pngenc ! filesink**

it is wrong:
WARNING: erroneous pipeline: no element "tisvideobufferfilter"

input:  gst-inspect-1.0
output:
[image]

but "tisvideobufferfilter" and "tiscolorize" don't exist. how to fix it and open the color camera?

TIS-Stefan, 2017-11-03:
Hello

Gstreamer 0.10 is outdated. Better use GStreamer 1.0. This will solve your issue.

Also, you mixed in the GST_PLUGIN_SYSTEM_PATH_1_0 variable the paths to GStreamer 0.10 and GStreamer 1.0 modules, which does not make any sense in your case. If you use the tiscamera  modules for GStreamer 1.0, you do not need to change GST_PLUGIN_SYSTEM_PATH_1_0

Please follow the instructions in the readme at https://github.com/TheImagingSource/tiscamera

yulpun, 2017-11-03:
Hi Stefan.
 **gst-launch-1.0 tcambin ! videoconvert ! autovideosink** is ok. And i do follow the instructions in the readme. But i can not find the plugins like "tiscolorize". The "tcambin" do exist.
If I run **gst-launch-1.0 tcambin ! videoconvert ! autovideosink ! tiscolorize**
The result is: WARNING: erroneous pipeline: no element "tiscolorize"
How to correct it ? Thank you!

TIS-Stefan, 2017-11-03:
Hi
tiscolorize does not exist for GStreamer 1.0, because it is not necessary. Its functionality is built into tcambin. Thus, please forget this old module.

Also you can not connect anything to the autovideosink. It has no src pad, if I am right.

yulpun, 2017-11-03:
So what is the command to open the color image?
I only get the gray image, so still in confused.
BTW, My camera is DFK 23UP031 and run in ubuntu16.04.

TIS-Stefan, 2017-11-03:
Hi

gst-launch-1.0 tcambin ! video-/raw, format=BGRx ! videoconvert ! autovideosink 

I admit, that is not very well documented.

yulpun, 2017-11-04:
Hello
 I tried the command
_gst-launch-1.0 tcambin ! video-/raw, format=BGRx ! videoconvert ! autovideosink_
the result is : 
WARNING: erroneous pipeline: could not link tcambin0 to videoconvert0

Then I tried the command 
gst-launch-1.0 v4l2src device=/dev/video1 ! bayer2rgb ! video/x-raw,format=RGBx  ! videoconvert ! ximagesink
The result is:
`Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2948): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.823865513
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
`
Maybe I can see one frame, then it would crash immediately.
Thanks.

TIS-Arne, 2017-11-06:
@yulpun I am very sorry but Stefan had a typo in the comand line he send you. This command should produce a color image:

```
gst-launch-1.0 tcambin ! video/x-raw, format=BGRx ! videoconvert ! ximagesink 
```

TIS-Stefan, 2017-11-06:
I am very sorry for the missing "x".

closed 2018-04-12 by TIS-Stefan