TheImagingSource/tiscamera #127: ARV_BUFFER_STATUS_TIMEOUT

haseeb-uob, 2017-11-09:


TIS-Stefan, 2017-11-10:
Hello

First make sure, you use an GigE network and GigE devices. If your switch or PoE Injector supports 100mBit only, you wont get images.

Enable Jumbo Frames. It is done like the following:
`ip link set enp2s0 mtu 9000`
"enp2s0" is the name of your network controller.

These are my first guesses.

haseeb-uob, 2017-11-10:
Yes, you are right. I am using a GigE switch with 1000mbps speed and also the mtu is 9000.

TIS-Stefan, 2017-11-10:
The I must continue guessing....

what is the exposure time or is exposure set to automatic?
Is the external trigger disabled?
Is the IP configuration correct, so there are no conflicts?

haseeb-uob, 2017-11-10:
Exposure time 10000
yes software and hardware trigger is disabled and I am using continuous mode
Yes, Ip configuration is fine. when I run ./arv-camera-test I could see the device properties, features etc...

haseeb-uob, 2017-11-10:
sudo ethtool eth1 | grep Speed
Speed: 1000Mb/s

TIS-Stefan, 2017-11-10:
Also the arv_viewer does not show an image?

haseeb-uob, 2017-11-10:
nope. I can see the camera name but can't see any images.

TIS-Stefan, 2017-11-10:
Which computer model / manufacturer do you use?
Which Linux version do you use?

haseeb-uob, 2017-11-10:
Ubuntu 14.04 64 bit
Manufacturer: AES-HM76Z1FL
Processor Intel® Core™ i7-3517UE CPU @ 1.70GHz × 4 
RAM 8 GB

TIS-Stefan, 2018-04-12:
Hello and sorry for not answering. 

I also should have looked in the "arv_camera_test.txt" you posted at start. 

The frame rate list of the camera seems to be wrong, so you must pass a valid frame rate to the camera. 

This should have been fixed with a newer firmware.

Stefan

closed 2018-04-12 by TIS-Stefan