TheImagingSource/tiscamera #128: Camera only returns images when closed/reopened

jveitchmichaelis, 2017-11-17:
Hi,

I'm using a DFK33UP2000 on an Nvidia Jetson TX2. When using any of the examples, the camera works every other time it's opened.

For example, I open the cpp live viewer demo. If I close and reopen the demo, the camera enumerates but delivers completely black images. If I close and reopen, everything is fine and I get normal images again. This seems to be regardless of how the camera is used - Python+OpenCV, gst-launch, the demo applications, etc.

Given that my code works on Windows, is this a library bug? I'll be able to test on another Ubuntu Linux machine in a couple of days.

Thanks!

TIS-Stefan, 2018-04-12:
Hello and sorry for not answering.

First of all, in Windows the access to the camera is completely different. In Linux using v4l2 we have a limited range of video format and frame rates, which must be specified in the USB descriptor of the camera. Therefore, you get the limited list of frame rates.

I would like you to run follwing command:
`gst-launch-1.0 tcambin serial=09719922 ! video/x-raw, format=BGRA,width=640,height=480,framerate=30/1 ! videoconvert ! ximagesink
`
I also tried
`gst-launch-1.0 tcambin serial=09719922 ! video/x-raw, format=BGRA,width=1920,height=1200,framerate=5/1 ! videoconvert ! ximagesink`
but I got no image. I need to check, what happened there.

TIS-Stefan, 2018-04-13:
Hello again

I used the DFK 33UP2000 on my Jetson TX with tcam-view and 1920x1200 resolution. I get all offered frame rates, even 1/30 working, I always get an image. 

Also I get a live video every time I start tcam-view. 

So this may is a firmware issue, therefore please give me the serial number of your camera.

Also there have been many updates on tiscamera since November, so maybe a git pull and rebuild does the job.

Stefan

closed 2018-04-13 by TIS-Stefan