TheImagingSource/tiscamera #130: USB 2.0 DMK 22BUC03 - can't take video output

aplastiras, 2017-11-29:
Dear all,

I am using the imaging source USB 2.0 DMK 22BUC03 camera with RPI3 and I am trying to capture at first video in Linux to see if everything is working fine. I have updated the camera firmware as described here https://github.com/TheImagingSource/tiscamera/wiki/Getting-Started-with-USB-2.0-CMOS-Cameras and now I am trying to take video capture with cheese software. Cheese is working fine with a simple webcam

The device is listed in the /dev/ folder as video0. 

the v4l2-ctl --all gives the results below:

pi@raspberrypi:~ $ v4l2-ctl --all
Driver Info (not using libv4l2):
        Driver name   : uvcvideo
        Card type     : DMx 22BUC03
        Bus info      : usb-3f980000.usb-1.4
        Driver version: 4.9.41
        Capabilities  : 0x84200001
                Video Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps   : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
        Width/Height      : 640/480
        Pixel Format      : 'GREY'
        Field             : None
        Bytes per Line    : 640
        Size Image        : 307200
        Colorspace        : Default
        Transfer Function : Default
        YCbCr/HSV Encoding: Default
        Quantization      : Default
        Flags             :
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 640, Height 480
        Default     : Left 0, Top 0, Width 640, Height 480
        Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 640, Height 480
Selection: crop_bounds, Left 0, Top 0, Width 640, Height 480
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 30.000 (30/1)
        Read buffers     : 0
                     brightness (int)    : min=0 max=255 step=1 default=-8193 value=128
                           gain (int)    : min=16 max=63 step=1 default=57343 value=16
              exposure_absolute (int)    : min=1 max=2500 step=1 default=127 value=127
                        privacy (bool)   : default=0 value=0

I have not yet install the tiscamera lib. 

Please let me know if I am doing something wrong

Thank you in advance
Apostolos

TIS-Stefan, 2017-11-29:
Hello

Unfortunately you let us not know the results you achieve with Cheese. Any error messages?

Stefan

aplastiras, 2017-11-29:
Sorry,

cheese in displaying that "There was an error playing video from the webcam."

Thank you

TIS-Stefan, 2017-11-29:
I think, Cheese does not like the Y800 mono format of the camera. Therefore it wont play. You may also look at the Cheese support or forums, maybe you will find there help for that program.

closed 2018-04-12 by TIS-Stefan