TheImagingSource/tiscamera #131: Example error on ubuntu 16.04

midjji, 2017-11-29:
The readme proposes trying: 
install deps

cmake 
make 
make install

gst-launch-1.0 tcambin ! videoconvert ! autovideosink
which gives 
WARNING: erroneous pipeline: no element "tcambin"
no file named tcambin is created by the script

BUILD_TOOLS=ON
makes no difference.

TIS-Stefan, 2017-11-30:
What about
`sudo make install`

midjji, 2017-11-30:
Yeah did that too once irealized install prefix wasnt respected,

He issue is including that and with the default prefix.

chutsu, 2017-12-05:
Can confirm that all the examples (C, C++, Python) do not work under Ubuntu 16.04 even after performing make install, it seems there is a fundamental problem with GStreamer not being able to access the camera. I have had some success in viewing images via `qv4l2` but not via the example code provided.

TIS-Stefan, 2017-12-11:
I must admit, I am lost. I use Ubuntu 16.04. Our modules are working fine on my machine. 

If there is no element "tcambin", then something went wrong. Have you been able to compile and link the modules? with the command "make" in the "build" directory?

Stefan

chutsu, 2017-12-11:
Ok, I think I resolved the issue, I forgot how I did it so I cannot explain here, I now have a different problem which I'll post a new issue.

TIS-Stefan, 2017-12-12:
ok.

closed 2017-12-12 by TIS-Stefan