TheImagingSource/tiscamera #132: Error using ROS example

Nick-Zhang1996, 2017-12-01:
Hi, I was trying the ROS example and got the following error message. I think the most important one is about shmsrc. Roscore is running, I was using 23U445 USB3.0 Camera. Could you provide some help? Thanks

```

tcambin serial="39710010" name=source ! video/x-raw,format=BGRx,width=1920,height=1080,framerate=30/1 ! tee name=t t. ! queue ! videoconvert ! video/x-raw,format=RGB ,width=1920,height=1080,framerate=30/1! shmsink socket-path=/tmp/ros_mem t. ! queue ! videoconvert ! ximagesink

(python3:7342): GStreamer-CRITICAL **: gst_caps_get_size: assertion 'GST_IS_CAPS (caps)' failed

(python3:7342): GStreamer-CRITICAL **: gst_caps_get_size: assertion 'GST_IS_CAPS (caps)' failed

(python3:7342): GStreamer-CRITICAL **: gst_caps_get_structure: assertion 'index < GST_CAPS_LEN (caps)' failed

(python3:7342): GStreamer-CRITICAL **: gst_structure_get_value: assertion 'structure != NULL' failed

(python3:7342): GStreamer-CRITICAL **: gst_structure_set_value: assertion 'G_IS_VALUE (value)' failed

(python3:7342): GStreamer-CRITICAL **: gst_structure_get_value: assertion 'structure != NULL' failed

(python3:7342): GStreamer-CRITICAL **: gst_structure_set_value: assertion 'G_IS_VALUE (value)' failed

(python3:7342): GStreamer-CRITICAL **: gst_structure_get_value: assertion 'structure != NULL' failed

(python3:7342): GStreamer-CRITICAL **: gst_structure_set_value: assertion 'G_IS_VALUE (value)' failed
Press Enter to end program[ INFO] [1512085734.058685460]: Using gstreamer config from env: "shmsrc socket-path=/tmp/ros_mem ! video/x-raw-rgb, width=1920,height=1080,framerate=30/1,bpp=24,depth=24,blue_mask=16711680, green_mask=65280, red_mask=255 ! ffmpegcolorspace"
[ INFO] [1512085734.066234291]: using default calibration URL
[ INFO] [1512085734.066447095]: camera calibration URL: file:///home/nick/.ros/camera_info/camera.yaml
[ INFO] [1512085734.066766618]: Unable to open camera calibration file [/home/nick/.ros/camera_info/camera.yaml]
[ WARN] [1512085734.066896066]: Camera calibration file /home/nick/.ros/camera_info/camera.yaml not found.
[ INFO] [1512085734.067003056]: Loaded camera calibration from 
[FATAL] [1512085734.075187251]: no element "shmsrc"

```

TIS-Stefan, 2017-12-01:
Hello

"width=1920,height=1080" can not work, since the camera's maximum resolution is 1280x960.  A frame rate of 30 fps is available.
Please update the code with 1280x960 in the tis constructor parameters. It is in tis_rosstater.py at
cam = tiscamera.Camera("17719924", 1920, 1080, 30, True, False)
It should be
cam = tiscamera.Camera("39710010", 1280, 960, 30, True, False)

Stefan

Nick-Zhang1996, 2017-12-02:
Thank you. I adjusted the values as you said. The same error persistes. I messed around a little bit and believe the problem is with ROS gscam. When I setup environment variable GSCAM_CONFIG and run 'rosrun gscam gscam' it complains no element 'xxx' whichever pipeline I try to run. Even thoses that would work when I call gst-launch directly. Any clues?

Nick-Zhang1996, 2017-12-02:
What's the test environment where the code is developed? Could you tell me your ROS version and Gstreamer version? I'm using kinetic with 1.0

TIS-Stefan, 2017-12-04:
The "xxx" is the interesting part. It seems, you need the GStreamer 0.10 modules to be installed, but I am not sure about that anymore. You may look at http://wiki.ros.org/gscam.

Please let me know, which modules "xxx" are missing. Thank you in advance!

Stefan

Nick-Zhang1996, 2017-12-04:
When I run the official example under ROS folder, the error was 'no element shmsrc'. I removed and reinstalled several packages to mess with 0.10 and 1.0 version of gst, and the original error disappeared. However, now it says something about unable to PAUSE the pipeline. I remember reading somewhere that this is a permission issue, so I chmod +777 to /dev/video0, yet the problem persists. I'm sorry that these information is vague. I will upload repeatable(hopefully) error scenario and screenshots as soon as possible. Meanwhile, could you make something out of what I said?

TIS-Stefan, 2017-12-05:
Does the standard pipeline
gst-launch-1.0 tcambin ! videovonvert ! ximagesink
give an image?

Nick-Zhang1996, 2017-12-05:
Yes it does. And I can get shmsrc and shmsink to work with gst-launch-1.0.

makabe0510, 2017-12-19:
Hello.
I was trying the ROS example and got the following error message.
I use DFKAFUJ003 USB3.0 camera.
I'm running ROS indigo on ubuntu14.04.
Then I canged tis_rosstarter line 12 and run this.
cam = tiscamera.Camera("17719924", 1920, 1080, 30, True, False)
↓
cam = tiscamera.Camera("12710001", 640, 480, 30, True, False)

"エレメント "shmsrc" がありません" in following error massage means "no element "shmsrc" ".
So I checked whether I have installed gstreamer library correspond to "shmsrc" element or not typing command.
$ gst-inspect-1.0 shmsrc
Since I could find shmsrc detail , I think gstreamer library has been installed.

Because of my ROS distribution, I installed "ros-indigo-gscam" instead of "ros-kinetic-gscam"
Can I use this ROS example under ROS indigo?

$ ./tis_rosstarter
tcambin serial="12710001" name=source ! video/x-raw,format=BGRx,width=640,height=480,framerate=30/1 ! videoconvert ! video/x-raw,format=RGB ,width=640,height=480,framerate=30/1! shmsink socket-path=/tmp/ros_mem
Press Enter to end program[ INFO] [1513666463.044921393]: Using gstreamer config from env: "shmsrc socket-path=/tmp/ros_mem ! video/x-raw-rgb, width=640,height=480,framerate=30/1,bpp=24,depth=24,blue_mask=16711680, green_mask=65280, red_mask=255 ! ffmpegcolorspace"
[ INFO] [1513666463.047287782]: using default calibration URL
[ INFO] [1513666463.047340407]: camera calibration URL: file:///home/leus/.ros/camera_info/camera.yaml
[ WARN] [1513666463.047909559]: [camera] does not match name narrow_stereo in file /home/leus/.ros/camera_info/camera.yaml
[ INFO] [1513666463.047952862]: Loaded camera calibration from
[FATAL] [1513666463.049343617]: エレメント "shmsrc" がありません
[FATAL] [1513666463.049369666]: Failed to initialize gscam stream!

$ gst-inspect-1.0 shmsrc
Factory Details:
Rank none (0)
Long-name Shared Memory Source
Klass Source
Description Receive data from the shared memory sink
Author Olivier Crete <olivier.crete@collabora.co.uk>

Plugin Details:
Name shm
Description shared memory sink source
Filename /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstshm.so
Version 1.2.4
License LGPL
Source module gst-plugins-bad
Source release date 2014-04-18
Binary package GStreamer Bad Plugins (Ubuntu)
Origin URL https://launchpad.net/distros/ubuntu/+source/gst-plugins-bad1.0

GObject
+----GInitiallyUnowned
+----GstObject
+----GstElement
+----GstBaseSrc
+----GstPushSrc
+----GstShmSrc

Pad Templates:
SRC template: 'src'
Availability: Always
Capabilities:
ANY

Element Flags:
no flags set

Element Implementation:
Has change_state() function: 0x7f2e1432d910

Element has no clocking capabilities.
Element has no indexing capabilities.
Element has no URI handling capabilities.

Pads:
SRC: 'src'
Implementation:
Has getrangefunc(): gst_base_src_getrange
Has custom eventfunc(): gst_base_src_event
Has custom queryfunc(): gst_base_src_query
Has custom iterintlinkfunc(): gst_pad_iterate_internal_links_default
Pad Template: 'src'

Element Properties:
name : The name of the object
flags: readable, writable
String. Default: "shmsrc0"
parent : The parent of the object
flags: readable, writable
Object of type "GstObject"
blocksize : Size in bytes to read per buffer (-1 = default)
flags: readable, writable
Unsigned Integer. Range: 0 - 4294967295 Default: 4096
num-buffers : Number of buffers to output before sending EOS (-1 = unlimited)
flags: readable, writable
Integer. Range: -1 - 2147483647 Default: -1
typefind : Run typefind before negotiating
flags: readable, writable
Boolean. Default: false
do-timestamp : Apply current stream time to buffers
flags: readable, writable
Boolean. Default: false
socket-path : The path to the control socket used to control the shared memory transport
flags: readable, writable
String. Default: null
is-live : True if the element cannot produce data in PAUSED
flags: readable, writable
Boolean. Default: false

TIS-Stefan, 2017-12-19:
Hello

I am very sorry, I am not a ROS expert. I am able to install the standard package with apt-get and this supports GStreamer 0.10 only. Therefore, there can be a problem to get the old modules.

I do not know, what "ros-indigo-gscam" is. In case it uses GStreamer 1.0, then the pipeline for it must be adapted. It should be something like

shmsrc socket-path=/tmp/ros_mem ! video/x-raw, format=rgb, width=640,height=480,framerate=30/1,bpp=24,depth=24,blue_mask=16711680, green_mask=65280, red_mask=255 ! videconvert"
 But I am not able to test this on my own

Stefan

Nick-Zhang1996, 2017-12-28:
@makabe0510 Hey Makabi:
After some messing around I solved the problem. It is likely cause by the lack of required gst packages. Roscam uses 0.1 (by default, it supports 1.0 in beta/experiment version too, but I didn't mess with that) version of gst while tiscamera uses 1.0. I forgot which exact package solved the problem but here's something that may be helpful:
[image]

mdemirst, 2018-05-22:
@Nick-Zhang1996 your installed gstreamer packages list helped me to get it solved! after installing missing packages that I noticed from your list, it stopped outputting error related to shmsrc. Thanks!

finphin, 2018-06-13:
@TIS-Stefan Hi, Stefan:

I use DFK 33UX290 camera.
I'm running ROS indigo on ubuntu14.04.
Also has the same issue as @makabe0510 

Then I  tried running ROS kinetic on ubuntu16.04
After install @Nick-Zhang1996 missing packages.
I can use rqt_image_view to see image_raw now.

May I ask if the current TIS ROS sample only works on ROS kinetic on ubuntu16.04?
Or how can I use it on ROS indigo on ubuntu14.04?

By the way, how can I check the 3A setting works
cam.set_property("Exposure Auto", True)
cam.set_property("Focus Auto", True)
cam.set_property("whitebalance-auto", True)
Because there is no difference when I use rqt_image_view to check image_raw

TIS-Stefan, 2018-06-13:
Hi

"3A" is new to me :-)

The only way checking whether they work is having a look on the colored image. If it looks fine, then they work. That means, the average brightness of the image is 128, in case there is enough light and the colors are not greenish. However the camera runs the automatics per default.
The GStreamer plugins are created for Ubuntu 16.04, on older Kernels on 14.04 you will encounter issues.

You may try the simple pipeline:

`gs-launch-1.0 tcambin ! videoconvert ! ximagesink`

That should give a nice image, if all works fine. That is nearly the same, our ROS example script does.

In th pyhton script is a line like the follwing,

`cam = tiscamera.Camera("17719924", 1920, 1080, 30, True, False)`

The fifth parameter should be "True". If it is "False", you will get a raw image GRAY8 with bayer pattern. 

Stefan

finphin, 2018-06-14:
@TIS-Stefan Hi, Stefan:

I use DFK 33UX290 camera.
I'm running ROS kinetic on ubuntu16.04 now.

I have use the following command to list all ROS supported properties in gst plugins
cam.list_properties()
And it works well to set, get and see the image_raw in rqt_image_view now

But the ROI function is strange to me, I have get the property value to double confirmed.
// ROI command sequence
cam.set_property("Auto Functions ROI Control", True)
cam.set_property("Auto Functions ROI Left", 0)
cam.set_property("Auto Functions ROI Top", 0)
cam.set_property("Auto Functions ROI Width", 32)
cam.set_property("Auto Functions ROI Height", 32)
=> The ROI function seems no work
=> it does no cropping either 3A detected area 
=> what is this ROI function for?

TIS-Stefan, 2018-06-14:
Hi 

The automatics should look only on the ROI rectangle for adjusting the 3A. So if the image is very bright in the upper left corner only, the complete image should become darker.

Stefan

finphin, 2018-06-15:
@TIS-Stefan Hi, Stefan:

As you said "So if the image is very bright in the upper left corner only, the complete image should become darker."
That's why I ask the question, that the ROI functions seems no work in your ROS sample code.
The test image is attached.
[image]

I use DFK 33UX290 camera.
My code is listed as following.
I have get the property value to double confirmed already.
Can you help me to check this ROI functions works well with gst plugins?
Maybe you can reproduce this situation also and find ROI works or not.

// ROI command sequence
cam.set_property("Auto Functions ROI Control", True)
cam.set_property("Auto Functions ROI Left", 0)
cam.set_property("Auto Functions ROI Top", 0)
cam.set_property("Auto Functions ROI Width", 32)
cam.set_property("Auto Functions ROI Height", 32)

TIS-Stefan, 2018-06-15:
Yes, that should have been the expected behavior.
However, If I try that with tcam-view,  it works as expected. 
The exposure and gain automatics are still enabled and I think, the camera does not run in trigger mode (In this case, the image should become much brighter to the bottom, because the frame rate and not the exposure time will determine the real exposure time.)

You may build the examples/cpp files , then go into examples/cpp/build/tcam-view and run the tcam-view program

Stefan

finphin, 2018-06-22:
@TIS-Stefan  Hi, Stefan:

The image test result is just like you mentioned in tcam-view.

But we use the ROS topic camera/image_raw to receive the image, which has no ROI effect in image.
I am using rqt_image_view to verify the ROS image.
Can you check how can I use your ROS example to get ROI function works well in ROS topic camera/image_raw ?

Thanks. :)

TIS-Stefan, 2018-06-22:
Do you run the x290 camera triggered? If yes, then the exposure time has no effect.
Also I must admit, I am totally unexperienced with ROS. Working with this is always a restart with high effort,
Is automatic for gain and exposure enabled? If not ....
However, I tested the sample tis_rosstarter.py:

```
import tiscamera
import time

# Open the camera. Parameters are serial number, width, height, frame rate, color and liveview.
cam = tiscamera.Camera("49619914", 1920, 1080, 30, True, True)

# Start the live stream from the camera and also "rosrun"
cam.start_pipeline()

# Set some properties
cam.set_property("Exposure Auto", True)
cam.set_property("Gain Auto", True)
cam.set_property("Brightness Reference", 128)

# ROI command sequence
cam.set_property("Auto Functions ROI Control", True)
cam.set_property("Auto Functions ROI Left", 0)
cam.set_property("Auto Functions ROI Top", 0)
cam.set_property("Auto Functions ROI Width", 32)
cam.set_property("Auto Functions ROI Height", 32)

xmax = 1880
xpos = 0
try:
    while True:
        if xpos == 0:
            xdir = 1

        if xpos == xmax:
            xdir = -1

        xpos += xdir
        cam.set_property("Auto Functions ROI Left", xpos)

        time.sleep(0.01)

except KeyboardInterrupt:
    print("Keyboard interrupt")

# Stop the camera pipeline.
cam.stop_pipeline()

print('Program ended')
```
The live view is enabled, so I can see it without ROS. The example works as expected. The ROI is moved on top from left to right and the automatic makes adjusts the image according to the ROI brightness.

Stefan

closed 2018-06-22 by TIS-Stefan