TheImagingSource/tiscamera #134: Unable to set exposure manually

ndjankov, 2017-12-15:
Greetings, 

I have a DFK 33UX264 camera and I am trying to capture images within Python (2.7) and OpenCV (3.3) on Linux (Ubuntu 14.04). The capture part is working pretty well; however, I am unable to figure out how to disable the auto exposure and fix it to a specific value. I have found documentation on how to enable auto exposure ([here](https://github.com/TheImagingSource/tiscamera/wiki/Gstreamer-1.0)) using the tcamautoexposure module, but nothing on how to turn it off. 

I've tried several gstreamer pipelines on the command line from:
"gst-launch-1.0 tcamsrc ! xvimagesink"
to:
"gst-launch-1.0 tcamsrc ! video/x-bayer,width=2448,height=2048 ! bayer2rgb ! videoconvert ! xvimagesink"
to:
gst-launch-1.0 tcamsrc ! video/x-bayer,width=2448,height=2048 ! tcamautoexposure auto-exposure=false ! bayer2rgb ! videoconvert ! xvimagesink
to:
gst-launch-1.0 tcamsrc name=src ! video/x-bayer,width=2448,height=2048 ! tcamautoexposure auto-exposure=true exposure-min=25 exposure-max=25 ! bayer2rgb ! videoconvert ! xvimagesink

In all cases the exposure is always being adjusted automatically. (In my test I have the camera pointing at a light, so when I cover it the exposure/gain get cranked up, then saturates to white when I remove my hand, then gradually returns to proper "automatic" exposure.

I know there must be something trivial that I am missing, I just haven't been able to find it. 

Thank you, 

Nick.

Myzhar, 2018-02-03:
I have the same issue. Have you solved it?

TIS-Stefan, 2018-02-05:
Hello

if you work from command line only. then you can use the "tcam-ctrl" program for setting these properties. Simply start the program without parameter, then you get a help.
Also I would like you to update, build and install the tiscamera repository, because there was a problem with exposure being listed twice.
If you want to use fixed exposure time you might disable gain too, because it has its own automatic.

```
tcam-ctrl -p -s "Exposure Auto=false" <SERIAL>
tcam-ctrl -p -s "Gain auto=false" <SERIAL>
tcam-ctrl -p -s "Gain=0" <SERIAL>
tcam-ctrl -p -s "Expsure=33333" <SERIAL>
```
The last line sets an exposure time.

In Python the apprach is a little bit different. First of all, we have our pipeline:

```
class TIS:
    def __init__(self,serial, width, height, framerate, color):
        p = 'tcambin serial="%s" name=source ! video/x-raw,format=%s,width=%d,height=%d,framerate=%d/1' % (serial,format,width,height,framerate,)
            p += ' ! videoconvert ! tee name=t'
            p += ' t. ! queue ! appsink name=sink'
            p += ' t. ! queue ! ximagesink'

self.pipeline = Gst.parse_launch(p)
```
The "tcambin" got a name "source" here. We need that for setting properties:
`self.source = self.pipeline.get_by_name("source")`

Now we can get and set properties:
```
def Get_Property(self, PropertyName):
            return CameraProperty(*self.source.get_tcam_property(PropertyName))

def Set_Property(self, PropertyName, value):
            self.source.set_tcam_property(PropertyName,GObject.Value(type(value),value))
```
You can also list the properties:
```
def List_Properties(self):
        for name in self.source.get_tcam_property_names():
            print( name )
```
Usage:
```
Tis = TIS.TIS("14610615",2592, 1944, 5,False )
print("Gain Auto : %s " % Tis.Get_Property("Gain Auto").value)
print("Gain : %d" % Tis.Get_Property("Gain").value)
print("Exposure : %d" % Tis.Get_Property("Exposure").value)

# Check, whether gain auto is enabled. If so, disable it.
if Tis.Get_Property("Gain Auto").value :
        Tis.Set_Property("Gain Auto",False)
        print("Gain Auto now : %s " % Tis.Get_Property("Gain Auto").value)

Tis.Set_Property("Gain",16)

# Now do the same with exposure. Disable automatic if it was
[tiscamera.zip](https://github.com/TheImagingSource/tiscamera/files/1694647/tiscamera.zip)
 enabled
# then set an exposure time.
if Tis.Get_Property("Exposure Auto").value :
        Tis.Set_Property("Exposure Auto",False)
        print("Exposure Auto now : %s " % Tis.Get_Property("Exposure Auto").value)

Tis.Set_Property("Exposure",3000)
'``
The sample code is not complete and without error handlers for simplifying. But the zip file contains the complete class and sample program.

Stefan

Myzhar, 2018-02-05:
I work in C++, I have my own appsink and I would like to disable auto-gain and auto-exposure from the gstreamer pipeline.
`"tcamsrc serial=%1"
 " ! video/x-bayer,width=%2,height=%3,framerate=%4/1"
 " ! tcamautoexposure"
 " ! tcamwhitebalance"
 " ! bayer2rgb"`
where 
%1 is the camera serial
%2,%3 the frame size (752x480)
%4 the FPS (100)
If I disable automatic white balance I get a "black image" if I do not correctly set red,green and blue "gains"
if I disable auto-exposure and auto-gain I get no effects.
I cloned the last version of the repository

TIS-Stefan, 2018-04-12:
Hello  and sorry for the delay of my answer. 

I am not sure, how your source code looks like, therefore, I make a guess.

First of all you pass a name to the tcamsrc in your pipeline. Also you remove the tcamautoexposure and tcamwhitebalance, because the camera does this in hardware:

`tcamsrc serial=%1 name=source ! video/x-bayer,width=%2,height=%3,framerate=%4/1 ! bayer2rgb ...`

Once the pipeline is established, you can get the source with

`GstElement *source= gst_bin_get_by_name(GST_BIN(pipeline), "source");`

This is used for setting properties:

```
void set_property_camera_enum(GstElement* source, const char *Property, const int Value )
{
   GValue val = {};
   char PropertyType[20];
   TcamProp*  Properties = TCAM_PROP(source); // from above
   if( Properties != NULL )
   {
      strcpy( PropertyType, tcam_prop_get_tcam_property_type(Properties, (gchar*)Property));
      
//      if( strcmp( PropertyType, "enum") == 0 )
      {
         g_value_init(&val, G_TYPE_STRING);
         if( Value == 1 )
            g_value_set_string(&val, "Continuous");
         else
            g_value_set_string(&val, "Off");
      }
      
      if( !tcam_prop_set_tcam_property(Camera.Properties ,(char*)Property , &val) )
      {
         printf("Failed to set %s\n", Property);
      }
   }
   else
   {
      printf("Error: pipeline not built! call start_camera first.\n");
   }
}
```
You can use tcam-ctlr -p serial number for getting the property names and pass them to the function above.

Also you can use the class in tcamcamera.cpp and tcamcamera.h in  "/examples/cpp/common/". It wraps all the Gstreamer code.