TheImagingSource/tiscamera #136: Setting exposure error for DFK AFU420-CCS

limbolily, 2017-12-19:
I used TcamCamera::set_property interface to set the exposure value to 80,then the exposure value I got by TcamCamera::get_property is 36.

In the file src/libusb/AFU420DeviceProperties.cpp,function create_shutter:
```
bool AFU420Device::create_shutter ()
{
    auto prop = create_empty_property(TCAM_PROPERTY_FOCUS); //Not TCAM_PROPERTY_SHUTTER?

    prop.value.b.value = false;
    prop.value.b.default_value = false;

    auto property = std::make_shared<PropertyBoolean>(property_handler, prop, Property::BOOLEAN);

    property_handler->properties.push_back({property});

    return true;
}
```
Is it relevant to my issue?

TIS-Edgar, 2017-12-19:
Hi,
You are correct that create_empty_property(TCAM_PROPERTY_FOCUS) is wrong.
It does not how ever explain the issues you are having. 
The standard exposure range is 100 - 30000000 with a step size of 100.
So something is definitely wrong.
Are one on current master?
Is there a pipeline running in the background? If there is a tcamautoexposure element in your pipeline it will overwrite your manual changes.
You can also set "Exposure Auto" to false to prevent that.

limbolily, 2017-12-19:
I checkout the current master, and it works correctly,thanks.

closed 2017-12-19 by limbolily