TheImagingSource/tiscamera #137: The DFK 23UV044 camera is not recognized on the USB 3.0 port.

danielsmorais, 2017-12-22:
Hi,

I'm using a DFK 23UV044 camera. However, my notebook does not recognize it when used on the USB 3.0 port. I already tested it on Ubuntu 16.04 and Windows 10 and the problem remains.

Att.,

Daniel

TIS-Arne, 2018-01-03:
@danielsmorais I appologize for the late reply. We worked with reduced staff because of the christmas season. 

If the camera is not detected at all, is not a tiscamera issue (or a software issue in general). Could you please open a support case by sending an email with a description of the issue to support@theimagingsource.com? Thanks!

danielsmorais, 2018-01-03:
Hi @TIS-Arne, 

Thanks for the answer! 
The problem still remains. I'll contact support.

Thanks!

closed 2018-01-05 by TIS-Arne