TheImagingSource/tiscamera #138: list_property python example doesn't list all properties?

Nick-Zhang1996, 2017-12-28:
In the ROS examples, I saw a line:
`cam.set_property("Exposure Auto", True)`

However, when I run the python example list_properties, it does not show this particular setting. 
`We will use the default device.
Gain(integer) value: 822 default: 174 min: 174 max: 1023 grouping: Exposure - Gain
Exposure(integer) value: 33333 default: 33333 min: 100 max: 300000000 grouping: Exposure - Exposure
Gain (dB/100)(integer) value: 0 default: 0 min: 0 max: 3039 grouping: Unknown - INVALID_PORPERTY
Trigger Mode(boolean) value: False default: False grouping: Special - Trigger Mode
Software Trigger(button) grouping is Special -  Trigger Mode
Trigger Delay (us)(integer) value: 150 default: 0 min: 0 max: 10000000 grouping: Special - Trigger Mode
Strobe Enable(boolean) value: False default: False grouping: Special - Strobe Enable
Strobe Polarity(boolean) value: False default: False grouping: Special - Strobe Enable
Strobe Exposure(boolean) value: True default: False grouping: Special - Strobe Enable
Strobe Duration(integer) value: 100 default: 0 min: 0 max: 10000 grouping: Special - Strobe Enable
Strobe Delay(integer) value: 0 default: 0 min: -10000 max: 10000 grouping: Special - Strobe Enable
GPOut(integer) value: 0 default: 0 min: 0 max: 1 grouping: Special - GPIO
GPIO(integer) value: 0 default: 0 min: 0 max: 1 grouping: Special - GPIO
`
There is Exposure, which is an integer. But there's no item called 'Exposure Auto' that takes a boolean value. Could someone explain this to me? 

Also, are all the properties here settable with syntax `cam.set_property("Exposure Auto", True)`? 
Also, are these properties specific to tiscameras? If so, how can we get a complete list or manual of all available options? If this is something standard, please point me toward the right direction. 

Thank you!

TIS-Arne, 2018-01-05:
This is likely an issue of the list_properties.py example. I will investigate and fix this issue soon. What causes the issue is that the camera does not have native (hardware) support for automatic exposure controls, so the "Exposure Auto" property has to be implemented in software. The "tcambin" handles this but it is only able to detect that software auto exposure is required when the elements are in "READY" state. 

So in short, if you list the properties after calling 
```tcambin.set_state(Gst.State.READY)```
it should list all properties, including "Exposure Auto".

Nick-Zhang1996, 2018-01-09:
I'll test on that. Thanks. 

Also, could you briefly explain the difference between Exposure and Gain? Which one is equivalent to ISO setting? What's the logic behind these too settings?  Is there somewhere we can get an explanation of what each and every setting means?

TIS-Stefan, 2018-01-10:
Hi Nick
Exposure time = time the sensor is light sensitive.
Gain : electronic amplify
There is no equivalent to ISO setting. The sensor manufacturer does not give that information.

Stefan

Nick-Zhang1996, 2018-01-11:
Does Exposure have a unit? like ms?

TIS-Stefan, 2018-01-12:
Yes. It depends on the sensor, which uinit is used. Usually it is µs.

closed 2018-01-12 by TIS-Stefan