TheImagingSource/tiscamera #140: tis_rosstarter fail to work in colorless mode

Nick-Zhang1996, 2018-01-09:
I'm working with 23U445, on an arm platform (Odroid XU4).

When I test the tis_rosstarter with color set to True, the program seems to work properly, though the window does not show the actual video stream. However I was able to read the image in my own code.

When I set the color to False,  the video stream is properly displayed( seems to be at a lower framerate than my setting) However, the output window says:

gst_caps_get_structure: assertion 'GST_IS_CAPS (caps)' failed

gst_structure_get_int: assertion 'structure !=NULL' failed

GStreamer image buffer underflow: Expected frame to be 10759456 bytes but got only 3686400 bytes. (make sure frames are correctly encoded)

What's the problem here? My settings are 1280*960@30fps

Thanks

TIS-Stefan, 2018-01-10:
Hello Nick

Do you use a DMK 23U445 or a DFK 23U445?
Does the simple pipeline
`gst-launch-1.0 tcambin ! videoconvert ! aasink `
show an ASCII graphics live video?
I wonder, where the  number 10.759.456 comes from, since 1280*960 is 1.228.800 only.
How does the constructor call in "tis_rostarter" at line 12
`cam = tiscamera.Camera("17719924", 1920, 1080, 30, True, False)`
looks in your script?
Should be 
`cam = tiscamera.Camera("17719924", 1280, 960, 30, False, False)`

Nick-Zhang1996, 2018-01-10:
Hi Stefan.
Thank you for your reply. DFK 23U445

    cam = tiscamera.Camera("39710010", 1280, 960, 30, False, True)

is my line in the script.
The number of bytes sent in each frame is correct, the number expected is not. I think the pipeline is missing a cap in one link, but I don't know enough about gst to figure that out.

I remember the simple pipeline gives me same result as if I set color to True.  Nothing from command line output (the last line is time offset = xxxxxxx ) and a window that doesn't display anything. I'll test again tomorrow.