TheImagingSource/tiscamera #141: GigE C++ change sink location

Mitchell-Hampton, 2018-01-15:
Can't seem to find how to set up the full pipeline. I am able to change frame-rate/resolution but that is it.

Here is the pipeline I need to set up:
#gst-launch-1.0 tcamsrc \
#! video/x-bayer,width=1280,height=720,framerate=14/1 \
#! bayer2rgb \
#! videoconvert \
#! x264enc threads=8 bitrate=1000 tune=zerolatency \
#! 'video/x-h264, profile=(string)main' \
#! rtph264pay config-interval=1 pt=127 \
#! udpsink host=127.0.0.1 port=5001

Any advice? This is where I was told to ask questions

TIS-Stefan, 2018-01-16:
Hello

What exactly you want to do? 
Which camera do you use? 
Which problems do you encounter
Which udp receiver do you use?

Stefan

TIS-Stefan, 2018-04-10:
Never got an answer. Closed.

closed 2018-04-10 by TIS-Stefan