TheImagingSource/tiscamera #145: DFK AFU420-CCS fails to display video

tadmorgan, 2018-01-30:
I'm running Ubuntu 16.04 on both a ThinkPad P50 development system and Nvidia Jetson TX2 target system and attempting to use the DFK AFU420-CCS. Under Windows on the ThinkPad the camera works fine. Under Linux both the ThinkPad & Jetson have the same issues:

1) The AFU420 is not added as a /dev/video source for V4L2.
2) The standard streaming command doesn't work:
```
root@tegra-ubuntu:~# gst-launch-1.0 tcambin ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...

(gst-launch-1.0:29638): GStreamer-CRITICAL **: gst_caps_get_size: assertion 'GST_IS_CAPS (caps)' failed

(gst-launch-1.0:29638): GStreamer-CRITICAL **: gst_caps_get_size: assertion 'GST_IS_CAPS (caps)' failed
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source: Internal data flow error.
Additional debug info:
gstbasesrc.c(2948): gst_base_src_loop (): /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.050626062
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```
3) tcam_view captures a single frame & stops, outputting the following along the way:
```

(tcam_view:15652): GLib-GObject-WARNING **: value "-inf" of type 'gdouble' is invalid or out of range for property 'value' of type 'gdouble'

(tcam_view:15652): GLib-GObject-WARNING **: value "-inf" of type 'gdouble' is invalid or out of range for property 'value' of type 'gdouble'
double property 'Gain Min': value = 0.000000 default = 64.000000 min = 64.000000 max = 520.000000 step_size = 1.000000
double property 'Gain Max': value = 520.000000 default = 0.000000 min = 64.000000 max = 520.000000 step_size = 1.000000

```
4) live_video.py generates:
```
TypeError: Must be number, not NoneType

(live_video.py:16355): GStreamer-CRITICAL **: gst_caps_get_size: assertion 'GST_IS_CAPS (caps)' failed

(live_video.py:16355): GStreamer-CRITICAL **: gst_caps_get_size: assertion 'GST_IS_CAPS (caps)' failed
Segmentation fault (core dumped)
```

I have built the code according to the instructions and didn't see any issues.

This is the output from lsusb -v for that device:
```
Bus 002 Device 005: ID 199e:0804 The Imaging Source Europe GmbH 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x199e The Imaging Source Europe GmbH
  idProduct          0x0804 
  bcdDevice            0.0c
  iManufacturer           1 The Imaging Source Europe GmbH
  iProduct                2 DFK AFU420-CCS
  iSerial                 6 35710488
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           31
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              224mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              14
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength           22
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000c
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   3
      Lowest fully-functional device speed is SuperSpeed (5Gbps)
    bU1DevExitLat           0 micro seconds
    bU2DevExitLat           0 micro seconds
Device Status:     0x0000
  (Bus Powered)
```

Please let me know if there are any other details I can provide or tests I can do to help resolve this issue.

Thank you.

beligum, 2018-01-30:
Sorry for cross-posting (see #146 ), but I'm just figuring out we're having the same problem.
The example that does seem to work for me is `live-stream.sh` which doesn't really do anything but displaying the camera feed, but it's a start...

tadmorgan, 2018-01-30:
Alas I get basically the same result with live-stream.sh as with the standard command line.

```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstTcamSrc:tcamsrc0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2948): gst_base_src_loop (): /GstPipeline:pipeline0/GstTcamSrc:tcamsrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.142486789
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

tadmorgan, 2018-02-01:
Thanks, that has fixed many of the issues!

2) Oddly enough, the standard command line and live-stream.sh don't work on Jetson, but do on the ThinkPad:
```
root@tegra-ubuntu:# gst-launch-1.0 tcambin ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
ERROR: from element /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source: Internal data flow error.
Additional debug info:
gstbasesrc.c(2948): gst_base_src_loop (): /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:01.177383092
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

or with live-stream.sh:
```
ERROR: from element /GstPipeline:pipeline0/GstTcamSrc:tcamsrc0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2948): gst_base_src_loop (): /GstPipeline:pipeline0/GstTcamSrc:tcamsrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.267363936
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```
3) tcam_view works just fine on both Jetson & ThinkPad!
4) live_video.py DOES work on both. It does get the libusb errno=12 messages, though.

This definitely gives me a start, hopefully the gstreamer from command line can be fixed. Is there any additional info I can provide that might help? Thanks again!

TIS-Edgar, 2018-02-01:
errno 12 typically means that your system prevents further allocations.
40 transfer buffers is probably too much for your system. 
If have reduced the number to 12. That should make that error go away on the Jetson.

tadmorgan, 2018-02-05:
Where is that value adjusted (in this code, gstreamer, libusb, kernel...)? Thanks!

TIS-Edgar, 2018-02-06:
Ok, I have no real idea where that comes from.
Could you give me the output of the following command:

    TCAM_LOG=DEBUG gst-launch-1.0 --gst-debug=tcamsrc:5,tcambin:5 --gst-debug-no-color tcambin ! videoconvert ! autovideosink

That should contain all information needed to find the source of this problem.

Thanks!

tadmorgan, 2018-02-06:
Thanks, here's the output as an attachment (since its kinda long):

[tiscamdebug.txt](https://github.com/TheImagingSource/tiscamera/files/1700463/tiscamdebug.txt)

tadmorgan, 2018-02-06:
So, studying the live_video.py script and comparing with the live-stream.sh and the standard command line, I find this combination works:
```
sudo gst-launch-1.0 tcambin name=src ! queue max_size_buffers=2 ! videoconvert ! capsfilter caps="video/x-raw,format=BGRx" ! videoconvert ! gtksink name=sink
```
At full 42 MP resolution & frame rate I get frequent warnings about dropped buffers on the Jetson. If I drop the resolution as follows this does go away:

```
sudo gst-launch-1.0 tcambin serial=35710488  name=src ! queue max_size_buffers=2 ! videoconvert ! capsfilter caps="video/x-raw,format=BGRx,width=1920,height=1080,framerate=30/1" ! videoconvert ! gtksink name=sink
```

tadmorgan, 2018-02-08:
So it seems that the sinks that have problems are the Nvidia ones, namely nvvideosink & nvoverlaysink. I have good success using nveglglessink however. It looks like autovideosink is choosing one of the first two (not sure which one).

TIS-Edgar, 2018-02-08:
I am looking into some optimizations for whitebalance as it seems to be a bottleneck.

Concerning your pipelines: is there a reason for the first videoconvert? From my understanding it would simply pass the buffers through and not convert anything.

autovideosink will choose the first display element it deems fit for usage. This will often not be the best one for your case. Choosing a sink for your use case and sticking to it will be better. How does nveglglessink compare to a xvimagesink?

Do you want to run the camera at full resolution? How does your setup look like? Maybe there are a few performance tweaks that we can try...

TIS-Edgar, 2018-02-09:
I have pushed the branch afu420_improvements that contains a few changes that will help you performance wise. That branch will also reset exposure and gain to their base values to ensure the auto exposure branch will always have the correct values.
It would be nice if you could try that branch out.

tadmorgan, 2018-02-10:
1) I'm just starting to play with gstreamer so I've just been piecing different pieces from different examples, so I probably have extras & wrong bits in there. I'd certainly appreciate any suggestions. It looks like removing one of the videoconverts was fine.

2) I think nveglglessink & xvimagesink are roughly equal, xvimagesink may take a little more CPU.

3) I'm interested in max resolution in some cases and high frame rate in other cases. Setup is a Nvidia Jetson TX2 with a 4 channel USB3 PCIe card. Lenses are kinda various at the moment.

4) Thank you very much for the aft420_improvements branch. I tried it and it seems to work much better, especially the exposure. Before I was having problems with it being a bit dark. I will continue to test that.

closed 2018-04-12 by TIS-Stefan