TheImagingSource/tiscamera #146: tcam-capture tools fails on Ubuntu 16.04 for DFK AFU420-L62

beligum, 2018-01-30:
Hi,

I'm test-driving the DFK AFU420-L62 on vanilla Ubuntu 16.04, but when I launch the tcam-capture script (as root), it shows video for about 2 seconds and then closes the shutter.

Can anyone help me fix this?

All the best,

b.

TIS-Edgar, 2018-01-31:
Hi,

I have pushed some fixes that should allow for a more stable stream.
Please be aware that at full resolution the gstreamer pipeline will likely drop frames as many elements are not multi-threaded. Lower resolutions should fix this.

I also have to mention that tcam-capture is not 100% stable, which is why it is currently not installable.
Please let me know if the performance is in an acceptable range or further improvements are needed.

Regards,

Edgar

beligum, 2018-01-31:
Thank you Edgar, I'll give it a try tomorrow. Is there a way to lower the camera resolution during startup of tcam-capture so it runs more smoothly?

TIS-Edgar, 2018-02-01:
There is currently no such option, sorry.

beligum, 2018-02-01:
Hi Edgar,

I've just tested the new update. Camera feed shows continuous now, although delayed, but acceptable for debugging/testing. So thank you already for the quick fix and swift feedback.

As few a side notes while playing with it a little bit: 

- default cmake configuration, installing to /usr/local instead of /usr doesn't work (but the cmake example command explicitly sets the /usr flag, so I took it as a hint)
- the autofocus ROI sliders in the tcam-capture don't work (at least, they don't seem to move)

The tcam-capture tools offers me exactly what I needed: a quick and dirty tool to test your camera, before making a decision, diving in, and spending hours coding to your API.

closed 2018-02-01 by beligum