TheImagingSource/tiscamera #147: DFK 31BU03.H - Green images

JonasR888, 2018-01-31:
We've set up the camera and tried it with the "properties.c"-example from this repo. (Linux - Raspberry Pi 3 B)
In the program we get the option to change stuff like:
Brigthness, Saturation, Hue etc, but when changing the sliders no change happen to the image(stream).

Options like "Gamma" and "Gain" do however make (live) changes.
We have "whitebalance blue" and "whitebalance red"-options displayed, but why is there no "whitebalance green" there? 
The problem is that our images/livefeed are completely misscolored and turn most colors green.

What's the reason for this and how can we change this? Also why won't the Brightness etc work?

We're also wondering if there is more examples or guides for this on Lunix other than in this repo. 
https://github.com/TheImagingSource/tiscamera/wiki/Getting-Started-with-USB-2.0-cameras-on-a-Raspberry-PI-or-on-other-Linux-Systems sounded promising, but it's obsolete.

Regards, Jonas

TIS-Stefan, 2018-01-31:
Hello

If you use the BA81 video format, you get bayer raw data from the camera. On this data the saturation and hue have no effect. Also the bayer start pattern seems to be wrong interpreted, therefore you get green where red is expected.

If you use YUYV video format, all properties will have an effect. The color properties have no effect on the bayer raw data. But you have half frame rate relative to BA81, because with YUYV two bytes per pixel are transported, wile BA81 transports only one byte per pixel.

Brightness controls the black level power on the sensor, therefore the visible effect is very little.

The white balance in the camera has no green slider. This is intended.

I suggest to use the qv4l2 program, select the YUYV format and try the properties.

Stefan

JonasR888, 2018-01-31:
Thanks for replying.

We managed to see 'normal' colors when running the IC Capture (end-user software program) on Windows by enabling Debayering. But as said, we want it to work on our Linux machine.

We installed the qv4l2 program on a Ubuntu 16.04 machine and tried out the reccomended format, but the picture was complete black. Changing the Gain slider did not change anything either, so it doesn't look like it's feeding. (We get the camera listed and selected in Input as Camera 1).

Using the repo examples(i.e. "properties (python)) gives a image, but it has the same problem as before; green-tinted images. 

How can we make it so that we get the same result like when we were using the Windows-IC Capture's Debayering?

We fail to find a "config" file for the camera. How are we suppose to change values, resolutions etc by code if we don't have any "physical" place to change it?
For example enabling Debayering by python code.

TIS-Stefan, 2018-01-31:
IC Capture makes the debayering in software, if BY8 selected.
Completely black? I tried the very same on my PC before writing you and that worked very fine.

`Using the repo examples(i.e. "properties (python)) gives a image, but it has the same problem as before; green-tinted images.`
Which video format is used?

`We fail to find a "config" file for the camera. How are we suppose to change values, resolutions etc by code if we don't have any "physical" place to change it?`
I do not know about a config file.
I will have a look tomorrow for a pipeline.

JonasR888, 2018-02-01:
We got the qv4l2 to show normal colored images with the YUYV format (15 fps), but is it not possible to obtain ~30 FPS and colored images with the other formats? 

Does the DFK 31BU03.H camera 'record' in raw bayern format which is black-white-images that we can turn into color ones? Pardon our imageing knownledge.

TIS-Stefan, 2018-02-01:
Hello
Please try following command line:
`gst-launch-1.0 tcamsrc serial=serialnumber ! video/x-bayer,format=bggr, width=1024, height=768, framerate=30/1 !  capssetter join=false caps="video/x-bayer,format=gbrg" ! bayer2rgb ! videoconvert ! ximagesink`

I tested that with a DFK 41BU02, since I can not find my DFK 31BU03.

The trick is 

- use tcamsrc, so we can select the video format
- use the capssetter, that allows to correct the format from "bggr" to "grbg", so bayer2rgb converts correctly

With
`tcam-ctrl -c serialnumber`
you can check, which bayer raw format is delivered by the camera.

Doing so, you can get 30 fps.

I hope, this helps

Stefanh

JonasR888, 2018-02-05:
Thanks for your reply! It has gotten us a good step further.

Next we're trying to start start filming with the camera, and saving it in a folder. 
We attempted to run the python example "live_video.py", but got the following error:

> TypeError: super() takes at least 1 argument (0 given)
> 
Like it's broken from the start of.

Could you get us on the right path to make it happen via python code?
- Start/stop recording 
- Save the videos in a specified folder
- Save it in different video formats(?)

All we got to work with is the few examples provided. Really missing a documentet library ;-)

TIS-Stefan, 2018-02-06:
Hello

the error occurs, if Python 2.7 is used.  Please install Python 3 or higher.

I am very sorry, that this is not mentioned in the readme.md file.

Stefan

Source:
https://stackoverflow.com/questions/38963018/typeerror-super-takes-at-least-1-argument-0-given-error-is-specific-to-any

closed 2018-02-06 by TIS-Stefan