TheImagingSource/tiscamera #152: DFK33GX249 fails to change resolution

willingtonvr, 2018-02-12:
Hello, 

today i got a couple DFK33GX249 cams.
 
when i run 

gst-launch-1.0 tcambin ! videoconvert ! autovideosink 

i got a tiny strip of 480x4 , if i go via arv-viewer i can successfully change to 1920x1200, but every resolution that i try with gst leads me to the same size (480x4).
 
attached useful debug info.
command for debug info: 

`TCAM_LOG=DEBUG gst-launch-1.0 --gst-debug=tcamsrc:5,tcambin:5 --gst-debug-no-color tcambin format=video/x-raw,format=BGRx,width=1920,height=1200 ! videoconvert ! autovideosink
`

Thank you !

[debuginfo.txt](https://github.com/TheImagingSource/tiscamera/files/1715360/debuginfo.txt)

TIS-Stefan, 2018-02-12:
Hello,
your pipeline is not correct. It should be
`gst-launch-1.0 tcambin ! video/x-raw,format=BGRx,width=1920,height=1200 ! videoconvert ! autovideosink
`
The "tcambin has no "format" parameter..
You can also add your debugging settings, I removed them for simplifying the line.

Stefan

closed 2018-02-12 by TIS-Stefan

willingtonvr, 2018-02-12:
Hello, i've tried the procedure and give me the followin error

`gst-launch-1.0 tcambin ! video/x-raw, format=BGRx,width=1920,heigth=1200 ! videoconvert ! autovideosink`

Estableciendo el conducto a PAUSA …

(gst-launch-1.0:2928): GStreamer-CRITICAL **: gst_caps_get_size: assertion 'GST_IS_CAPS (caps)' failed

(gst-launch-1.0:2928): GStreamer-CRITICAL **: gst_caps_get_size: assertion 'GST_IS_CAPS (caps)' failed
El conducto está vivo y no necesita PREPARARSE …
Estableciendo el conducto a REPRODUCIENDO …

(gst-launch-1.0:2928): GStreamer-CRITICAL **: gst_caps_get_structure: assertion 'index < GST_CAPS_LEN (caps)' failed

(gst-launch-1.0:2928): GStreamer-CRITICAL **: gst_structure_get_value: assertion 'structure != NULL' failed

(gst-launch-1.0:2928): GStreamer-CRITICAL **: gst_structure_set_value: assertion 'G_IS_VALUE (value)' failed

(gst-launch-1.0:2928): GStreamer-CRITICAL **: gst_structure_get_value: assertion 'structure != NULL' failed

(gst-launch-1.0:2928): GStreamer-CRITICAL **: gst_structure_set_value: assertion 'G_IS_VALUE (value)' failed

(gst-launch-1.0:2928): GStreamer-CRITICAL **: gst_structure_get_value: assertion 'structure != NULL' failed

(gst-launch-1.0:2928): GStreamer-CRITICAL **: gst_structure_set_value: assertion 'G_IS_VALUE (value)' failed
New clock: GstSystemClock
ERROR: del elemento /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source: Error en el flujo de datos interno.
Información adicional de depuración:
gstbasesrc.c(2948): gst_base_src_loop (): /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source:
streaming task paused, reason not-negotiated (-4)

attached the debug info

Thank you

[debuginfo2.txt](https://github.com/TheImagingSource/tiscamera/files/1716304/debuginfo2.txt)

TIS-Stefan, 2018-02-12:
Hmm, weird. I tried the very same without  any problems on my computer and works fine on my machine.
Do you use latest repository version? You may perform "git pull" and recompile and install.

willingtonvr, 2018-02-12:
Hello i'm prety sure that's is latest version my machine is Ubuntu 16.04
The examples also fails to show the properties of the camera.

Thank you

TIS-Arne, 2018-02-15:
@willingtonvr Sorry for the late response. Could you please run this command: 

```
GST_DEBUG=5 gst-launch-1.0 tcambin ! video/x-raw, format=BGRx,width=1920,heigth=1200 ! videoconvert ! xvimagesink 2>gsttcam.log
```

and post the resulting "gsttcam.log" file? 

Thanks!

willingtonvr, 2018-02-15:
Hello Here is the file... 

I discover that if you pass the whole format string it works, so in the C/python examples the width and height are missing so happens the same problem, also it wont survive to typos (one very common is change  height with "heigth") so the function fall back with the lowest  height possible so it gets 4 . 

Also the following code doesn't work: (so, what i'm doing wrong?)

```
GstElement *source;
// create the source element
source = gst_element_factory_make("tcambin", "source");
// put the serial on it
GValue val = {};
g_value_init(&val, G_TYPE_STRING);
g_value_set_static_string(&val, "427100042");
g_object_set_property(G_OBJECT(source), "serial", &val);
// change the device-caps 
GValue val = {};
g_value_init(&val, G_TYPE_STRING);
g_value_set_static_string(&val, "video/x-raw,format=BGRx,width=1920,height=1200");
g_object_set_property(G_OBJECT(source), "device-caps", &val);
```
here is the gstcamlog.log with the typo 
[gsttcam.log](https://github.com/TheImagingSource/tiscamera/files/1729752/gsttcam.log)

and here is corrected 

[gsttcam2.log.zip](https://github.com/TheImagingSource/tiscamera/files/1729789/gsttcam2.log.zip)

Thank you very much.