TheImagingSource/tiscamera #154: Trying to run live_video.py keep getting "'VariantDict object is not subscript ble"

jrupel, 2018-02-17:
I'm attempting to run this program on a raspberry pi model 3 with Raspbian GNU/Linux 9.1
I attempt to run the program by typing sudo python3 live_video.py -s 47410203
The camera I'm using is the DMK 42BUC03(usb 2.0 camera) and the serial number is 47410203

I then get the following error
Traceback (most recent call last):
  File "live_video.py", line 205, in do_command_line
    if options.contains("serial") and options["serial"] is not None:
TypeError: 'VariantDict' object is not subscriptable

What am I doing wrong?

jrupel, 2018-02-17:
Another note: when I run sudo tcam-ctrl --list with my camera connected to the usb port the command outputs

No devices found
Hope that helps.

TIS-Stefan, 2018-02-19:
Hello

I guess, you did not change the firmware to UVC/Linux firmware. Please have a look at https://github.com/TheImagingSource/tiscamera/wiki/Getting-Started-with-USB-2.0-CMOS-Cameras for an explanation and manual.

Stefan