TheImagingSource/tiscamera #155: Cannot make a GigE camera work with a direct connection

RhiobeT, 2018-02-19:
Hi,

As a part of a project, I need to use a DMK 23GM021 camera with a Raspberry Pi.

After spending several days trying all the things I could think about, I managed to have the camera detected by the tiscamera tools (and also the Aravis tools) when plugged to my home router.
It also works well when plugged directly to a Windows computer.
However, I just can't get it to be detected when plugged directly to the Raspberry Pi, or to any other computer running Linux actually.

It may very well be a network issue, but I'm far from being competent in networking and I just can't tell what could me missing here.
I tried to setup a DHCP server on the Raspberry but it never receives any DHCPREQUEST messages from the camera (also, do I even need a DHCP server?)
The only packets I see with Wireshark are DISCOVERY_CMD, plus the DHCPDISCOVER and DHCPOFFER messages.
I also tried to ping the whole subnet (169.254.0.0/16) using the fping and nmap commands, but nothing came out.

I am a bit stuck here :/
The camera was bought under the pretense that it was perfectly compatible with Linux, but I had mixed results so far...

Some help would be great.

TIS-Stefan, 2018-02-19:
Hello

"DMK 23GM021 camera with a Raspberry Pi." I am very sorry, but that wont work, because the PI has no GigE network controller. The camera needs the GigE speed, otherwise it denies to work.

Also the camera must be powered. Either by a PoE device or an external 12V power supply.

Yes, you can use the camera with Linux, but you must use compatible hardware. If you are bound to a PI and can not change to e.g. Odroid with GigE interface, then you should consider an USB camera, e.g.  DFK 42BUC03 or DFG 42AUC03.

Stefan

RhiobeT, 2018-02-19:
I must admit that I never considered the speed of the Ethernet port...
I was using a PoE device by the way.

Well, we should have thought of that during the design step.

Thank you for your answer, you can close this issue.

TIS-Stefan, 2018-02-20:
You may consider using
- Twisted pair network cable
Then the camera should at least ask for DHCP and the yellow LED on the camera should blink.
- lowest possible frame rate
That could do the job, but must not be stable.

Stefan

closed 2018-02-27 by TIS-Arne