TheImagingSource/tiscamera #158: gst launch before using arv-viewer

mehtajaghvi, 2018-03-02:
[image]

1.  Every time the camera is powered down and restarted or if the setup is done  for the first time, I see a dark image, the exposure and gain values are likely reset, but when I try to use arv-viewer I can set the gain and exposure but it does not do anything, I always see a black view
2. So, to get the view that was set before the camera shutdown
3. I run the gst-launch tcambin serial="" ! videoconvert ! autovideosink
4. It sets the pipeline 
5.  After this, If I run the arv-viewer, I am able to set the exposure and gain to get my "live" view back and not a black image
Is this because gst-launch is setting certain properties that aren't preset by arv-viewer, can you please explain the difference, thanks

TIS-Stefan, 2018-03-05:
Hello

Which camera do you use?

Stefan

mehtajaghvi, 2018-03-06:
Camera vendor : The Imaging Source Europe GmbH
Camera model  : DFK 33GV024
Camera vendor : The Imaging Source Europe GmbH
Camera model  : DFK 33GX290e