TheImagingSource/tiscamera #161: About the version of Cmake

AbnerCSZ, 2018-03-15:
In CMakelist.txt:

```

#Specify the version being used aswell as the language
cmake_minimum_required(VERSION 3.2)
```

Can we use cmake 2.8 to build this project and run in ROS (ubuntu14.04--indigo)

TIS-Arne, 2018-03-15:
@AbnerCSZ  We do not officially support ubuntu14.04 and recommend to update to a more recent Ubuntu version. 

That being said, there is a "ubuntu_lts_14_04" branch in the tiscamera repository that should work. You might want to try that one.

AbnerCSZ, 2018-03-16:
When I run the example in ROS, I met a problem:
```
anber@anber-XPS-8700:~/github/tiscamera/examples/ROS$ ./tis_rosstartertcambin serial="48414292" name=source ! video/x-raw,format=BGRx,width=640,height=480,framerate=60/1 ! tee name=t t. ! queue ! videoconvert ! video/x-raw,format=RGB ,width=640,height=480,framerate=60/1! shmsink socket-path=/tmp/ros_mem t. ! queue ! videoconvert ! ximagesink
Traceback (most recent call last):
  File "/home/anber/github/tiscamera/examples/ROS/tiscamera.py", line 81, in start_pipeline
    self.pid = subprocess.Popen(["rosrun", "gscam", "gscam"])
  File "/usr/lib/python3.4/subprocess.py", line 859, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.4/subprocess.py", line 1457, in _execute_child
    raise child_exception_type(errno_num, err_msg)
FileNotFoundError: [Errno 2] No such file or directory: 'rosrun'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./tis_rosstarter", line 15, in <module>
    cam.start_pipeline()
  File "/home/anber/github/tiscamera/examples/ROS/tiscamera.py", line 83, in start_pipeline
    except GLib.Error as error:
TypeError: catching classes that do not inherit from BaseException is not allowed

```

And I can run this normally. 
```
gst-launch-1.0 tcambin ! videoconvert ! autovideosink
```

What bothers me is that I can run ROS example normally in Ubuntu16.04 with ROS kinetic. But for some reason, I must use **Ubuntu14 with ROS indigo**. Can u help me to deal with it in branch "ubuntu_lts_14_04"?  
And even in this branch, the README.txt in exmaple/ROS said that 
```
sudo apt-get install ros-kinetic-gscam
```
I can only use 
```
sudo apt-get install ros-indigo-gscam
```

closed 2018-03-19 by AbnerCSZ