TheImagingSource/tiscamera #162: there is wrong with "gst-launch-1.0 tcambin ! videoconvert ! autovideosink"

xiat5, 2018-03-19:
on ubuntu 16.04 nvidia Tx1,with camera DFK 72BUC02
follow the readme,
install deps
cmake
make
sudo make install
and run "gst-launch-1.0 tcambin ! videoconvert ! autovideosink"
"WARNING: erroneous pipeline: no element "videoconvert"

how can i fix it?

TIS-Stefan, 2018-03-19:
The "videoconvert" belongs to the standard GStreamer 1.0 plugins, which are installed with the runtime files. It is installed, if you follow the instructions at https://github.com/TheImagingSource/tiscamera:

# Build dependencies
sudo apt-get install git g++ cmake pkg-config libudev-dev libudev1 libtinyxml-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libglib2.0-dev libgirepository1.0-dev libusb-1.0-0-dev libzip-dev uvcdynctrl python-setuptools libxml2-dev libpcap-dev libaudit-dev libnotify-dev autoconf intltool gtk-doc-tools

```
# Runtime dependencies
sudo apt-get install gstreamer1.0-tools gstreamer1.0-x gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly libxml2 libpcap0.8 libaudit1 libnotify4
```

Which Linux distribution do you use?

Stefan

xiat5, 2018-03-20:
Thank you for your reply.
all  deps are installed.  The problem is still unsolved.
Linux version 4.4.38-tegra

xiat5, 2018-03-20:
i delete environment variables "GI_TYPELIB_PATH", “GST_PLUGIN_SYSTEM_PATH_1_0”, and this problem solved.

closed 2018-03-20 by xiat5