TheImagingSource/tiscamera #163: something wrong with "gst-launch-1.0 tcambin ! videoconvert ! autovideosink"

xiat5, 2018-03-20:
$ gst-launch-1.0 tcambin ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source: Internal data flow error.
Additional debug info:
gstbasesrc.c(2948): gst_base_src_loop (): /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:07.759286915
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

please help me

TIS-Stefan, 2018-03-20:
Hello

please use "ximagesink" instead of "autovideosink".

Also please please let me know
- camera model
- Linux distribution
- computer model

Stefan

TIS-Stefan, 2018-04-12:
Hello

I would like you to try smaller video formats too:

`gst-launch-1.0 tcambin  ! video/x-raw, format=BGRA,width=640,height=480,framerate=15/1 ! videoconvert ! ximagesink
`
And see. what happens.

closed 2019-07-04 by TIS-Edgar