TheImagingSource/tiscamera #164: undefined reference to gsttcam::

aplastiras, 2018-03-24:
Hello to all,

I am trying to connect a GigE Camera DFK33GX249 with an application in Ubuntu 16.04 with Eclipse C++. 

I have installed the libraries needed with the dependencies as described in the readme tiscamera/README.md. I also tested with the compiled tools that the camera is recognized correctly, and also managed to take video signal from gst-launch using the command below in command line:

`gst-launch-1.0 tcambin ! videoconvert ! autovideosink`

In other words the hardware is working fine. The problem comes when for example i try to compile and run the simple tcam_demo example.
The libraries included in the project are with this order:
tcamgstbase, tcamprop, tcam, girepository-1.0, gstapp-1.0, gstbase-1.0, gstreamer-1.0, gobject-2.0, glib-2.0

Also the includes are (as it seems) correctly set. After build attempt i get the results above.

```
15:37:30 **** Build of configuration Debug for project tcamTest ****
make all 
Building file: ../src/tcamTest.cpp
Invoking: GCC C++ Compiler
g++ -std=c++0x -I/usr/local/include/ -I/usr/include/gobject-introspection-1.0 -I/usr/include/ -I/usr/include/gstreamer-1.0 -I/usr/lib/x86_64-linux-gnu/gstreamer-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -O0 -g3 -Wall -c -fmessage-length=0 -pthread -MMD -MP -MF"src/tcamTest.d" -MT"src/tcamTest.o" -o "src/tcamTest.o" "../src/tcamTest.cpp"
../src/tcamTest.cpp: In function ‘GstFlowReturn new_frame_cb(GstAppSink*, gpointer)’:
../src/tcamTest.cpp:28:28: warning: unused variable ‘data’ [-Wunused-variable]
             unsigned char* data = info.data;
                            ^
../src/tcamTest.cpp:20:18: warning: unused variable ‘caps’ [-Wunused-variable]
         GstCaps *caps = gst_sample_get_caps(sample);
                  ^
Finished building: ../src/tcamTest.cpp
 
Building target: tcamTest
Invoking: GCC C++ Linker
g++ -L/usr/local/lib -L/usr/lib -o "tcamTest"  ./src/tcamTest.o   -ltcamgstbase -ltcamprop -ltcam -lgirepository-1.0 -lgstapp-1.0 -lgstbase-1.0 -lgstreamer-1.0 -lgobject-2.0 -lglib-2.0
./src/tcamTest.o: In function `main':
/home/irmica/Documents/eclipse-workspace/tcamTest/Debug/../src/tcamTest.cpp:47: undefined reference to `gsttcam::get_device_list()'
/home/irmica/Documents/eclipse-workspace/tcamTest/Debug/../src/tcamTest.cpp:72: undefined reference to `gsttcam::TcamCamera::TcamCamera(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
/home/irmica/Documents/eclipse-workspace/tcamTest/Debug/../src/tcamTest.cpp:75: undefined reference to `gsttcam::TcamCamera::get_format_list()'
/home/irmica/Documents/eclipse-workspace/tcamTest/Debug/../src/tcamTest.cpp:79: undefined reference to `gsttcam::VideoFormatCaps::to_string[abi:cxx11]()'
/home/irmica/Documents/eclipse-workspace/tcamTest/Debug/../src/tcamTest.cpp:83: undefined reference to `gsttcam::TcamCamera::get_camera_property_list()'
makefile:45: recipe for target 'tcamTest' failed
/home/irmica/Documents/eclipse-workspace/tcamTest/Debug/../src/tcamTest.cpp:93: undefined reference to `gsttcam::TcamCamera::get_property(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
/home/irmica/Documents/eclipse-workspace/tcamTest/Debug/../src/tcamTest.cpp:110: undefined reference to `gsttcam::TcamCamera::set_new_frame_callback(std::function<GstFlowReturn (_GstAppSink*, void*)>, void*)'
/home/irmica/Documents/eclipse-workspace/tcamTest/Debug/../src/tcamTest.cpp:114: undefined reference to `gsttcam::TcamCamera::enable_video_display(_GstElement*)'
/home/irmica/Documents/eclipse-workspace/tcamTest/Debug/../src/tcamTest.cpp:117: undefined reference to `gsttcam::TcamCamera::start()'
/home/irmica/Documents/eclipse-workspace/tcamTest/Debug/../src/tcamTest.cpp:121: undefined reference to `gsttcam::TcamCamera::disable_video_display()'
/home/irmica/Documents/eclipse-workspace/tcamTest/Debug/../src/tcamTest.cpp:125: undefined reference to `gsttcam::TcamCamera::stop()'
/home/irmica/Documents/eclipse-workspace/tcamTest/Debug/../src/tcamTest.cpp:72: undefined reference to `gsttcam::TcamCamera::~TcamCamera()'
/home/irmica/Documents/eclipse-workspace/tcamTest/Debug/../src/tcamTest.cpp:72: undefined reference to `gsttcam::TcamCamera::~TcamCamera()'
./src/tcamTest.o: In function `std::shared_ptr<gsttcam::DoubleProperty> std::dynamic_pointer_cast<gsttcam::DoubleProperty, gsttcam::Property>(std::shared_ptr<gsttcam::Property> const&)':
/usr/include/c++/5/bits/shared_ptr.h:458: undefined reference to `typeinfo for gsttcam::DoubleProperty'
/usr/include/c++/5/bits/shared_ptr.h:458: undefined reference to `typeinfo for gsttcam::Property'
collect2: error: ld returned 1 exit status
make: *** [tcamTest] Error 1

15:37:31 Build Finished (took 1s.798ms)
```

I have searched all the libraries to check if there are in the right place and I found that they are. 

Are there any checks that can be done in order to find out what is going on? Are there any dependencies that I am missing?

I would be grateful if you could help me solve this problem.

Thank you in advance,
Apostolos

TIS-Stefan, 2018-03-26:
Hello Apostols

the tcam_demo is built with the the complete examples in the "examples/cpp" directory. Therefore, I would like you to follow the instructions at "https://github.com/TheImagingSource/tiscamera/tree/master/examples/cpp" first.

The "tcam_demo.cpp" depends on the cpp wrapper class in "tcamcamera". The header and cpp files, that contain this wrapper can be found in "examples/cpp/common/"

A sample CMakeList.txt, that uses this wrapper class can look like that:

```
project(tcamsoftwaretrigger)
 
cmake_minimum_required(VERSION 2.6)
set(CMAKE_CXX_STANDARD 11)

find_package(PkgConfig REQUIRED)

# Set the following to your tiscamera repository directory:
set( TISCAMERA_DIR /home/bvtest/projects/tiscamera) 

if(NOT EXISTS( ${TISCAMERA_DIR}/examples/cpp/common/tcamcamera.h))
    message( "CONFIGURATION ERROR : TISCAMERA_DIR not set to tiscamera directory." )
    #return()
endif()

pkg_check_modules(GSTREAMER REQUIRED gstreamer-1.0 gstreamer-app-1.0 gstreamer-video-1.0)
pkg_check_modules(TCAMLIB tcam)

include_directories( ${CMAKE_CURRENT_BINARY_DIR} ${TISCAMERA_DIR}/examples/cpp/common  ${GSTREAMER_INCLUDE_DIRS} ${TCAM_INCLUDE_DIRS} )
add_definitions(${GSTREAMER_CFLAGS_OTHER})  

add_executable(tcamsoftwaretrigger main.cpp ${TISCAMERA_DIR}/examples/cpp/common/tcamcamera.cpp )

target_link_libraries(tcamsoftwaretrigger ${TCAMLIB_LIBRARIES} ${GSTREAMER_LIBRARIES} )

install(TARGETS tcamsoftwaretrigger RUNTIME DESTINATION bin)
```
I hope, this helps.

Stefan

TIS-Stefan, 2018-04-10:
No more answer. Closed.

closed 2018-04-10 by TIS-Stefan