TheImagingSource/tiscamera #165: DMK 27BUJ003 on Linux

beligum, 2018-03-30:
Hi,

I've recently bought a DMK 27BUJ003 camera (after having used the AFU420-L62 successfully), but I can't seem to get any frames out of it on Linux (haven't tested on Windows since I don't have a Win machine available).

Launching both `gst-launch-1.0 tcambin ! videoconvert ! autovideosink` or `tcam-capture` doesn't show any image, although the camera discovery seems to work correctly.

Am I missing something? Does this camera need additional work, compared to the AFU420?

best,

b.

TIS-Stefan, 2018-04-10:
Hello and sorry for the delay.

Also I must ask a question back first.
Which computer model do you use?

Stefan

beligum, 2018-04-10:
Hello Stefan,

I'm running Ubuntu 16.04.4 LTS on a Dell XPS 15 9560

best,

b.

TIS-Stefan, 2018-04-12:
`gst-launch-1.0 tcambin  ! video/x-raw, format=BGRA,width=640,height=480,framerate=15/1 ! videoconvert ! ximagesink
`

beligum, 2018-04-13:
This doesn't really do anything I'm afraid.
This is my dmesg output:
```
bram@escher:~$ dmesg | tail
[249788.744647] usb 2-2: LPM exit latency is zeroed, disabling LPM.
[249788.745451] usb 2-2: New USB device found, idVendor=199e, idProduct=8609
[249788.745456] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[249788.745460] usb 2-2: Product: DMK 27BUJ003
[249788.745464] usb 2-2: Manufacturer: The Imaging Source Europe GmbH
[249788.745468] usb 2-2: SerialNumber: 10810497
[249788.746286] uvcvideo: Found UVC 1.10 device DMK 27BUJ003 (199e:8609)
[249788.747186] uvcvideo 2-2:1.0: Entity type for entity Extension 3 was not initialized!
[249788.747191] uvcvideo 2-2:1.0: Entity type for entity Processing 2 was not initialized!
[249788.747195] uvcvideo 2-2:1.0: Entity type for entity Camera 1 was not initialized!
```

So the camera seems to be recognized okay. But this is the output of your command:

```
bram@escher:~$ gst-launch-1.0 tcambin ! video/x-raw, format=BGRA,width=640,height=480,framerate=15/1 ! videoconvert ! ximagesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
```

and then, nothing happens...
Can you point me in the right direction or do I have a broken camera?

beligum, 2018-04-27:
Hi Stefan,

This issue has been resolved in the mean time.
Thank you for your help.

b.

closed 2018-04-27 by beligum