TheImagingSource/tiscamera #166: live_video.py can't not use

RozenAstrayChen, 2018-04-10:
Hi i use python to run __live_video.py__ but  i got some error:
```
Traceback (most recent call last):
  File "live_video.py", line 234, in <module>
    app = Application()
  File "live_video.py", line 198, in __init__
    self.add_main_option("serial", ord("s"), GLib.OptionFlags.NONE, GLib.OptionArg.STRING,
AttributeError: 'Application' object has no attribute 'add_main_option'
```
I have no idea the method ```add_main_option``` in where .

System : ubuntu14
Python : python3.4

thx!

TIS-Stefan, 2018-04-10:
Hello 

I searched in the internet for your error message and found

"This requires GLib 2.42 and 14.04 has 2.40."

https://github.com/lutris/lutris/issues/464

Stefan

RozenAstrayChen, 2018-04-11:
Ok I have install GLib 2.42
```
download from http://ftp.gnome.org/pub/gnome/sources/glib/2.42/glib-2.42.1.tar.xz
$ ./configue
$ make
$sudo make install
$ sudo reboot
```
after reboot my GUI it not work.look like i need to update to 16.04

RozenAstrayChen, 2018-04-11:
I have slove GUI can't use problem! but I still got same error, how can I update glib?
```
self.add_main_option("serial", ord("s"), GLib.OptionFlags.NONE, GLib.OptionArg.STRING,
AttributeError: 'Application' object has no attribute 'add_main_option'
```

TIS-Stefan, 2018-04-11:
Hello 

I spoke with the programmers. They said, the minimum Ubuntu version required is 16.04. If you use older versions, you could adapt the Python script in order to run it. 

Stefan

closed 2018-04-11 by TIS-Stefan