TheImagingSource/tiscamera #167: (gst-launch-1.0:7304): GStreamer-CRITICAL

RozenAstrayChen, 2018-04-11:
Hi I have another problem,when i try this tools
```
$ gst-launch-1.0 tcambin ! videoconvert ! autovideosink
```
and got this ...
```
gst-launch-1.0 tcambin ! videoconvert ! autovideosink 
Setting pipeline to PAUSED ...

(gst-launch-1.0:7347): GStreamer-CRITICAL **: gst_caps_get_size: assertion 'GST_IS_CAPS (caps)' failed

(gst-launch-1.0:7347): GStreamer-CRITICAL **: gst_caps_copy_nth: assertion 'GST_IS_CAPS (caps)' failed

(gst-launch-1.0:7347): GStreamer-CRITICAL **: gst_caps_get_size: assertion 'GST_IS_CAPS (caps)' failed

(gst-launch-1.0:7347): GStreamer-CRITICAL **: gst_caps_get_size: assertion 'GST_IS_CAPS (caps)' failed
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
```
and I got same error in __sample/python/pipline__
look like as #166 ...
my use camera is __DMK 33GP1300e__

TIS-Stefan, 2018-04-11:
Can you please use the "ximagesink" instead of "autovideosink" and try again?

TIS-Stefan, 2018-04-11:
May be a firmware issue. What is the camera's serialnumber?

TIS-Stefan, 2018-04-11:
ok, the camera is from last year.

Please do 
`tcam-gigetool info 11714092`
This lists the current firmware in your camera. The last release is 1574. If your firmware is older, please create a request at https://www.theimagingsource.com/support/contact/ because I send firmware files only by email.

The tcam-gigetool is used for firmware updates too.

Stefan

RozenAstrayChen, 2018-04-11:
I don't have ```tcam-gigetool``` just have ```tcam-ctrl```
look like this tools in 
```
/tiscamera/tools/tcam-gigetool
```
should do ```cmake``` in this dir?

RozenAstrayChen, 2018-04-11:
hi I try use ```tcam-ctrl info 11714092``` and got some infomation
```
~/tiscamera/tools$ tcam-ctrl info 11714092
Found 44 propert(y/ies)
GPIO                                    (int) min=0 max=3 step=1 default=2 value=2 category=Special
Strobe Operation                        (enum)  default=Exposure category=Special
						value=Exposure
							Exposure
Strobe Polarity                         (enum)  default=ActiveLow category=Special
						value=ActiveLow
							ActiveHigh
							ActiveLow
Strobe Enable                           (enum)  default=Off    category=Special
						value=Off
							Off
							On
GPIn                                    (int) min=0 max=1 step=1 default=1 value=1 category=Special
GPOut                                   (int) min=0 max=1 step=1 default=0 value=0 category=Special
AutoFunctionsROIEnable                  (bool) default=false value=false category=Unknown
AutoFunctionsROIPreset                  (enum)  default=AutoFunctionsROIPreset_Center50Pct category=Unknown
						value=AutoFunctionsROIPreset_Center50Pct
							AutoFunctionsROIPreset_BottomHalf
							AutoFunctionsROIPreset_Center25Pct
							AutoFunctionsROIPreset_Center50Pct
							AutoFunctionsROIPreset_Custom
							AutoFunctionsROIPreset_Full
							AutoFunctionsROIPreset_TopHalf
AutoFunctionsROILeft                    (int) min=0 max=240 step=1 default=64 value=64 category=Unknown
AutoFunctionsROITop                     (int) min=0 max=0 step=1 default=0 value=0 category=Unknown
AutoFunctionsROIWidth                   (int) min=16 max=192 step=1 default=128 value=128 category=Unknown
AutoFunctionsROIHeight                  (int) min=16 max=16 step=1 default=16 value=16 category=Unknown
Gain                                    (double) min=0 max=9.20819 step=0.001 default=9.20819 value=9.20819 category=Exposure group=429785092
Gamma                                   (double) min=0.01 max=5 step=0.001 default=1 value=1 category=Image group=429785123
Gain Auto                               (enum)  default=Continuous category=Exposure
						value=Continuous
							Continuous
							Off
Gain Auto Lower Limit                   (double) min=0 max=9.20819 step=0.001 default=0 value=0 category=Exposure group=429785092
Gain Auto Upper Limit                   (double) min=0 max=9.20819 step=0.001 default=9.20819 value=9.20819 category=Exposure group=429785092
DecimationVertical                      (int) min=1 max=2 step=1 default=1 value=1 category=Unknown
DecimationHorizontal                    (int) min=1 max=2 step=1 default=1 value=1 category=Unknown
Binning Horizontal                      (int) min=1 max=2 step=1 default=1 value=1 category=Unknown
Binning Vertical                        (int) min=1 max=2 step=1 default=1 value=1 category=Unknown
Offset X                                (int) min=0 max=1024 step=1 default=512 value=512 category=Partial Scan
Offset Y                                (int) min=0 max=1008 step=1 default=504 value=504 category=Partial Scan
Offset Auto Center                      (enum)  default=On     category=Partial Scan
						value=On
							Off
							On
TimestampLatchValue                     (int) min=0 max=281474976710655000 step=1 default=0 value=0 category=Unknown
TimestampLatch                          (button) category=Unknown
TimestampReset                          (button) category=Unknown
AcquisitionBurstFrameCount              (int) min=1 max=1000 step=1 default=1 value=1 category=Unknown
Trigger Denoise                         (double) min=0 max=100000 step=0.001 default=0 value=0 category=Special group=429785097
Trigger Mask                            (double) min=0 max=1e+06 step=0.001 default=0 value=0 category=Special group=429785097
Trigger Debouncer                       (double) min=0 max=1e+06 step=0.001 default=0 value=0 category=Special group=429785097
Trigger Delay (us)                      (double) min=0 max=1e+06 step=0.001 default=54.3 value=54.3 category=Special group=429785097
Software Trigger                        (button) category=Special
Trigger Activation                      (enum)  default=RisingEdge category=Special
						value=RisingEdge
							FallingEdge
							RisingEdge
Trigger Source                          (enum)  default=Any    category=Special
						value=Any
							Action0
							Any
							Line1
							Software
Trigger Mode                            (enum)  default=Off    category=Special
						value=Off
							Off
							On
Trigger Selector                        (enum)  default=FrameStart category=Special
						value=FrameStart
							FrameStart
Highlight Reduction                     (bool) default=false value=false category=Exposure
Exposure Auto Upper Limit Auto          (bool) default=false value=false category=Exposure
Exposure Auto Upper Limit               (double) min=50 max=1e+07 step=0.001 default=66 value=66 category=Exposure group=429785090
Exposure Auto Lower Limit               (double) min=50 max=1e+07 step=0.001 default=10 value=10 category=Exposure group=429785090
Exposure Auto Reference                 (int) min=0 max=255 step=1 default=128 value=128 category=Exposure
Exposure Auto                           (enum)  default=Continuous category=Exposure
						value=Continuous
							Continuous
							Off
Exposure                                (int) min=50 max=10000000 step=1 default=66 value=66 category=Exposure
```

TIS-Stefan, 2018-04-11:
Yes, you got the available properties, but I suppose, you do not get a live video, right? That is the real issue. Is this correct?

TIS-Stefan, 2018-04-12:
Hello
as I wrote, please follow that:

Please do
tcam-gigetool info 11714092
This lists the current firmware in your camera. The last release is 1574. If your firmware is older, please create a request at https://www.theimagingsource.com/support/contact/ because I send firmware files only by email.

The tcam-gigetool is used for firmware updates too.

Stefan

closed 2019-11-29 by TIS-Stefan