TheImagingSource/tiscamera #169: DFK 27BUR0135 Detected but only show monochrome color

javierhilariop, 2018-04-19:
Dear TIS team,
 The OS that I am using is Ubuntu 16.04-LTS. My DFK 27BUR0135 camera is detected as /dev/video0 and I can launch scripts using opencv2 with this dev but it only show monochrome images.
 The webpage say that the DFK27BUR0135 has Linux support. Please anyone else experienced the same issue?

TIS-Stefan, 2018-04-19:
Hello

the camera provides Bayer raw data, which looks gray scale. The debayering must be done in software, therefore, we made our GStreamer modules.
However, if you got your images as  one byte per pixel raw data, you can debayer it with OpenCV functions;

`cvCvtColor(pOpenCVImage,pColorCVImage,CV_BayerGR2RGB);`

The patter "CV_BayerGR2RGB" may need to be changed, until you receive correct colors. After doing so, you will see are greenish image, because the camera does not do automatic white balance. This must also be done in software. Unfortunately I have no OpenCV functions for that. 

If you compile OpenCV 3 with GStreamer support and use our Gstreamer modules,  you can pass a pipeline to the cv::VideoCapture. Currently I do not have a matching sample code, but I can build that.

Stefan

closed 2018-04-19 by TIS-Stefan

javierhilariop, 2018-04-20:
Thanks for the answer. I tried the same code on Windows 10 with opencv without GStreamer and the image is better than in Ubuntu 16.04 which show a monochrome image but the white appear only with direct lights. I appreciate so much a sample code with a color image.
Thanks in advance.

TIS-Stefan, 2018-04-20:
This is, because the Windows driver does a lot of things with the camera images, which V4L2 does not do. 
Regarding color image, I gave a few approaches in my first answer. The automatics and debayering must be done in software, as we do with the Windows driver in software.

Stefan

gonzalesluisfrancisco, 2018-04-24:
Dear TIS-Stefan,
 This is the same issue that I have.  Following your suggestion I had to convert it to gray and then debayer it. Doing that OpenCV shows a colored image but with low quality in comparison with the live_vide.py example. I understand that the reason is the use of the GStreamer functions and wrappers but the CPU use percentage reach the 35% without any additional processing. With an histogram and canny filter it reach the 95%. Only with Opencv function it reach 20% without filters and 80% with canny functions. When I tested the same code on Windows and Opencv the percentage is much less, 7% without filters and 70% with filters. Are these percentage usual or is my Ubuntu or driver with something wrong?

TIS-Stefan, 2018-04-30:
Hello

Please excuse the very long delay of my answer, I was on an exhibition. 

Please describe "low quality". However, I guess, you get a green image, because the camera does not do white balance. Same goes with automatics for gain and exposure. All automatics are done by the GStreamer modules for this camera, therefore, the image does not look as expected. And therefore, you have higher CPU load using the GStreamer modules. 

Stefan