TheImagingSource/tiscamera #170: How to save images by example/cpp/tcam_demo.cpp

AbnerCSZ, 2018-04-20:
I built example/cpp/tcam_demo.cpp and run ./tcam_demo. The program showed the properties normally. But the callback function 'new_frame_cb' does not respond. I don't know why I can't get printf out from this code: 
```
        g_print("Captured frame %d, Timestamp=%" GST_TIME_FORMAT "            \r",
                framecount,
                GST_TIME_ARGS(timestamp));
```
In addition, I can run `gst-launch-1.0 tcambin ! videoconvert ! autovideosink` normally.  
And could you tell me what is the second parameter means in the function 'TcamCamera::set_new_frame_callback(std::function<GstFlowReturn(GstAppSink *appsink, gpointer data)>callback, gpointer data)'  
And  tcam_demo.cpp call this function.:
```
cam.set_new_frame_callback(new_frame_cb, NULL);
```

TIS-Stefan, 2018-04-20:
Hello Abner

maybe the video format and frame rate the tcam_demo uses is too much for your computer. Therefore, please add after

`TcamCamera cam(serial);`

the line

`cam.set_capture_format("BGRA", FrameSize{640,480}, FrameRate{30,1});`

The second parameter of set_new_frame_callback is used as pointer to user data, which can be passed to the callback.

Stefan

AbnerCSZ, 2018-04-20:
nice!  I can show image now. But how can i enter the 
```
if(gst_buffer)map(buffer, &info, GST_MAP_READ))
{
  gst_sample_unmap(buffer, &info);
}
```

TIS-Stefan, 2018-04-20:
I do not understand the question. Did you set a break point there or what are you doing?

AbnerCSZ, 2018-04-20:
Thank you very much! You solved my problem!

closed 2018-04-20 by TIS-Stefan