TheImagingSource/tiscamera #173: Cannot connect more than 1 camera. get_device_list()

tsybaya, 2018-04-23:
Hello!

I am using DMK 33GP2000e 4 cameras, connected to PC. I don't know why, but I cannot open more than 1 camera. The function

```
std::vector<CameraInfo>
gsttcam::get_device_list()
{
    if (!gst_is_initialized())
        throw std::runtime_error("GStreamer is not initialized! gst_init(...) needs to be called before using this function.");
    GstElement* source = gst_element_factory_make("tcambin", "source");
    if (!source)
        throw std::runtime_error("'tcambin' could not be initialized! Check tiscamera installation");
    GSList* serials = tcam_prop_get_device_serials(TCAM_PROP(source));
    std::vector<CameraInfo> ret;
    for (GSList* elem = serials; elem; elem = elem->next)
    {
        gchar *serial = reinterpret_cast<gchar *>(elem->data);
        gchar *name = nullptr;
        gchar *identifier = nullptr;
        gchar *connection_type = nullptr;
        if (tcam_prop_get_device_info(TCAM_PROP(source),
                                      serial,
                                      &name, &identifier, &connection_type))
        {
            CameraInfo info;
            info.serial = serial;
            info.name = name;
            info.identifier = identifier;
            info.connection_type = connection_type;
            ret.push_back(info);
        }

        g_free(name);
        g_free(identifier);
        g_free(connection_type);
    }
    gst_object_unref(source);
    g_slist_free_full(serials, free);

    return ret;
}
```

return only one camera. Actually, only the **last** camera, which has been connected to my pc. If I unplug camera, the method will return vector, which contain only one , but is another camera. I tried all Ethernets ports and different PCI PoE grabber cards, and switch DHCP connection to manual, but it did not help me.

On windows I "see" all cameras.

TIS-Stefan, 2018-04-23:
Hello

I see, that is the same code used in /examples/cpp/common/tcamcamera.cpp, which works on my machine. I would like you to enter in the terminal

`tcam-ctrl -l`

and let me know the output. 
Stefan

tsybaya, 2018-04-23:
TIS-Stefan,

The output is

```
$ tcam-ctrl -l
Available devices:
Model		Type	Serial

DMK 33GP2000e	Aravis	17714167

```

TIS-Stefan, 2018-04-23:
Interesting. Aravis finds one camera.  IP Address conflicts? Please quote me the output of

`tcam-gigetool list`#

(And before you blame me, I could have asked that earlier, but I did not remember that in time. Sorry.)

Stefan

tsybaya, 2018-04-23:
No problem, it is ok.

The interesting detail, that actually I don't have such command
```
$ tcam-gigetool list
tcam-gigetool: command not found
```

I install tiscamera following all your descriptions of steps, but it looks like it is missed

TIS-Stefan, 2018-04-23:
Oh sorry. It is in 

`/build/tools/tcam-gigetools`

You may need to 
```
make
./tcam-gigetool list
```
Stefan

tsybaya, 2018-04-23:
tsybaya@tsybaya-PC:~/Documents/tiscamera/build/tools/tcam-gigetool$ make
[ 81%] Built target tcam-network
[100%] Built target tcam_gigewrapper
[100%] Built target target
tsybaya@tsybaya-PC:~/Documents/tiscamera/build/tools/tcam-gigetool$ sudo make install
[ 81%] Built target tcam-network
[100%] Built target tcam_gigewrapper
[100%] Built target target
Install the project...
-- Install configuration: ""
-- Up-to-date: /usr/lib/libtcam_gigewrapper.so
-- Up-to-date: /usr/lib/python2.7/site-packages/tcamgigetool
-- Up-to-date: /usr/lib/python2.7/site-packages/tcamgigetool/version.py
-- Up-to-date: /usr/lib/python2.7/site-packages/tcamgigetool/__init__.py
-- Up-to-date: /usr/lib/python2.7/site-packages/tcamgigetool/controller.py
-- Up-to-date: /usr/lib/python2.7/site-packages/tcamgigetool/tcam_gigetool.py
-- Up-to-date: /usr/lib/python2.7/site-packages/tcamgigetool/__main__.py
tsybaya@tsybaya-PC:~/Documents/tiscamera/build/tools/tcam-gigetool$ ./tcam-gigetool list
bash: ./tcam-gigetool: No such file or directory

tsybaya, 2018-04-23:
Oh, finally. I have installed it. 

```
tcam-gigetool list
terminate called after throwing an instance of 'tis::SocketCreationException'
  what():  Socket creation returned -1.
Aborted (core dumped)
```
Such a not good things, I think.

TIS-Stefan, 2018-04-23:
Well, nice, not expected.

Can you add a screenshot of the GigE IP Camera Config program, that shows all cameras?
Which Linux computer model do you use?
Which Linux distribution do you use?

tsybaya, 2018-04-23:
Yes, sure.

[image]

All cameras have DHCP option.

This is Firmware tab:
[image]

I'm running Ubuntu 16.04.4 LTS on 
Motherboard: Motherboard Name	Asus X99-Deluxe II
CPU: Intel(R) Core(TM) i7-6850K CPU @ 3.60GHz
GPU: nVIDIA GeForce GTX 560 Ti
RAM: 2x8 GB DDR4-2133 DDR4 SDRAM
Storage: SAMSUNG SSD 830 Series + WDC WD1002FAEX-00Y9A0
Network: 2x PCIe-GIE64+

TIS-Stefan, 2018-04-23:
Ok, that wont work.  The operating systems wont be able to differ between the networks, because all of them start with 169.254 and the subnet mask is 255.255.0.0.

Please follow the instructions for Windows and Linux at:

https://s1.dl.theimagingsource.com/api/2.0/packages/publications/whitepapers-cameras/wpgigecamerastartup/d8fc864407/wpgigecamerastartup_1.0.en_US.pdf

Give each network a fixed IP as
- 192.168.2.1: 255.255.255.0
- 192.168.3.1: 255.255.255.0
- 192.168.4.1: 255.255.255.0
- 192.168.5.1: 255.255.255.0
And the cameras accordingly
- 192.168.2.2: 255.255.255.0
- 192.168.3.2: 255.255.255.0
- 192.168.4.2: 255.255.255.0
- 192.168.5.2: 255.255.255.0

tsybaya, 2018-04-24:
I set up each network and camera as you said. Right now I have two cameras instead of one in tcam-gigetool list
```
tsybaya@tsybaya-PC:~/CLionProjects/LtNanoPi/bin$ tcam-gigetool list

------------------------ Interface: enp4s0 [192.168.5.1] ------------------------

     Model Name      |  Serial Number   |  User Defined Name   |     Current IP   
DMK 33GP2000e        | 17714170         | ir3                  | 192.168.5.2      

------------------------ Interface: enp6s0 [192.168.3.1] ------------------------

     Model Name      |  Serial Number   |  User Defined Name   |     Current IP   
DMK 33GP2000e        | 17714168         | ir1                  | 192.168.3.2    
```

but get_device_list(); (from example) return only one camera (DMK 33GP2000e        | 17714170         | ir3                  | 192.168.5.2      ) and second camera is inaccessible.

tsybaya, 2018-04-24:
[image]

```
tsybaya@tsybaya-PC:~$ tcam-gigetool list
terminate called after throwing an instance of 'tis::SocketCreationException'
  what():  Socket creation returned -1.
Aborted (core dumped)
```

tsybaya, 2018-04-24:
Sometimes the output is:
```
$ tcam-gigetool list

------------------------ Interface: enp4s0 [192.168.5.1] ------------------------

     Model Name      |  Serial Number   |  User Defined Name   |     Current IP   
DMK 33GP2000e        | 17714170         | ir3                  | 192.168.5.2      

------------------------ Interface: enp5s0 [192.168.4.1] ------------------------

     Model Name      |  Serial Number   |  User Defined Name   |     Current IP   
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
[... 230 lines cut ...]
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2      
[... cut ...]
DMK 33GP2000e        | 17714168         | ir1                  | 192.168.3.2      

------------------------ Interface: enp7s0 [192.168.2.1] ------------------------

     Model Name      |  Serial Number   |  User Defined Name   |     Current IP   
DMK 33GP2000e        | 17714167         | ir0                  | 192.168.2.2    

```
but if i try again:

```
tsybaya@tsybaya-PC:~$ tcam-gigetool list
terminate called after throwing an instance of 'tis::SocketCreationException'
  what():  Socket creation returned -1.
Aborted (core dumped)

```

tsybaya, 2018-04-25:
TIS-Stefan.

The problem was with the camera 
`DMK 33GP2000e        | 17714169         | ir2                  | 192.168.4.2  `
I replaced it within another one, and everything works great. Thank you so much!

closed 2018-04-25 by tsybaya