TheImagingSource/tiscamera #174: timestamp from camera

tsybaya, 2018-04-23:
Hello,

I am using DMK 33GP2000e.

How to get timestamp exactly from the camera driver (or device)?

I am going to synchronize camera between each other by timestamps, but for this I need reset clocks on cameras simultaneously every time. 
[Here example for resetting counter for another cameras](
https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&ved=0ahUKEwilmeHznNDaAhVHqJQKHR49DTMQFggqMAA&url=https%3A%2F%2Fwww.matrix-vision.com%2Fprofessional-article%2Ftimes-are-changing.html%3Ffile%3Dtl_files%2Fmv11%2FWhitepapers%2FEN%2FTimestamp%2520synchronization.pdf&usg=AOvVaw1x_CHLBjRlhwtovuK_vbwb)

Does it possible to reset counter by trigger? or start counter on all cameras simultaneously by trigger?

tsybaya, 2018-04-23:
TIS-Stefan,

I am using C++.

tsybaya, 2018-04-23:
TIS-Stefan,

Thank you for the information. I would appreciate if you provide any information for it.
My email (In a case): tsyba.y.a@gmail.com

tahasadr, 2018-05-22:
Hi tsybaya.
could you get timestamp?

closed 2019-11-29 by TIS-Stefan