TheImagingSource/tiscamera #178: [DFK27BUR0135]Trigger-mode

gonzalesluisfrancisco, 2018-04-27:
Dear TIS team,
The trigger-images Python example shows only a static image after the "source.set_tcam_property("Trigger Mode", True)" line. When I try to snap other picture by the space(" ") and enter nothing happen. Other examples like the properties does not show any video or capture and give an error("valve is broken"), the property_dialog shows the video live desired but when I switch to trigger mode it freeze and after certain seconds a disconnected device message box pop-up.
Please I appreciate so much any help about how to capture pictures by software using python for the DFK27BUR0135 camera.
Note: The  list_properties example needs the select_camera function implemented in list_devices.

TIS-Stefan, 2018-04-30:
Hello

I tested the sample on my computer and had no issues.  
Do you get an image stream, if the line
`source.set_tcam_property("Trigger Mode", True)`
is commented out?

The sample, that uses the "valve" seems to have an issue with GStreamer 1.0... so please ignore that sample, until next update.

Stefan

gonzalesluisfrancisco, 2018-04-30:
Dear Stefan,
 Thanks for your answer. When I comment the line that you indicates the stream is fine as expected. But the space command no cause any new snap in the original script.
Thanks for the answer.
Best regards.

TIS-Stefan, 2018-05-02:
Hello Gonzales

I found it..... The 27UR0135 has hardware trigger only, no software trigger. For unknown reasons the software triggering did not work.

I am very sorry for the long delay of figuring that out.

Stefan

gonzalesluisfrancisco, 2018-05-02:
Dear TIS-Stefan,
 Thanks for the answer. I understand the response but I still confuse because on IC Capture the snap function works correctly.
 Is there any way to implement the snap using Python, OpenCV and the gstreamer modules?
 Best regards

TIS-Stefan, 2018-05-03:
Hello

In my notices to the camera it says, the DFK 27BUR0135 has no software trigger. Unfortunately I do
not have such a camera on stock, therefore, I am not able to confirm, that software trigger works.

You really enabled trigger mode in the Device Properties dialog and clicked the "Softwaretrigger"
button and got one single image?

Also, I wonder, whether you understand trigger mode correctly. Trigger mode means, the camera sends an image only, if an extern trigger pulse, by hardware or software, if supported, is sent to the camera. While there is no trigger pulse, you wont get any images streamed.
Trigger mode does not mean, you get the images automatically somewhere, where you can handle them.
I have the faint suspect, you only want to get one image from the live streamed images from the camera. Am I correct here or am I completely wrong?

Stefan

gonzalesluisfrancisco, 2018-05-03:
Dear TIS team,
 I try the snap capacity on IC Capture - Windows. The image attached is the capture using the snap button. On Ubuntu 16.04 LTS the script of properties for Trigger Mode gives an error. My objective is snap a picture with a serial command through "COM" port. I already wrote some code  to achieve this based on your TIS.py example but I am testing it. In case of success I share the code but is only pull if a new sample is available on certain condition.
Thanks for your answer.

[image]

TIS-Stefan, 2018-05-04:
Aha, as expected. Snap means, IC Capture software takes an image from the live stream. The camera is *not* triggered. The trigger mode button is two buttons left from the "snap" button.
That means, in Windows you dot *not* run the camera triggered, as you do in Linux with the "trigger-image" sample. That means you do completely different things.
That also means, the "trigger-image" sample is not, what you need. You only want to snap am image form the live stream.  What you need is shown in the "live-video.py" sample!

Stefan

closed 2019-11-29 by TIS-Stefan