TheImagingSource/tiscamera #180: Unable to set framerate with GRAY8

Wurzelsepp123, 2018-05-14:
I'm trying to use two monochrome USB3.0 cameras with an active USB3.0 hub and encountered some problems. Using both cameras at FHD with default FPS breaks the bandwidth of USB3.0 on my PC and the cameras flicker. Reducing the image size is not possible for my project so i tried to reduce the framerate, but the framerate settings are somehow ignored (always 90fps in v4l2-ctl).  

> gst-launch-1.0  -v tcamsrc serial=serial1 !  video/x-raw,format=GRAY8,width=1920,height=1080,framerate=30/1 ! videoconvert !  autovideosink tcamsrc serial=serial2 ! video/x-raw,format=GRAY8,width=1920,height=1080,framerate=30/1 ! videoconvert ! autovideosink

Using the same command with 640x480 resolution, i can change the FPS to all the supported values.
I finally found a workaround by using GRAY16_LE instead of GRAY8, which allows me to change the FPS even with FHD.  I would prefer to use GRAY8 though and use the remaining bandwidth for more fps/higher resolution, instead of color depth.

Any Idea how to fix this with GRAY8 (possibly with higher fps)?

Supported formats:
> video/x-raw, format=(string)GRAY8, width=640, height=480, fps={300.000000, 240.003840, 120.000000, 60.000000, 30.000000, 15.000000, 5.000000, 1.000000}
> video/x-raw, format=(string)GRAY8, width=1920, height=1080, fps={90.000000, 120.000000, 60.000000, 30.000000, 15.000000, 5.000000, 1.000000}
> video/x-raw, format=(string)GRAY8, width=2048, height=1536, fps={60.000000, 30.000000, 15.000000, 5.000000, 1.000000}
> video/x-raw, format=(string)GRAY16_LE, width=640, height=480, fps={300.000000, 240.003840, 120.000000, 60.000000, 30.000000, 15.000000, 5.000000, 1.000000}
> video/x-raw, format=(string)GRAY16_LE, width=1920, height=1080, fps={90.000000, 60.000000, 30.000000, 15.000000, 5.000000, 1.000000}
> video/x-raw, format=(string)GRAY16_LE, width=2048, height=1536, fps={50.000000, 30.000000, 15.000000, 5.000000, 1.000000}

TIS-Stefan, 2018-05-14:
Hello

Which camera model do you use?

Stefan

Wurzelsepp123, 2018-05-14:
I'm using DMK 33UX265 on Ubuntu 18.04 with latest gstreamer etc.

TIS-Stefan, 2018-05-16:
Hello 

It seems to be a firmware issue of the DMK cameras in Linux. In Windows they work as expected.  We investigate into that. 

Stefan

TIS-Stefan, 2018-05-23:
Hello Wurzelsepp

the problem is fixed with a new firmware.
[fw926.zip](https://github.com/TheImagingSource/tiscamera/files/2030191/fw926.zip)
In the zip file is a Windows program for the upload.  I tried the Linux firmware-update program
./firmware-update -u -d 02710360 -f ./33UFirmware.926.fwpack ``
but it gave me an error code. If you try that in Linux too, you may ignore the warning about "Do not disconnect the camera". Better use Windows.

The maximum frame rate for GRAY8 640x480  is still wrong, is is 300 but should be 370.

Stefan

Wurzelsepp123, 2018-05-23:
Hello Stefan,
i flashed both cameras with the new firmware(using windows) and now everything works as expected.
So far the USB bandwidth seems to be enough to handle both cameras with GRAY8 @60FPS, which is all i wanted. 
Thanks for your help.

closed 2018-05-23 by Wurzelsepp123

TIS-Stefan, 2018-05-23:
Hi WurzelSepp

Thank you for your feedback.

Stefan