TheImagingSource/tiscamera #181: Images coming with artefacts

Kolya96, 2018-05-19:
I've connected the camera to a microcomputer (particularly, Odroid XU4) and got very strange result while triggering the camera.
Camera works good with normal PC and Raspberry Pi 3 model B, but with this Odroid I get only such images:

[image]

Kolya96, 2018-05-20:
More examples:
[image]
[image]

TIS-Stefan, 2018-05-22:
Hello

The image data is incomplete, therefore blocks are missing and the rest of the image is "moved up". So you get these tile style.

Which camera do you use?
Which software do yo use?
Please try lower frame rates and see, whether the issue disappears.
If other devices are connected to the same USB controller, please disconnect them and try again.
You may also disconnect the network cable and try again. The network and USB use the same controller.

Stefan
--

Kolya96, 2018-05-22:
Hello, Stefan!
Thanks for your response.

Computer: Odroid XU4 (https://wiki.odroid.com/odroid-xu4)
Camera: DFK 33UX178
System: ubuntu-16.04.3-4.14-minimal

Camera is connected via usb3 to the computer. Near with camera here is connected just one usb-modem for ssh, nothing more. 

As a code I use base code of pipeline.c with slight modifications.

```
    pipeline = gst_parse_launch("tcambin name=src \
                                ! tee name=t t. \
                                ! queue \
                                ! valve name=valve \
                                ! videoconvert \
                                ! jpegenc \
                                ! multifilesink \
                                 name=filesink post-messages=true \
                                 location=images/image-live-%d.jpg t. \
                                ! queue \
                                ! videoconvert ", &err);
```

After I just set the state of pipeline to "GST_STATE_PLAYING" and images start to come. Every image comes with missing blocks.

Same code was used with Raspberry Pi 3 B (system Raspbian) with usb2 port. Images as expected came slower, but there was no missing blocks.

P.S.: I have installed all build and runtime dependencies (also I had to include gtk+-3.0 for gtk.h file) to a "fresh" and clear systems.

TIS-Stefan, 2018-05-22:
Hello Kolya

I do not have an own Odroid XU4 currently.

Please try a lower resolution on the console
`gst-launch-1.0 tcambin ! video/r-raw, format=BGRA,width=640,height=480,framerate=30/1 ! videoconvert ! ximagesink`

If this works, list with 
`tcam-ctrl -c <serialnumber>`
avaiable video formats and try higher resolutions, until you get a problem. Please let me know, at which resolution and frame rate trouble begins.

Stefan

Kolya96, 2018-05-22:
I have inserted your code into my, so I have now:

```
    pipeline = gst_parse_launch("tcambin name=src \
                                ! tee name=t t. \
                                ! queue \
                                ! valve name=valve \
                                ! video/r-raw, format=BGRA,width=640,height=480,framerate=30/1 \
                                ! videoconvert \
                                ! jpegenc \
                                ! multifilesink \
                                 name=filesink post-messages=true \
                                 location=tmp/imgs/image-live.jpg t. \
                                ! queue \
                                ! videoconvert ", &err);
```
And now I get no images at all.

Also I tried to run it directly from a ssh and I get this error:
```
hedef@odroid:~/tiscamera/examples/c$ gst-launch-1.0 tcambin ! video/r-raw, format=BGRA,width=640,height=480,framerate=30/1 ! videoconvert ! ximagesink
WARNING: erroneous pipeline: could not link tcambin0 to videoconvert0, videoconvert0 can't handle caps video/r-raw, format=(string)BGRA, width=(int)640, height=(int)480, framerate=(fraction)30/1
```

Kolya96, 2018-05-22:
Now I tried to run following command:
`sudo gst-launch-1.0 v4l2src device=/dev/video0 ! queue ! videorate ! video/x-raw,width=640,height=480,framerate=30/1 ! jpegenc ! multifilesink location="./frame%06d.jpg"`

I got missing blocks in first 20 frames.
When I changed resolution to 1920x1080 I had no missing blocks in any image. 
At 2048x2048 I had missing blocks in EVERY (of 65) image. Same for 3072x2048.
Also in every try I executed command for like 4 seconds and in last 2 executions (2048x2048 and 3072x2048 resolutions) I got 65 and 21 frames respectively.

Kolya96, 2018-05-22:
When I tried you command I get this error:
```
hedef@odroid:~/tiscamera/examples/c/test_cam$ sudo gst-launch-1.0 tcambin ! video/x-raw, format=BGRA,width=640,height=480,framerate=30/1 ! videoconvert ! ximagesink
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstXImageSink:ximagesink0: Could not initialise X output
Additional debug info:
ximagesink.c(860): gst_x_image_sink_xcontext_get (): /GstPipeline:pipeline0/GstXImageSink:ximagesink0:
Could not open display
Setting pipeline to NULL ...
Freeing pipeline ...
```
So I will proceed with command i have user previously.
I got no missing blocks with 15 fps at 3072x2048 and first 5 images with missings at 5 fps.
Also I have noticed that execution of command for 15 fps has resulted in 42 frames for 6.64 seconds of execution. That is like 6 fps in reality. And 31 frame for 7.23 sec of execution for 5 fps, that is like 4 fps.

P.S.: But

TIS-Stefan, 2018-05-22:
Is it possible, you do not run XWindows?  That would explain, why you removed the display sink in your pipeline.

Try this:
`gst-launch-1.0 tcambin ! video/x-raw, format=BGRA,width=640,height=480,framerate=30/1 ! videoconvert ! aasink
`

Kolya96, 2018-05-22:
We will use the camera in unmanned aircraft, and we will connect to it only via ssh, so we are not able to use any desktop environment, only console. If you meant this.

Your command at 640x480 worked good. At 1920x1080 I got some latency (like 0.8 sec I think), at 2048x2048 latency was greater, like 1 sec.

TIS-Stefan, 2018-05-22:
However, it is hard to see, whether you have incomplete images with the aasink. Therefore, you may try to setup stream into an UDP or TCP sink. GStreamer modules are available for this. With some tricks you can see the result in VLC in your other computer.

The latency is unfortunately to be expected, since the amount of data is very high. 

Stefan
--

Kolya96, 2018-05-22:
Since we do not need live-stream in its exact meaning (we just need to get N images in a second with equal time intervals to process them), so stream is not necessary. Our main problem is solved for now — we can get correct images, but with lower fps.

So, you think it is because of some restrictions in file size to transfer in Odroid? Because we got correct images with raspberry pi 3 B which hasn't usb3 interface and has worse processor.

closed 2018-05-22 by Kolya96