TheImagingSource/tiscamera #182: what is GstClockTime timestamp

tahasadr, 2018-05-23:
Hi TIS team,
What is timestamp provided by the below line in tcam_demo.cpp:
`GstClockTime timestamp = GST_BUFFER_PTS(buffer);`
Is it timestamp created by the camera?

TIS-Stefan, 2018-05-23:
Hello

Currently the time stamps are generated by software and not by cameras.

Stefan

tahasadr, 2018-05-26:
Hi @TIS-Stefan
Thanks for your reply.
How can i get the camera timestamp ?
There is another open issue about camera that.
you said that we can get it by aravis directly. could you help us to get camera timestamp by a sample code?

TIS-Stefan, 2018-05-28:
Hello

I have to admit never used Aravis on my own.. However, I searched in the internet and found 
https://aravisproject.github.io/docs/aravis-0.4/ArvBuffer.html
There is a "arv_buffer_get_timestamp ()" on the "ArvBuffer". That should do the job.

Stefan

closed 2019-11-29 by TIS-Stefan